SpikeInterface/spikeinterface #2020: read_phy and read_kilosort have varying arguments

Cumol, 2023-09-19:
read_phy is missing the "remove_empty_units" argument and read_kilosort is missing the "explude_cluster_groups" argument.

Its puzzling since it seems like both the functions should be identical per https://github.com/SpikeInterface/spikeinterface/blob/fc954653588e274f239f8b6967918495bb6a316b/src/spikeinterface/extractors/phykilosortextractors.py#L256

`sorting_curated = se.read_phy('/mnt/hpc_slurm/projects/OWzeronoise/Jove_paper/369/20221101/sorted/sorter_output/', exclude_cluster_groups = ["noise"], remove_empty_units=True)`

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[4], line 2
      1 ##sorting_curated = se.read_kilosort('/mnt/hpc_slurm/projects/OWzeronoise/Jove_paper/369/20221101/sorted/sorter_output/', exclude_cluster_groups = ["noise"], remove_empty_units=True)
----> 2 sorting_curated = se.read_phy('/mnt/hpc_slurm/projects/OWzeronoise/Jove_paper/369/20221101/sorted/sorter_output/', exclude_cluster_groups = ["noise"], remove_empty_units=True)

TypeError: __init__() got an unexpected keyword argument 'remove_empty_units'
```

`sorting_curated = se.read_kilosort('/mnt/hpc_slurm/projects/OWzeronoise/Jove_paper/369/20221101/sorted/sorter_output/', exclude_cluster_groups = ["noise"], remove_empty_units=True)`

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[6], line 1
----> 1 sorting_curated = se.read_kilosort('/mnt/hpc_slurm/projects/OWzeronoise/Jove_paper/369/20221101/sorted/sorter_output/', exclude_cluster_groups = ["noise"], remove_empty_units=True)

TypeError: __init__() got an unexpected keyword argument 'exclude_cluster_groups'
```

alejoe91, 2023-09-19:
Hi! Indeed they are based off the same class, but they serve different purposes. The underlying format is the same, but Phy adds additional labels (hence the `exclude_cluster_groups` option). KS instead can return emtpy templates, so we give the option to remove it.

The 2 functions are therefore NOT identical ;)

Cumol, 2023-09-19:
Then I might be confused in how to use them properly. 

Looking at the help for read_kilosort, it suggests that there should be multiple arguments, like exclude_cluster_groups:

```
Help on class KiloSortSortingExtractor in module spikeinterface.extractors.phykilosortextractors:

class KiloSortSortingExtractor(BasePhyKilosortSortingExtractor)
 |  KiloSortSortingExtractor(folder_path, keep_good_only=False, remove_empty_units=True)
 |  
 |  Load Kilosort format data as a sorting extractor.
 |  
 |  Parameters
 |  ----------
 |  folder_path: str or Path
 |      Path to the output Phy folder (containing the params.py).
 |  exclude_cluster_groups: list or str, optional
 |      Cluster groups to exclude (e.g. "noise" or ["noise", "mua"]).
 |  keep_good_only : bool, default: True
 |      Whether to only keep good units.
 |      If True, only Kilosort-labeled 'good' units are returned.
 |  remove_empty_units : bool, default: True
 |      If True, empty units are removed from the sorting extractor.
```

But using it outputs the above error.

alejoe91, 2023-09-19:
Ah you're right! The docstring is wrong and it doesn't correspond to the actual function signature! I'll push a fix!

PR #2022 (2023-09-21): Fix Kilosort Phy reader docstrings
Fixes #2020
closed 2023-09-21 by alejoe91