SpikeInterface/spikeinterface #2021: store path to data in phy export to relative instead of absolute path

Cumol, 2023-09-19:
When exporting to phy, the `params.py `file contains the path to the recording.dat file as an absolute path. The moment the folder is moved to another machine, phy fails to open as the path is wrong.

https://github.com/SpikeInterface/spikeinterface/blob/843b451282369565ca8e62863b4814c772c479f8/src/spikeinterface/exporters/to_phy.py#L152C18-L152C18

I wonder if it makes sense to turn it into a relative path. I tried it on one session and it resolved the issue. Unless that breaks some other functionality I am not aware of.

alejoe91, 2023-09-19:
Hi @Cumol 

Yes I think we should give the option to use relative paths with respect to the output folder. We have a similar option for the waveform extractor that we could propagate here. Thanks for the suggestion!

zm711, 2023-09-20:
Just in case people search here in the future how the `export_to_phy` behaves related to native KS behavior:

KS1 and KS2 supply just the name of the binary file in the `dat_path` 
KS2.5 and KS3 supply the absolute path in the `dat_path` (specifically to the `temp_wh.dat`.

zm711, 2023-09-27:
@Cumol, the option to use a relative path is now merged with #2041. To use you would need to install spikeinterface from source ([instructions](https://spikeinterface.readthedocs.io/en/latest/installation.html#from-source)) or wait for the next pypi release. The new keyword argument is `use_relative_path`. 

https://github.com/SpikeInterface/spikeinterface/blob/3826a2d8b5a02e53af597f568256fa27b20ee620/src/spikeinterface/exporters/to_phy.py#L68-L70

closed 2023-09-27 by zm711