SpikeInterface/spikeinterface #2023: Dense vs Sparse for `export_to_phy

zm711, 2023-09-19:
While we are on the thinking about changes to the `export_to_phy`. Phy uses the presence of `template_ind.npy` to assume sparsity whereas for KS2 the file KS2 returns is actually `templates_ind.npy` and so Phy assumes that the templates will be dense. Should spikeinterface use this fact such that if the user provides sparsity it returns an actual `template_ind.npy` so Phy can create the sparse templates and if the user does not provide sparsity/have sparsity the exporter instead will write the `np.ones` to `templates_ind.npy`.

https://github.com/cortex-lab/phylib/blob/2b567cd2e2c382bd5e8926e9abd9a2f9353bfd76/phylib/io/model.py#L691-L727 For how phylib handles sparse vs dense templates.

relevant code lines in spikeinterface
https://github.com/SpikeInterface/spikeinterface/blob/2aabd146b919ac4c907d8fd017115c7c60717486/src/spikeinterface/exporters/to_phy.py#L91-L98
https://github.com/SpikeInterface/spikeinterface/blob/2aabd146b919ac4c907d8fd017115c7c60717486/src/spikeinterface/exporters/to_phy.py#L172-L189

alejoe91, 2023-09-19:
@zm711 I think I don't understand the question. We save the `temnplate_ind.npy` (the correct name) and we populate it with the actual sparsity of the waveform extractor or the externally provided sparsity. Aren't we?

alejoe91, 2023-09-19:
Or are you suggesting to just avoid saving the `template_ind.npy` if no sparsity is used?

zm711, 2023-09-19:
There's this line

```python
else:
    used_sparsity = ChannelSparsity.create_dense(waveform_extractor) 
```

I'm not sure how often it actually get calls. But in this case they are dense. And you could either save no `template_ind.npy` or save it as `templates_ind.npy` instead.

zm711, 2023-09-19:
phylib relies on a lot of `IOError` calls to determine the structure of the data. So I agree minimizing the extra files would make troubleshooting easier. The error that inspired this is: https://github.com/cortex-lab/phy/issues/1219. I think it ended up being an absolute vs relative path issue (similar to #2021) with either absolute path being too long-which I asked them to test), but I'm waiting for the user to respond.

PR #2148 (2023-11-02): Change phy exporter to not save `template_ind` in the case of dense waveform_extractor
Fixes #2023. Just trying to do a cleanup of the issue tracker for 0.99.0. Definitely could go later though.

2 things in this PR
1) As discussed in #2023, there is no point in saving the `template_ind.npy` if the data is dense since it is suppose to be the sparsity for phy. So I just set a boolean to not save `template_ind.npy` if the waveform extractor is dense. Phy will use the lack of `template_ind.npy` to treat it as a dense dataset so Phy's behavior isn't affected (might even be improved since it will t [... cut]
closed 2023-11-02 by alejoe91