SpikeInterface/spikeinterface #2030: Are hosted docker images ever changed?

JoeZiminski, 2023-09-20:
Just a quick question about the sorter images. I am considering pre-downloading them and storing them on our HPC so people do not have to worry about pulling them from docker, which is causing some problems on our HPC at the moment.

I guess the images that are download do need to be versioned because they are very light, just serving as a base that pikeinterface is installed in. However, I just wanted to check they are not liable to change between SI versions as this might require extra checks for my plan above.

Cheers!

alejoe91, 2023-09-21:
Hi @JoeZiminski 

They do change very sporadically. This is mainly true for Kilosort, because we have a `.m` wrapper in SI that needs to be updated in the image if something changes. 

One note: if you run sorter `docker_image/singularity_image=True` and the image available on the machine is not the `latest`, it will download and cache the new one (at least docker will, not sure about singularity). The alternative could be setting an image explicitly when calling the `run_sorter` function, e.g.:

```
sorting_KS25_fixed = ss.run_sorter(..., singularity_image="spikeinterface/
kilosort2_5-compiled-base:0.2.0")
```

If we modify the image, we make a new tag so in this case you are absolutely sure that the image will never change.

JoeZiminski, 2023-10-16:
Hey @alejoe91 thanks a lot, apologies for the delayed reply. This is really useful, cheers!

closed 2023-10-16 by JoeZiminski