SpikeInterface/spikeinterface #2034: channel map discrepancy, spikeinterface 0.98.1, probeinterface 0.2.18, read_openephys

gmaggi, 2023-09-21:
Hi,

We are using spikeinterface 0.98.1 and probeinterface 0.2.18 (just updated via pip install git+https://github.com/SpikeInterface/probeinterface.git). 

A user has found a discrepancy between the channel map (.prb file) that is given as an input to read_prb at probeinterface.  She uses cambridge neurotech probe. The .prb file is attached as .txt
[CambridgeNT_H5_CB.txt](https://github.com/SpikeInterface/spikeinterface/files/12685134/CambridgeNT_H5_CB.txt)

we ran the following function for the extractor part:

[image]

Then the user runs kilosort2

The output generated (here visualized with phy2): 
[image]

Many channels seem to be missing. I expect 64 channels with this geometry:  

[image]
 
Any help would be much appreciated.

alejoe91, 2023-09-21:
Hi @gmaggi 

The probe is loaded correctly on my side:

```
import probeinterface as pi
from probeinterface.plotting import plot_probegroup

probegroup = pi.read_prb("/home/alessio/Downloads/CambridgeNT_H5_CB.txt")
plot_probegroup(probegroup)
```

[image]

Even after loading it into a "fake" spikeinterface recording, the probe appears correct:
```
import spikeinterface as si
import spikeinterface.widgets as sw

rec = si.generate_recording(num_channels=72)
rec_prb = rec.set_probes(probegroup)

sw.plot_probemap(rec_prb)
```

[image]

```

Note that KS removes channels with low activity. So that might be the reason. You can try to `export_to_phy` from SpikeInterface!

gmaggi, 2023-09-22:
@alejoe91 Thank you very much for your help and great support.

closed 2023-09-22 by gmaggi