SpikeInterface/spikeinterface #2038: `recording.set_probegroup()` fails because because channel count in both match

stothe2, 2023-09-24:
I'm wondering why the below is `>=` and not `>`? As when attaching a probe group to a recording object, shouldn't it be OK if the channel counts in both exactly match? 

https://github.com/SpikeInterface/spikeinterface/blob/ff1b8cdb59f0febb21354cb14a3ea2fc7577b202/src/spikeinterface/core/baserecordingsnippets.py#L147

Sorry if I'm missing something very obvious.

stothe2, 2023-09-25:
Thank you for the detailed reply!

That makes sense. But is it required for `device_channel_indices` to be 0-indexed (or for that matter, even in sequential order?)? I had created a probe with indices starting from 1, and thus ran into this issue. You're correct that `len(device_channel_indices)` wouldn't work, but how about `len([index for index in device_channel_indices if index != -1])`?

alejoe91, 2023-09-25:
@zm711 you are perfectly right! Thee `device_channel_indices` needs to be 0-based since it is the index of the traces which each contact correspond to, and Python is 0-based ;)

It would be good to double check those lines given the note, so let's keep this open for the moment!

alejoe91, 2023-09-25:
@stothe2 it doesn't need to be on sequential order! It should reflect the channel mapping between your probes and the acquisition system. We are correctly working on improving the docs to better explain this important step

samuelgarcia, 2023-09-25:
Hi.
The reponse of Zach is correct.

device_channel_indices is **always zero based** in probeinterface/spikeinterface api.

This is the link between contact_id and channel_id but with knowing in advance the channel_ids list.
This is very convinient to describe a probe + mapping (that depend on a hardware) wihtout the channels names that mainly depend on the acquisition sofware (openephyx vs blackrock for instance.

For instance we could have 72 channels with 2 probes of 32 and 8 channels that are extra input.
The probeA will certanilly be 0-31 channel indices and probeB 32-63 and then 64-71 would be the no prbe recording part.
But sometimes some headstage can mix channels in a very strange order and the 2 probes can be interleaved!
This is why we need this flexible device_channel_indices on probeinterface side.

zm711, 2023-09-25:
Also including links to the current probeinterface docs that demonstrate some real and some 'toy cases' of the wiring while the docs are being updated.

https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html
https://probeinterface.readthedocs.io/en/main/examples/ex_05_device_channel_indices.html#sphx-glr-examples-ex-05-device-channel-indices-py

stothe2, 2023-10-02:
Thank you all for the quick, detailed responses! It truly helped. I assumed `device_channel_indices` needn't be 0-based, and was thus running into issues.

closed 2023-10-02 by stothe2