SpikeInterface/spikeinterface #2039: concatenate recordings spikeinterface 0.98.1

gmaggi, 2023-09-25:
Hi,

I have a question.

Imagine there are two sub recordings that have to be analyzed together (via GLX extractor, then KS2). How would it work considering there are in the same folder?. I mean, for read_openglx, one has to pass an entire directory path (no an individual recording), so if the two of them are in the same folder, what read_openephys will do?

after having rec1 and rec2, like:
rec1 = read_openephys(WHATPATH)
rec2 = read_openephys(WHATPATH)

I can imagine I should use `from spikeinterface import concatenate_recordings`

Thanks in advance for your help,
Giuliano.

alejoe91, 2023-09-27:
Hi Giuliano,

Sorry I don't understand, do you have Open Ephys recordings? (Or is it SpikeGLX - I guess it's what you mean with GLX)?
In that case, you can just `read_openephys`. If you have multiple recordings, they will show up as multiple segments. If you `print(recording)` you'll see there is more than one segment.

From a multi-segment recording, you can just use the  `concatenate_recordings` as follows:

```
recording_mono_segment = si.concatenate_recordings([recording])
```

The `recording_mono_segment` is then compatible with Kilosort2.

gmaggi, 2023-09-27:
Indeed, this is GLX recording.
Ok, thanks! now is even simpler compared to 0.12. At that time we did:

```
recordings=[]
for inputFile in inputFileS:
       segment = se.SpikeGLXRecordingExtractor(inputFile)
       recordings.append(segment)
       
 recording = se.MultiRecordingTimeExtractor(recordings)      
```

zm711, 2023-11-01:
@gmaggi,

Did you test Alessio's fix and you're happy with it or did you need to clarify more what you wanted? Just checking up on issues and making sure they've been answered sufficiently.

gmaggi, 2023-11-03:
Hi,

Thank you for checking this and sorry for my lack of feedback.
we have not really tested yet. But feel free to close this ticket and if needed I can re open it.

Thanks!

zm711, 2023-11-03:
Sounds like a plan. Thanks!

closed 2023-11-03 by zm711