SpikeInterface/spikeinterface #2042: run_sorter_jobs() function not found

bengrannan, 2023-09-26:
I am attempting to use run_sorter_jobs() for batch sorting per https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html?highlight=run_sorter_jobs#running-several-sorters-in-parallel . I am not able to import the function, however. The documentation instructs to use run_sorter_jobs() over run_sorters() (deprecated).  Is run_sorter_jobs() in current version? I am using version 0.98.2. 

[image]

zm711, 2023-09-26:
@bengrannan 

It is not in 0.98.2. To use the new function you would have to install from source. [Instructions for install](https://spikeinterface.readthedocs.io/en/latest/installation.html). The documentation is updated as new features are added to prevent them from getting out of sync, but that means it often recommend users install from source following the instructions (I linked above) to get all the latest features. Eventually this will not be the case, but in this pre-1.0 stage, SpikeInterface is still making a lot of optimizations and thus some API changes. Does that make sense?

samuelgarcia, 2023-09-27:
Hi Ben and Zach.
Alternatively you check the doc of the latest relase with this kind of url : https://spikeinterface.readthedocs.io/en/0.98.2/
In the latest release you need to use `run_sorters()` but I recommand to switch to the new run_sorter_jobs and so go for sources as Zach mentioned.

alejoe91, 2023-09-27:
Maybe in the README we should have both the a link to the released docs and the latest one?

zm711, 2023-09-27:
That's probably a good idea. This is the 2nd or 3rd time that I've seen this issue.

zm711, 2023-10-06:
@alejoe91 @samuelgarcia 

Added a link to the pypi docs on the main page and linked to install from source. Did you want to still leave this open or are we good to close?

@bengrannan you can always feel free to open more issues if things come up!

alejoe91, 2023-10-06:
I think we can close! Should be clear now :)

closed 2023-10-06 by alejoe91