SpikeInterface/spikeinterface #2063: Saving data in uV

JonathanAMichaels, 2023-10-03:
How can I ensure that processed data is saved in uV?
In this example using a spikeglx recording object, I can't figure out if gain scaling is occuring. I think it's saving in the same sample units as the original recording.

  ```
    rec_filtered = si.bandpass_filter(raw_rec, freq_min=1., freq_max=300.)
    rec_shifted = si.phase_shift(rec_filtered)
    rec_preprocessed = si.resample(rec_shifted, 1000)
    rec_preprocessed.save(folder=lfp_folder, **job_kwargs)
```

samuelgarcia, 2023-10-03:
Hi Jonathan.
At the moment we do not have the option `rec.save(scaled=True)` so the recording is save as the original format BUT  the gain and offset are propagated to rec_preprocessed and so they are save as properties in the saved folder.
So you should able to load it and play with traces in microvolt.
```python
rec_loaded = si.load_extractor(lfp_folder)
# This is micro volts :
traces = rec_loaded.get_traces..., return_scaled=True)
# This is raw (very often int16) :
traces = rec_loaded.get_traces..., return_scaled=False)

```

JonathanAMichaels, 2023-10-03:
Thanks!

I needed to prepare this data to export to someone else, so I used the scale method to apply the gain myself and saved as int16.