SpikeInterface/spikeinterface #2065: How to remove spikes with low amplitude

rat-h, 2023-10-03:
Hi there,
After spike sorting, we got plenty of units and some units have a mixture of low ant high amplitude spikes (at least phy shows that). We want to remove spikes with amplitude less than a threshold in all units. Is there a simple way to do so?

samuelgarcia, 2023-10-03:
Hi,
here in step 7 curation, you should have the main guide.
https://github.com/SpikeInterface/spiketutorials/blob/f309f2c03fbecb04a5c4eeff540a874c6101d8d9/Official_Tutorial_SI_0.96_Oct22/SpikeInterface_Tutorial.ipynb

The idea is to compute metrics and then filter the dataframe with some threshold on some metrics.

@alejoe91 : I am not sure we have this clearly written i the doc.

rat-h, 2023-10-03:
@samuelgarcia correct me if I'm wrong, does curation work on the level of units, not spikes? What I need to remove all spikes with low amplitudes, not units.

samuelgarcia, 2023-10-03:
Oups sorry. I read it quickly.
Then you need to play with the compute amplitudes functions
Then use it as a a mask on spike vector.

```python
amplitudes = si.compute_spike_amplitudes which give amplitudes of all spikes (per segments)
spikes = sorting.to_spike_vector()
keep_spikes = spikes[amplitudes>my_thresh]

```

alejoe91, 2023-10-04:
Hi @rat-h 

If your threshold is global, what Sam suggested works. You can do the following:
```
import spikeinterface as si
import spikeinterface.postprocessing as spost

...
# assume we is a WaveformExtractor

amplitudes = spost.compute_spike_amplitudes(we)
# this returns a list with as many entries as number of segments:
amplitudes0 = amplitudes[0]
# make the mask based on threshold (note that this will be uV if you have scaling)
my_thresh = -50
mask = amplitudes0 < my_thresh

# mask the spikes vector
spikes = we.sorting.to_spike_vector()
spikes_large_amp = spikes[mask]
# make a Sorting object
sorting_large_amp = si.NumpySorting(spikes=spikes_large_amp, 
                                    sampling_frequency=we.sampling_frequency, 
                                    unit_ids=we.unit_ids)

# now we have to re-compute waveforms (we don't have tools to slice a waveform extractor for specific spikes)
we_large_amp = si.extract_waveforms(we.recording, sorting_large_amp, folder="waveforms_large_amp")

# then you can export to phy
spost.export_to_phy(we_large_amp, output_folder="phy_large_amp")
```

alejoe91, 2023-10-04:
We could also make an automatic function to do this in the `spikeinterface.curation` module

rat-h, 2023-10-04:
I got an error
```python
python
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import spikeinterface.full as si
>>> import spikeinterface.postprocessing as spost
>>> 
>>> rc = si.load_extractor("preprocessed")
>>> sr = si.load_extractor("sorting-saved")
>>> we = si.WaveformExtractor.load_from_folder('waveforms')
>>> amplitudes = si.compute_spike_amplitudes(we,peak_sign='both',outputs='concatenated')
Setting 'return_scaled' to False
extract amplitudes: 100%|##################################################################################################################################################################################################################| 4499/4499 [22:39<00:00,  3.31it/s]
>>> len(amplitudes)
1
>>> amplitudes0 = amplitudes[0]
>>> type(amplitudes0)
<class 'numpy.ndarray'>
>>> amplitudes0.shape
(16019740,)
>>> mask = amplitudes0 < -7.
>>> mask
array([False, False, False, ..., False, False, False])
>>> spikes = we.sorting.to_spike_vector()
>>> spikes_large_amp = spikes[mask]
>>> spikes_large_amp.shape
(868450,)
>>> sorting_large_amp = si.NumpySorting(spikes_large_amp, we.sampling_frequency, we.unit_ids)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: NumpySorting.__init__() takes from 2 to 3 positional arguments but 4 were given
>>> help(si.NumpySorting)
"""
Help on class NumpySorting in module spikeinterface.core.numpyextractors:

class NumpySorting(spikeinterface.core.basesorting.BaseSorting)
 |  NumpySorting(sampling_frequency, unit_ids=[])
 |  
 |  Method resolution order:
 |      NumpySorting
 |      spikeinterface.core.basesorting.BaseSorting
 |      spikeinterface.core.base.BaseExtractor
 |      builtins.object
 |  
 |  Methods defined here:
 |  
 |  __init__(self, sampling_frequency, unit_ids=[])
 |      Initialize self.  See help(type(self)) for accurate signature.
 |  

"""
>>> si.__version__
'0.98.2'
```

I think I have the latest version from `pip`:
```bash
$ pip install -U 'spikeinterface[full]'
Requirement already satisfied: spikeinterface[full] in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (0.98.2)
Requirement already satisfied: numpy in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (1.23.5)
Requirement already satisfied: neo>=0.12.0 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (0.12.0)
Requirement already satisfied: joblib in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (1.1.1)
Requirement already satisfied: threadpoolctl in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (3.1.0)
Requirement already satisfied: tqdm in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (4.64.1)
Requirement already satisfied: probeinterface>=0.2.17 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (0.2.17)
Requirement already satisfied: zarr in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (2.13.3)
Requirement already satisfied: h5py in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (3.7.0)
Requirement already satisfied: pandas in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (1.5.2)
Requirement already satisfied: xarray in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (2023.3.0)
[... 2 lines cut ...]
Requirement already satisfied: networkx in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (2.8.8)
Requirement already satisfied: distinctipy in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (1.2.2)
Requirement already satisfied: matplotlib in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (3.6.2)
Requirement already satisfied: numba in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (0.56.4)
Requirement already satisfied: cuda-python in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from spikeinterface[full]) (12.1.0)
Requirement already satisfied: packaging in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from neo>=0.12.0->spikeinterface[full]) (22.0)
Requirement already satisfied: quantities>=0.14.1 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from neo>=0.12.0->spikeinterface[full]) (0.14.1)
Requirement already satisfied: cython in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from cuda-python->spikeinterface[full]) (0.29.32)
Requirement already satisfied: contourpy>=1.0.1 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from matplotlib->spikeinterface[full]) (1.0.6)
Requirement already satisfied: cycler>=0.10 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from matplotlib->spikeinterface[full]) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from matplotlib->spikeinterface[full]) (4.38.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from matplotlib->spikeinterface[full]) (1.4.4)
[... cut ...]
Requirement already satisfied: fasteners in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from zarr->spikeinterface[full]) (0.18)
Requirement already satisfied: numcodecs>=0.10.0 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from zarr->spikeinterface[full]) (0.11.0)
Requirement already satisfied: entrypoints in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from numcodecs>=0.10.0->zarr->spikeinterface[full]) (0.4)
Requirement already satisfied: six>=1.5 in /home/rth/.local/apps/spikes/lib/python3.10/site-packages (from python-dateutil>=2.7->matplotlib->spikeinterface[full]) (1.16.0)

```

alejoe91, 2023-10-04:
Right, I think that we made the constructor from the spikes array recently so you would need to install form source for the time being. We'll make a new release in about a month, so then you can switch back to the released version!

alejoe91, 2023-10-04:
Should be ;)

rat-h, 2023-10-04:
Almost there
```
AttributeError: module 'spikeinterface.postprocessing' has no attribute 'export_to_phy'
```

rat-h, 2023-10-05:
I ended with this code. It still running, and I hope to get everything tomorrow.
```python
import spikeinterface.full as si
# import spikeinterface.postprocessing as spost

rc = si.load_extractor("preprocessed")
sr = si.load_extractor("sorting-saved")
we = si.WaveformExtractor.load_from_folder('waveforms')

amplitudes = si.compute_spike_amplitudes(we,peak_sign='both',outputs='concatenated')

amplitudes0 = amplitudes[0]

print(f"Original shape:{amplitudes0.shape}")

# make the mask based on threshold (note that this will be uV if you have scaling)
my_thresh = 7.
mask = abs(amplitudes0) >= my_thresh

# mask the spikes vector
spikes = we.sorting.to_spike_vector()
spikes_large_amp = spikes[mask]
# make a Sorting object

print(f"Large amplitude shape:{spikes_large_amp.shape}")

sorting_large_amp = si.NumpySorting(spikes_large_amp, we.sampling_frequency, we.unit_ids)

sorting_large_amp = sorting_large_amp.remove_empty_units()

# now we have to re-compute waveforms (we don't have tools to slice a waveform extractor for specific spikes)
we_large_amp = si.extract_waveforms(we.recording, sorting_large_amp, folder="waveforms_large_amp")

# we_large_amp = si.load_waveforms("waveforms_large_amp")

si.compute_spike_amplitudes(we_large_amp)
si.compute_principal_components(we_large_amp, n_components=3, mode='by_channel_global')
        
# then you can export to phy
si.export_to_phy(we_large_amp, output_folder="phy-large_amp")
```

**THANK YOU FOR HELP**!

closed 2023-10-05 by rat-h

NathanielH-snek, 2024-08-15:
> Hi @rat-h
>
> If your threshold is global, what Sam suggested works. You can do the following:
>
> ```
> import spikeinterface as si
> import spikeinterface.postprocessing as spost
>
> ...
> # assume we is a [...]

Has the API for any of the functions in this changed in the transition to using Sorting Analyzers? Running the numpy sorting returns a type error now (TypeError: only integer scalar arrays can be converted to a scalar index).

M-Hisham, 2025-05-23:
Hello, I am interested in the same problem, I am looking for a way to clean individual unit' spikes. for example removing low-amplitude spikes or spikes of certain amplitudes window. here is two examples from two units. these are two good units but runied with few outlier spikes. 

[image]

[image].

I am curating using the spike-interface GUI, and using spike-analyzer objects. is there a way to clean individual units without recomputing the whole spike-Analyzer. I used to use UltraMegaSort previously, and the impeded outlier tool were able to remove outlier spikes from individual units.

Thanks a lot for looking at this

M-Hisham, 2025-05-23:
> We could also make an automatic function to do this in the `spikeinterface.curation` module

Hello @alejoe91, is this function available?

Thank you

alejoe91, 2025-05-23:
Hi guys, not yet but we are actively working on the SpikeInterface GUI and adding splitting functionalities. There is a way to do that also from spikeinterface, but it would require recomputing the analyzer. Let me get back to you in the next few days with a working solution!