SpikeInterface/spikeinterface #2066: Having trouble importing neuroexplorer file using spikeinterface extractor

kthuang20, 2023-10-03:
Hello,
I am having difficulty importing my NeuroExplorer file using the SpikeInterface extractor module.
Here is a screenshot of the steps that I took and error message I receive when I try to use the read_neuroexplorer function:

[image]

alejoe91, 2023-10-04:
Hi @kthuang20 

The neuroexplorer reader was recently added (#1965) and so it's available only if you install from source (not from pip).

You can check this installation instructions: https://spikeinterface.readthedocs.io/en/latest/installation.html#from-source

alejoe91, 2023-10-19:
@kthuang20 were you able to load your neuroexplorer data?

kthuang20, 2023-10-19:
Thank you, it is now able to recognize the read_neuroexplorer function. However, when I try to load the file, I am getting another error
[image]

alejoe91, 2023-10-19:
Can you share the file with us?

kthuang20, 2023-10-19:
The file can be found here: https://drive.google.com/file/d/1WuUNCWhDoKRwPrI7UEMs4hd8zYfMlJBj/view?usp=sharing

alejoe91, 2023-10-19:
Thanks @kthuang20 

I think that the problem is that you don't have the raw data. The `read_neuroexplorer` reads the raw traces, but your file only contains spikes (which we don't have a reader for).

h-mayorquin, 2023-11-12:
I guess this the spike extractor is not available on neo here?