SpikeInterface/spikeinterface #2083: NumpySorting from_dict, from_unit_dict

magland, 2023-10-06:
I notice that `NumpySorting.from_dict` has changed to `NumpySorting.from_unit_dict`, so there is an inconsistency between the released version and the dev version. I understand the need for the change. How would you suggest that I manage that in my code that uses NumpySorting?

https://github.com/SpikeInterface/spikeinterface/blob/4cc4777c362a826833706bcb57bdc24012aa4240/src/spikeinterface/core/numpyextractors.py#L212

magland, 2023-10-06:
For now I am doing this

```
    try:
        # different versions of spikeinterface
        sorting = si.NumpySorting.from_dict(
            [units_dict], sampling_frequency=sampling_frequency
        )
    except:
        sorting = si.NumpySorting.from_unit_dict(
            units_dict, sampling_frequency=sampling_frequency
        )
```

samuelgarcia, 2023-10-06:
Hi Jeremy.
Sorry for that.
Yes I discover that the NumpySorting.from_dict over overwirtting the Base.from_dict().
It was an old and big mistake!
It was fixed during the summer.

I do not have better fix than you have otherwise version checking or maybe check that the method `from_dict` is the same as Base (then new behavior) or different (new old behavior).

closed 2023-10-06 by magland