SpikeInterface/spikeinterface #2092: Tetrode Probe Group Ignores Channel Indices

NathanielH-snek, 2023-10-10:
I'm working on some old Open Ephys data (2019) that appears to not have any sort of probe information attached. I found another issue that mentioned this and the answer is that you need to manually assign this. An additional complexity added to this project is that it uses a custom probe layout. As a result I have been attempting to create a probe using probe interface manually and use that (although I'm still unclear if position will matter). I have a channel list that I keep attempting to assign to the probe_group, but it seems to ignore it based on the outputted dataframe. I may be simply making a simple mistake but I can't seem to figure out why it's behaving this way. I'm also curious as to if I'm approaching this analysis pipeline the proper way since a lot of the tutorials seem to be somewhat contingent on either having probe information or using standardized probe layouts. Thanks! I've attached my somewhat hacky scripted as well. 
[TestExample.zip](https://github.com/SpikeInterface/spikeinterface/files/12861492/TestExample.zip)

zm711, 2023-10-12:
@NathanielH-snek,

You can just post the script here  with '```' (triple backtick) so it will be easy to read. Could you do that? I'll also attach the documentation to `probe interface` which goes through how to make a probe yourself.

https://probeinterface.readthedocs.io/en/main/examples/ex_01_generate_probe_from_sratch.html

And this page specifically talks about tetrodes:
https://probeinterface.readthedocs.io/en/main/examples/ex_07_probe_generator.html

NathanielH-snek, 2023-10-13:
Yep, I attached the script originally since I assumed it would be easier than pasting each individual cell. 
Here is what I'm using for my tetrode probe generation. 
'''
from probeinterface import Probe, ProbeGroup, generate_tetrode
from probeinterface.plotting import plot_probe_group
import numpy as np
import pandas as pd
probe = ProbeGroup()
for i in range(16):
    tetrode = generate_tetrode()
    tetrode.move([i * 50, 0])
    probe.add_probe(tetrode)
probe.set_global_device_channel_indices(channel_ids)
df = probe.to_dataframe()
pd.set_option('display.max_columns', None)
pd.set_option('display.max_rows', None)
print(df)

plot_probe_group(probe, with_channel_index=True, same_axes=True)
'''
This results in the right tetrode amount but they are labeled incorrectly as still '1.2,3,4'. Also I'm still a bit confused in your documentation in the use of probeinterface. Should I be attaching the probe I make to a recroding (although since its a probegroup I'm unsure if that's doable). I managed to get MS5 to run by assigning arbitrary positions to channels, but I'm assuming this is not the correct way to do this. Here is my analysis script right now. The recording is 67 channels with the last 3 channels auxillary and as such they are removed from the recording. 

'''
import spikeinterface.full as si
from pathlib import Path
global_job_kwargs = dict(n_jobs=4, chunk_duration="1s")
si.set_global_job_kwargs(**global_job_kwargs)
import spikeinterface.extractors as se
import spikeinterface.preprocessing as pp

dataLocal = se.read_openephys("ExperiementData" , stream_id = '1' , block_index = 1 , stream_name = 'Common_Avg_Ref-113_100.0')
recording = pp.bandpass_filter(dataLocal, freq_min=300, freq_max=6000)
import numpy as np
#recording.set_probe(dataLocal)
#channel_ids = np.array(channel_ids)
numChannels = 67
x = [0] * numChannels
y = list(range(1, numChannels+1))
locations = list(zip(x,y))
locations = np.array(locations)
locations.shape
origChanIDs = recording.get_channel_ids()
recording.set_channel_locations(locations,origChanIDs)
badChannels = ['AUX1','AUX2','AUX3']
recording = recording.remove_channels(badChannels)
badChannelIDs = pp.detect_bad_channels(recording)
recording = recording.remove_channels(badChannelIDs)
print(recording)
#recording = pp.phase_shift(recording)
recordingCom = pp.common_reference(recording, operator='median')
recordingCor = si.correct_motion(recordingCom,preset='kilosort_like')
print(recordingCor)
from spikeinterface import sorters
sortingMS5 = sorters.run_sorter(sorter_name="mountainsort5", recording= recording)
'''

alejoe91, 2023-10-16:
Hi,

So the problem is in how you use the `set_global_device_channel_indices()`.

Channel indices are NOT channel IDs. Channel IDs are a "name" of the channel, e.g., given by the acquisition device, while channel indices are an integer indicating to what contact each recorded trace corresponds to.
If the channel ids are a string containing the channel mapping, you should do:
```
device_channel_indices = [int(ch) for ch in channel_ids]
probe.set_global_device_channel_indices(channel_ids)
```
(I also recommend to rename `probe` to `probegroup`)

The second error is that you are generating the probe group, but not setting it to the recording.
You should use:
```
recording_w_probe = recording.set_probegroup(probegroup, group_mode="by_probe")
# or in place
recording = recording.set_probegroup(probegroup, group_mode="by_probe", in_place=True)
```

NathanielH-snek, 2023-10-16:
Hello, sorry I left out a part of my script in which the channel_id's is actually an integer list. The channel mapping is not in direct numerical order. I did the following for pulling the correct numbers from the channel id list.  I just happened to use the same variable name. The issue was when I graphed it wouldn't graph the correct channel numbers. It's probably just bad form that I used the same variable name temporarily. 
```
import re
channel_ids = recording.get_channel_ids()
num_channels = recording.get_num_channels()
channel_ids = channel_ids[0:64].tolist()
for i in range(len(channel_ids)):
    word = str(channel_ids[i])
    word = re.findall(r'\d+' , word)
    channel_ids[i] = int(word[0])
    
channel_ids
```
This results in the following list: 
[9, 10, 15, 16, 11, 12, 13, 14, 5, 6, 7, 8, 1, 2, 3, 4, 32, 31, 30, 29, 28, 27, 26, 25, 22, 21, 20, 19, 18, 17, 23, 24, 51, 52, 53, 54, 57, 58, 59, 60, 61, 62, 63, 64, 33, 34, 35, 36, 37, 38, 39, 40, 43, 44, 45, 46, 41, 42, 47, 48, 49, 50, 55, 56]
this is correct.
The reason I had not been setting it to the recording is because when I graph it it looks like this:
[image]
Looking at the dataframe that is output the channel indices I am setting are being ignored. This means either I am setting them wrong or plotting incorrectly. I'm assuming I should be assigning the contact ids instead of the channel indices? If so, how? Also will the numerical distance between tetrodes affect sorting? I can't find that information in the documentation.

alejoe91, 2023-10-16:
Ah I see! The plot only shows 1-2-3-4 because it doesn't show device indices, but contact IDs. These are all the same because each probe is a duplicate otlf the others. So that's ok. Then your problem should be solved if you run the `set_probegroup` with the `group_mode="by_probe"`. 

Let me know!

alejoe91, 2023-10-16:
The numerical distance will affect the sorting, but I recommend sorting each tetrode separately with a sorter that doesn't assume high-density probes (e.g. not Kilosort4, mountainsort could be a good option!).

To sort by group, you can check this documentation, which also includes how to handle tetrodes (I just realized!): 
https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html?highlight=run_sorter_by_property#spike-sorting-by-group

NathanielH-snek, 2023-10-16:
Thank you for your help and bearing with me! One last issue it seems. Everything seems to have worked as expected, except now I'm producing a value error despite having the same length of indices and channels. 
"ValueError: The given Probe have 'device_channel_indices' that do not match channel count 
64 vs 64 "

alejoe91, 2023-10-16:
Can you paste the full error?

alejoe91, 2023-10-16:
Ahh I see! Device channel indices should be 0-based! So you should just be able to fix it with:
`channel_ids=channel_ids-1`

I strongly recommend refactoring your variable naming and use `device_channel_indices` instead of `channel_ids` for clarity!

NathanielH-snek, 2023-10-16:
Thank you that seemed to have solved it. Dumb mistake.  I also did change the variable name after it was confusing earlier. Thanks for your help again!