SpikeInterface/spikeinterface #2095: 'spikeinterface.full' has no attribute 'plot_traces'

NJamann, 2023-10-12:
Hi,

I am running SpikeInterface Version 0.98.2 in a Jupyter Notebook on one of my Neuropixels recordings, according to the newest tutorial (Oct 23).
I am importing spikeinterface.full as si.
However, when I try to run plot.traces I get the following error:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[43], line 3
      1 get_ipython().run_line_magic('matplotlib', 'widget')
----> 3 u= si.plot_traces(raw_rec, time_range=[0, 10], backend="ipywidgets", mode='map')

AttributeError: module 'spikeinterface.full' has no attribute 'plot_traces'

Can you help me out?
Thank you!
Nora

zm711, 2023-10-12:
Hi Nora @NJamann ,

This was a change that just occurred right after the 0.98.2 release (in PR #1859) so the change in function name hasn't been incorporated yet into a release. 
The documentation for your version is [here](https://spikeinterface.readthedocs.io/en/0.98.2/ ). 
You can also install from source with these [instructions](https://spikeinterface.readthedocs.io/en/latest/installation.html) in which case the `plot_traces` will work as expected.

Previously `plot_traces` was `plot_timeseries`, but there have been a couple renamings to help make things clearer in the future (it's easier to remember to plot the traces if the function is called `plot_traces`). 

The specifics into the function you would need with the older version is [here](https://spikeinterface.readthedocs.io/en/0.98.2/modules/widgets.html?highlight=plot_timeseries).

Does all of that make sense?

NJamann, 2023-10-12:
Ok thanks!
I had tried plot_timeseries first and didn't get an error, but it also does not output a plot either. But that might be a different issue alltogether.
I will try to install from source to use plot_traces.

NJamann, 2023-10-13:
Hi @zm711,
I got plot_traces to run but not to work so Im at the same point as before with plot_timeseries.
For some reason I get an object as an output but not the actual plot:

[image]

zm711, 2023-10-13:
@NJamann  I think I hav the solution.

Instead of 
```python
%matplotlib widget
```
put
```python
%matplotlib inline
```

:)

NJamann, 2023-10-13:
Hi Zach! Great, for the matplotlib backend, that works! Thanks a lot

zm711, 2023-10-13:
Great Nora! (@NJamann )

So if you try 

```
%matplotlib inline
%matplotlib widget
```

Does ipywidgets work? I'm just trying to figure out if I need to see if there is a bug.

NJamann, 2023-10-13:
No that still doesnt do anything
[image]

zm711, 2023-10-13:
And when you installed from source you made sure to do

```
pip install .[widgets]
```

NJamann, 2023-10-13:
I thought I did but I just redid
pip install spikeinterface[full,widgets]
again.
The only thing that wasnt installed yet was:
[image]

However, I still dont get a plot, also after restarting the virtual environment.

zm711, 2023-10-13:
Okay I’m going to tag @samuelgarcia, since he’s the expert for plotting. He’ll look over things when he gets a moment. (I personally just use the matplotlib backend).

NJamann, 2023-10-13:
Thanks anyways for your help, Im already happy to visualize my data like this! :)

closed 2023-11-06 by alejoe91