SpikeInterface/spikeinterface #2099: Question about SI "group" property for multi-shank probe

JoeZiminski, 2023-10-13:
Hi All,

I have an OpenEphys recording (NP2.0, 4-shank) loaded with `read_openephys()` that loads the data and probe in a way that seems correct (plot of probe below). Althought, in the probe view only 2 of the shanks are plot, but indeed the activate channel position overlaid over the (missing) other two shanks looks correct.

However, the `recording.get_channel_groups()` is set to all zero. I had assumed behind the scenes this would set the channels to be grouped by shank (if I recall correctly this is the case with SpikeGLX output data). I am happy to set the groups manually but wanted to check if this should happen automagically / I am doing something wrong.

Cheers!
Joe

[image]

alejoe91, 2023-10-13:
Hi @JoeZiminski 

Indeed it should be automatic. I'll take a look. Can you sen me the settings file?

Cheers
Alessio

JoeZiminski, 2023-10-13:
Thanks @alejoe91!

Sorry I can't seem upload the `.xml` but please see the contents below. Also just for reference the load I am using to load / plot  is:

```
from spikeinterface.extractors import read_openephys

data_path = Path("...")

recording = read_openephys(data_path)

probe = recording.get_probe()
plot_probe(probe)
plt.show()

recording.get_channel_groups()

array([0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
       0, 0, 0, 0, 0, 0, 0, 0, 0, 0], dtype=int64)
(Pdb) 
```

```
<?xml version="1.0" encoding="UTF-8"?>

<SETTINGS>
  <INFO>
    <VERSION>0.6.5</VERSION>
    <PLUGIN_API_VERSION>8</PLUGIN_API_VERSION>
    <DATE>3 Oct 2023 18:57:09</DATE>
    <OS>Windows 10</OS>
    <MACHINE name="DESKTOP-V8K5R75" cpu_model="13th Gen Intel(R) Core(TM) i9-13900K"
             cpu_num_cores="32"/>
  </INFO>
  <SIGNALCHAIN>
    <PROCESSOR name="Neuropix-PXI" insertionPoint="0" pluginName="Neuropix-PXI"
               type="4" index="0" libraryName="Neuropix-PXI" libraryVersion="0.5.0"
               processorType="2" nodeId="100">
      <GLOBAL_PARAMETERS/>
      <STREAM name="ProbeA" description="description" sample_rate="30000.0"
              channel_count="384" device_name="Neuropixels 2.0 - Multishank">
        <PARAMETERS/>
      </STREAM>
      <CUSTOM_PARAMETERS/>
      <EDITOR isCollapsed="0" isDrawerOpen="0" displayName="Neuropix-PXI" activeStream="0"
              Type="Visualizer">
        <TAB Active="1" Index="3"/>
        <WINDOW Active="0"/>
        <NEUROPIXELS_EDITOR MainSyncSlot="-1" SendSyncAsContinuous="0" SyncDirection="0"
                            SyncFreq="0">
          <BASESTATION Directory="" Slot="2" NamingScheme="0" port1dock1="slot2-port1-1"
                       port1dock2="slot2-port1-2" port2dock1="slot2-port2-1" port2dock2="slot2-port2-2"
                       port3dock1="slot2-port3-1" port3dock2="slot2-port3-2" port4dock1="slot2-port4-1"
                       port4dock2="slot2-port4-2"/>
[... 269 lines cut ...]
  <DATAVIEWPORT selectedTab="4"/>
  <AUDIO sampleRate="48000.0" bufferSize="480" deviceType="Windows Audio">
    <DEVICESETUP deviceType="Windows Audio" audioOutputDeviceName="Dell S2716DG (NVIDIA High Definition Audio)"
                 audioInputDeviceName="" audioDeviceRate="48000.0"/>
  </AUDIO>
  <CONTROLPANEL isOpen="1" recordPath="C:\Users\Sara\Documents\Open Ephys" recordEngine="BINARY"
                clockMode="0"/>
  <AUDIOEDITOR isMuted="0" volume="50.0" noiseGate="10.0"/>
  <FILENAMECONFIG>
    <PREPEND state="0" value=""/>
    <MAIN state="1" value="2023-10-03_18-57-09"/>
    <APPEND state="0" value=""/>
  </FILENAMECONFIG>
  <PROCESSORLIST>
    <COLOR ID="801" R="59" G="59" B="59"/>
    <COLOR ID="804" R="241" G="90" B="41"/>
    <COLOR ID="802" R="0" G="174" B="239"/>
    <COLOR ID="803" R="0" G="166" B="81"/>
    <COLOR ID="805" R="147" G="149" B="152"/>
    <COLOR ID="806" R="255" G="0" B="0"/>
    <COLOR ID="807" R="0" G="0" B="0"/>
  </PROCESSORLIST>
  <UICOMPONENT isProcessorListOpen="1" isEditorViewportOpen="1"/>
</SETTINGS>
```

PR #2102 (2023-10-16): Fix grouping of OpenEphys NPIX
Fixes #2099 

@JoeZiminski can you test this out?
closed 2023-10-16 by samuelgarcia