SpikeInterface/spikeinterface #2101: Curation Tutorial Broken

zm711, 2023-10-13:
Curation tutorial was broken by #2061.

I didn't have time to look into it yet, but I wanted this on my radar to remind me.

alejoe91, 2023-10-13:
Interesting. I cannot reproduce the issue locally by running the `examples/modules_gallery/quality_metrics/plot_4_curation.py`.

@zm711 curious if it fails locally for you!

zm711, 2023-10-13:
It failed locally for me too!

```
    self._main_ids.dtype.kind in "uiSU"
AssertionError: Main IDs can only be integers (signed/unsigned) or strings, not object
```

zm711, 2023-10-13:
@alejoe91 so the issue is

```
>>> keep_mask.index.values
Index(['#0', '#1', '#2', '#3', '#4', '#5', '#6', '#7', '#8', '#9'], dtype='object')
```

It is being stored as an np object array because np does haven't string arrays. Do you want to fix this as a bug to allow numpy object arrays initialize ids or do you want me to fix the tutorial and pop out the strings into a list which I just tested and worked fine.

```
test =[idx for idx in keep_mask[keep_mask].index.values]
>>> curated_sorting = sorting.select_units(test)
```

samuelgarcia, 2023-10-14:
Why did we removed objetc in main_ids ?

alejoe91, 2023-10-14:
Because we always assume main IDs should be int or str. In my opinion, it is time to always use strings

samuelgarcia, 2023-10-14:
No lets keep int. It is super convinient in many places.

alejoe91, 2023-10-14:
Where? I think it's just annoying

PR #2103 (2023-10-16): Fix the `curation` tutorial
Fixes #2101.

I know there is a greater debate about what main unit ids should be allowed to be. But at least for the tutorial the path of least resistance is just to pop them out as strings (since everyone agrees strings should be allowed) and use that in the tutorial.
closed 2023-10-16 by samuelgarcia