SpikeInterface/spikeinterface #2107: How to sort only a segment of an Open Ephys recording

nmtimme, 2023-10-16:
Hello

I would appreciate any advice on how to sort only a segment of an open ephys recording. A portion of my recordings have a discontinuity in the timestamps which produces an error like #1194 . I've found that this error is caused by a short ~17 ms discontinuity ~34 ms before the end of the recording. The recording time stamps also do not start at 0, but I have other recordings that also do not start at zero and spike sort successfully (so long as they do not possess the discontinuity at the end).

I attempted to use the ignore_timestamps_errors option described in https://github.com/NeuralEnsemble/python-neo/issues/1210 and https://github.com/NeuralEnsemble/python-neo/pull/1213, but this did not work. I received a syntax error, but perhaps I am not using the most up to date version of the code. I'd rather just cut off the end of the recording in case there are other discontinuities in the middle of the recording that I should be aware of.

Is there a way in spike interface to only extract and sort a certain part of the recording, such as from time step 1 to time step end - 30000? I believe this would occur at:

raw_rec = si.read_openephys(openephys_folder,stream_name='Signals CH')

I investigated the block_index argument, but I'm not sure of the correct syntax to use. I would appreciate any advice that people could provide. I've looked around the issues and documentation and I couldn't find anything that directly addressed this point, but if I missed it, I apologize and please direct me to the correct resource. Thank you!

~Nick

zm711, 2023-10-16:
@nmtimme,

I'm trying to understand at which point the error is occurring-- at extracting from the file like in #1194 or after when trying to sort. 

If the extraction is successful then you can simply slice a recording like is explained [here](https://spikeinterface.readthedocs.io/en/latest/modules/core.html) (and I've copied below)
```python
sub_recording = recording.frame_slice(start_frame=int(fs * 60 * 5), end_frame=int(fs * 60 * 12))
```

If the error is in the extraction it would be beneficial to know your version of spikeinterface and Neo to see if this  a version issue. Could you post the exact error you're getting along with the version number of your packages? Thanks.

nmtimme, 2023-10-16:
Hi @zm711 !

Thank you for responding so quickly! 

The error is occurring at extraction. We have spike interface version 0.98.2 and neo version 0.12.0. This is the code we are running (Note, there is a lot of stuff in here related to loading a job list. The error occurs on the last line of what I've copied here.):

```
# Import necessary pieces of software
import spikeinterface.full as si
import probeinterface as pi
import neo
from probeinterface import Probe
from probeinterface.plotting import plot_probe
print(si.__version__)
print(neo.__version__)

import numpy as np
import matplotlib.pyplot as plt
import csv as csv
import pathlib
# from pathlib import Path

# Load the data set list
f = open('/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/cambridgeRecordings.txt', 'r')
csvreader = csv.reader(f, delimiter=',')
dataSetList =[]
for row in csvreader:
    dataSetList.append(row)
f.close()

# Iterate through each data set
for dataSet in dataSetList:
    
    # Set the open ephys folder path, data set name, channel map, and the location of the spike sorted data for this data set
    openephys_folder = dataSet[0]
    channelMapFile = dataSet[1]
    path = pathlib.PurePath(openephys_folder)
    dataSetName = path.name
    spikeSortedDataFolder = '/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/' + dataSetName
    
    print('Processing Data Set:',dataSetName)
    print('Channel Map Name:',channelMapFile)

    # Load the channel map
    f = open('/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeChannelMapFiles/' + channelMapFile + '.txt', 'r')
    reader = f.readlines()
    channelMapFileData = []
    for row in reader:
        channelMapFileData.append(row)
    f.close()
    probeName = channelMapFileData[0].strip()
    nChannels = len(channelMapFileData) - 1
    print('nChannels:',nChannels)
    device_channel_indices = np.arange(nChannels)
    for i in np.arange(1,nChannels + 1):
        device_channel_indices[i - 1] = int(channelMapFileData[i]) - 1
    print('Probe Name:',probeName)
    print('device_channel_indices:',device_channel_indices)

    # Read the ephys data (stream names are 'Signals AUX', 'Signals CH', 'Signals ADC')
    raw_rec = si.read_openephys(openephys_folder,stream_name='Signals CH')
```

This is the error we get:

```
Traceback (most recent call last):
  File "/geode2/home/u050/lapishla/BigRed200/spikeInterfaceVer1/spikeSortCambridgeData.py", line 63, in <module>
    raw_rec = si.read_openephys(openephys_folder,stream_name='Signals CH')
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py", line 280, in read_openephys
    recording = OpenEphysLegacyRecordingExtractor(folder_path, **kwargs)
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py", line 56, in __init__
    NeoBaseRecordingExtractor.__init__(
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 185, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 25, in __init__
    self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 64, in get_neo_io_reader
    neo_reader.parse_header()
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/neo/rawio/baserawio.py", line 179, in parse_header
    self._parse_header()
  File "/N/u/lapishla/BigRed200/.conda/envs/si_env/lib/python3.9/site-packages/neo/rawio/openephysrawio.py", line 118, in _parse_header
    assert np.all(diff == RECORD_SIZE), \
AssertionError: Not continuous timestamps for 100_AUX1.continuous. Maybe because recording was paused/stopped.
```

We've found (with the help of @jsiegle ) that the last record in the open ephys recordings is all zeros and possesses a discontinuity in the timestamps (https://github.com/open-ephys/open-ephys-python-tools/issues/25). We're hoping there is a way in spike interface to extract everything except this last part of the recording when we are loading in the open ephys data. We'd appreciate any advice you can provide. Thank you!

~Nick

zm711, 2023-10-16:
@nmtimme 
Hey Nick,

I see. So the potential fix for this issue [neo1213](https://github.com/NeuralEnsemble/python-neo/pull/1213) has not been included in a Neo release yet (it will be included in Neo 0.13.0 whenever it is released). So your best solution is to install Neo from source (so 0.13.0.dev). 

To do this you have two options do an install with pip calling to the git repository like this (gets dev version without a local editable copy hanging around):
```
pip install git+https://github.com/NeuralEnsemble/python-neo.git
```

Or you can clone the source repository like this (with and editable copy in case you want to pull changes if more patches are made): 

```
git clone https://github.com/NeuralEnsemble/python-neo.git
cd python-neo
pip install -e .
cd ..
```

Either way make sure you are in your working `si_env` so you don't install it in the wrong place. 

Then you can run the extraction (with the ignore_missing_timestamps), then perform the splicing step (like I said above) and then you can follow the advice of https://github.com/NeuralEnsemble/python-neo/issues/1210 and save the slice as a binary recording for future reuse without having to worry about the small signal drop outs.

Does that all make sense?

Zach

nmtimme, 2023-10-18:
Hi Zach (@zm11)!

Thank you for those suggestions. We may try that in the future, but we were able to get a solution up and running where we just trim the open ephys data (https://github.com/open-ephys/open-ephys-python-tools/issues/25). Thank you again!

~Nick

nmtimme, 2023-10-20:
Hi Zach (@zm711)!

I'm reopening this issue because we've pivoted back to the method you suggested with updating neo and we're encountering a problem with sliced data in the kilosort3 container. This data set has a weird discontinuity at the end of the recording (in the last 34 ms). We also have some recordings with weird discontinuities in the first second. Therefore, we tried to cut off the first and last two seconds.

Here is the code we are running (we are running this on a cluster, so there's some code related to that functionality):

```
import spikeinterface.full as si
import probeinterface as pi
import neo
from probeinterface import Probe
from probeinterface.plotting import plot_probe
print('si version:',si.__version__)
print('neo version:',neo.__version__)

import numpy as np
import matplotlib.pyplot as plt
import csv as csv
import pathlib
import sys
import shutil
import os
# from pathlib import Path

# Get the task ID
taskID = int(sys.argv[1])
# taskID = 1
print('Task ID:',taskID)

# Load the data set list
f = open('/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/cambridgeRecordings.txt', 'r')
csvreader = csv.reader(f, delimiter=',')
dataSetList =[]
for row in csvreader:
    dataSetList.append(row)
f.close()

# Set the open ephys folder path, data set name, channel map, and the location of the spike sorted data for this data set
openephys_folder = dataSetList[taskID - 1][0]
channelMapFile = dataSetList[taskID - 1][1]
path = pathlib.PurePath(openephys_folder)
dataSetName = path.name
spikeSortedDataFolder = '/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/' + dataSetName
spikeSortTempFolder = '/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/' + dataSetName + '/temp'

print('Processing Data Set:',dataSetName)
print('Channel Map Name:',channelMapFile)

# Load the channel map
f = open('/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeChannelMapFiles/' + channelMapFile + '.txt', 'r')
reader = f.readlines()
channelMapFileData = []
for row in reader:
    channelMapFileData.append(row)
f.close()
probeName = channelMapFileData[0].strip()
nChannels = len(channelMapFileData) - 1
print('nChannels:',nChannels)
device_channel_indices = np.arange(nChannels)
for i in np.arange(1,nChannels + 1):
    device_channel_indices[i - 1] = int(channelMapFileData[i]) - 1
print('Probe Name:',probeName)
print('device_channel_indices:',device_channel_indices)

# Read the ephys data (stream names are 'Signals AUX', 'Signals CH', 'Signals ADC')
raw_rec = si.read_openephys(openephys_folder,stream_name='Signals CH',ignore_timestamps_errors=True)
print('raw_rec',raw_rec)

# Remove the first and last two seconds
sampling_frequency = raw_rec.sampling_frequency
[... 127 lines cut ...]
    new_kwargs[name] = _load_extractor_from_dict(value)
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/core/base.py", line 998, in _load_extractor_from_dict
    new_kwargs[name] = _load_extractor_from_dict(value)
  [Previous line repeated 1 more time]
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/core/base.py", line 1017, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py", line 79, in __init__
    NeoBaseRecordingExtractor.__init__(
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 185, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 25, in __init__
    self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py", line 64, in get_neo_io_reader
    neo_reader.parse_header()
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/neo/rawio/baserawio.py", line 178, in parse_header
    self._parse_header()
  File "/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/CambridgeProbeSpikeSortingResults/2022-02-25_09-04-11/in_container_python_base/lib/python3.9/site-packages/neo/rawio/openephysrawio.py", line 120, in _parse_header
    raise ValueError(
ValueError: Not continuous timestamps for 100_AUX1.continuous. Maybe because recording was paused/stopped.

```

It seems like the container isn't properly using the sliced recording and that it is attempting to reload the recording. I'd appreciate any thoughts people might have on this issue. Thank you!

~Nick

zm711, 2023-10-20:
@nmtimme, I honest don't know the containers very well, I'll loop in @alejoe91 at this point so he can double check.

alejoe91, 2023-10-20:
Hi @nmtimme 

@zm711 thanks for helping with this!

So the problem seems to be related to the version of NEO that is installed in the container. If your spikeinterface version is from source (e.g., `si.__version__` has a `dev` in it), then NEO will be installed from its master branch and the problem should be solved. If you are using a release version of SI, for example 0.98.2, NEO will be installed from pip and we have to wait for a new NEO release. 

So I recommend installing SpikeInterface from source. See instructions here: https://spikeinterface.readthedocs.io/en/latest/installation.html#from-source

nmtimme, 2023-10-20:
Hi @alejoe91 !

Thanks for your quick response! When we start up the code, we print out the versions of si and neo and we get:

si version: 0.99.0.dev0
neo version: 0.13.0.dev0

So, it seems like we're using the proper versions of si and neo, right? I deleted the kilosort3-compiled-base:latest.sif and reran the analysis to make sure we downloaded the container with the most recent version of si, but the error persists. Is there a way to print the version of neo being used in the container? All it reports back is:

Installing extra requirements: ['neo']

I'd appreciate any other suggestions you can provide. Thank you!

~Nick

zm711, 2023-10-28:
@nmtimme,

There's a release prep happening right now, but I'll make sure this doesn't fall off the radar. I also linked an issue on the Neo side about some issues with dropped signals in case you want to read about this (if you're planning to use a lot of openephys through spikeinterface/neo).

zm711, 2023-11-06:
@alejoe91, when you get a moment. Can you look at this one again? There was an additional issue with the container that @nmtimme was asking and I promised I would ping you back in after the release.

alejoe91, 2023-11-06:
@nmtimme can you paste here the output of:
```
print(raw_rec._kwargs)
```
I think we might not be correctly saving the `ignore_timestamps_error` internally

alejoe91, 2023-11-06:
@nmtimme actually I think I found the issue. Can you install from this PR? 
https://github.com/SpikeInterface/spikeinterface/pull/2174

claudio1212, 2023-11-17:
Hi, I'm having the exact same problem as this user. I'm trying to sort an OE legacy recording (one .continuous file per recording) through the last version of spikeinterface, but it returns the same error ad in this thread even using the "ignore timestamp error=true" flag in read_open_ephys. I'm pretty sure there are no gaps in my recording, given that I've checked it through Neuroexplorer.

zm711, 2024-04-11:
We've added more updates to the OE legacy reader in Neo including getting rid of ignore_timestamp_error and instead have a fill option. Feel free to test and let us know if that works!