SpikeInterface/spikeinterface #2117: Exporting consensus sorting to Phy

adam-hockley, 2023-10-19:
Is it possible to export consensus-based sorting to Phy format? 
The AgreementSortingExtractor from get_agreement_sorting does not seem to work as the sorting input to extract_waveforms, is there another way to extract the consensus-sorted waveforms for Phy export?

Error showing on si.extract_waveforms: "Object of type MultiSortingComparison is not JSON serializable"

Relevant code:
    # Compare sorter outputs and calculate consensus
    MultiSort = si.compare_multiple_sorters(sorting_list=[sorting_sc2, sorting_tdc2, sorting_ms5],name_list=['SC2', 'TDC2', 'MS5'],verbose=True,)
    Agree_2 = MultiSort.get_agreement_sorting(minimum_agreement_count=2)
    
    # Export to Phy format
    we = si.extract_waveforms(recording=rec_cmr, sorting=Agree_2, folder='waveforms', sparse=True)

    # some computations are done before to control all options
    compute_spike_amplitudes(waveform_extractor=we)
    compute_principal_components(waveform_extractor=we, n_components=3, mode='by_channel_global')

    # the export process is fast because everything is pre-computed
    export_to_phy(waveform_extractor=we, output_folder='2',copy_binary=False)

Thanks!!
Adam

alejoe91, 2023-10-19:
Hi Adam!

You should be able to `save` it first and then extract waveforms:
```
Agree_2_saved = Agree_2.save(folder="your-preferred-folder")
we = si.extract_waveforms(recording=rec_cmr, sorting=Agree_2_saved, folder='waveforms', sparse=True)
```

adam-hockley, 2023-10-20:
Saving the agreement sorting, doesn't work either, I get the same error:

Agree_2_saved = Agree_2.save(folder="agreement")
Traceback (most recent call last):

  Cell In[32], line 1
    Agree_2_saved = Agree_2.save(folder="agreement")

  File ~\anaconda3\envs\SIenvLap\lib\site-packages\spikeinterface\core\base.py:749 in save
    loaded_extractor = self.save_to_folder(**kwargs)

  File ~\anaconda3\envs\SIenvLap\lib\site-packages\spikeinterface\core\base.py:818 in save_to_folder
    self.dump(provenance_file)

  File ~\anaconda3\envs\SIenvLap\lib\site-packages\spikeinterface\core\base.py:558 in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)

  File ~\anaconda3\envs\SIenvLap\lib\site-packages\spikeinterface\core\base.py:596 in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),

  File ~\anaconda3\envs\SIenvLap\lib\json\__init__.py:238 in dumps
    **kw).encode(obj)

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:201 in encode
    chunks = list(chunks)

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:431 in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:405 in _iterencode_dict
    yield from chunks

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:405 in _iterencode_dict
    yield from chunks

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:438 in _iterencode
    o = _default(o)

  File ~\anaconda3\envs\SIenvLap\lib\site-packages\spikeinterface\core\core_tools.py:106 in default
    return super().default(obj)

  File ~\anaconda3\envs\SIenvLap\lib\json\encoder.py:179 in default
    raise TypeError(f'Object of type {o.__class__.__name__} '

TypeError: Object of type MultiSortingComparison is not JSON serializable

alejoe91, 2023-10-20:
@adam-hockley see intructions here: https://spikeinterface.readthedocs.io/en/latest/installation.html#from-source

adam-hockley, 2023-10-20:
Perfect, works fine on this updated version (0.99.0.dev0)

Thanks!!