SpikeInterface/spikeinterface #2118: Getting KeyError: 'relative_path' while implementing any sorting algorithms.

Ananyamuguli, 2023-10-21:
I have a NumpyRecording object. While running Sorters, it given me a KeyError: 'relative_path' . Any help would be much appreciated.
SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/Users/ananyamuguli/Documents/BrainResearch/Data/in_container_sorter_script.py", line 17, in <module>
    sorting = run_sorter_local(
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/runsorter.py", line 174, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 269, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py", line 234, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/sorters/external/mountainsort4.py", line 92, in _run_from_folder
    recording = load_extractor(
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/core/base.py", line 1065, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/core/base.py", line 659, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
  File "/usr/local/lib/python3.8/site-packages/spikeinterface/core/base.py", line 428, in from_dict
    if dictionary["relative_paths"]:
KeyError: 'relative_paths'

Spike sorting failed. You can inspect the runtime trace in /Users/ananyamuguli/Documents/BrainResearch/Data/mountainsort4_output/spikeinterface_log.json.

samuelgarcia, 2023-10-23:
Hi Ananya.
NumpyRecording do not play well with sorters iniside docker. The best is to save it in our binary folder format before running a sorter.

```python
rec = NumpyReording(...)

rec_saved = rec.save(folder='/some/folder/', format='binary')

sorting = run_sorter('moutainsort4', rec_saved, ...)

```

Ananyamuguli, 2023-10-26:
Thank you so much!! That worked :)

closed 2023-10-26 by alejoe91