SpikeInterface/spikeinterface #2120: Quality Metrics Test Failure `Amplitude Cutoff`

zm711, 2023-10-23:
FAILED src/spikeinterface/qualitymetrics/tests/test_quality_metric_calculator.py::QualityMetricsExtensionTest::test_peak_sign
```
array([0.00091969, 0.00272539, 0.00087973, 0.00071159, 0.00156904,\n       0.00126182, 0.00047331, 0.00348382, 0.00120492, 0.00035983]), 
array([0.00279436, 0.00272539, 0.00087973, 0.00071159, 0.00156904,\n       0.00126182, 0.00047331, 0.00348382, 0.00120492, 0.00035983]), atol=0.001)

array([0.00091969, 0.00272539, 0.00087973, 0.00071159, 0.00156904,\n       0.00126182, 0.00047331, 0.00348382, 0.00120492, 0.00035983]), 
array([0.00279436, 0.00272539, 0.00087973, 0.00071159, 0.00156904,\n       0.00126182, 0.00047331, 0.00348382, 0.00120492, 0.00035983]), atol=0.001)

```
from @h-mayorquin recent PR

```
array([0.00324951, 0.00035942, 0.00971237, 0.00269361, 0.00908869,\n       0.00062211, 0.00282141, 0.00120306, 0.0005949 , 0.00176459]), 
array([0.00324951, 0.00035942, 0.00971237, 0.00269361, 0.00908869,\n       0.00292164, 0.00282141, 0.00120306, 0.0005949 , 0.00176459]), atol=0.001)

array([0.00324951, 0.00035942, 0.00971237, 0.00269361, 0.00908869,\n       0.00062211, 0.00282141, 0.00120306, 0.0005949 , 0.00176459]), 
array([0.00324951, 0.00035942, 0.00971237, 0.00269361, 0.00908869,\n       0.00292164, 0.00282141, 0.00120306, 0.0005949 , 0.00176459]), atol=0.001)

```
from @samuelgarcia recent PR

It looks to me like there is one failure that re-occurs during a particular PR (maybe there is a seeding thing?).  In the two PRs being compared the same failure keeps occurring, but it is a different failure between PRs.

closed 2023-10-23 by alejoe91

alejoe91, 2023-10-23:
@zm711 there was also a missing seed in Ramon's PR: https://github.com/SpikeInterface/spikeinterface/blob/52261e2199d5095258b066c251ceae943cc3b939/src/spikeinterface/qualitymetrics/tests/test_quality_metric_calculator.py#L213

zm711, 2023-10-23:
There are potentially a few more seeds missing right? I think I remember a PR changing the seed behavior that led to a lot of tests needing seeds added. So I suppose eventually all the missing tests seeds will be found (if not already 🤞 )

h-mayorquin, 2023-10-24:
It is just strange though that the tests passing relies on a specific seed.

Here for example:
https://github.com/SpikeInterface/spikeinterface/issues/1972

Is this the case here as well?

If the test only passess for certian seeds, then there is something wrong with the test. Am I wrong?

alejoe91, 2023-10-24:
It's not for a specific seed. The failing tests was testing that metrics extracted using the `neg` peak are the same as the `pos` peak, when the recording is inverted. In this case, we have to make sure that the sampeld waveforms are the same. Note that the seed is on the `extract_waveforms`|

h-mayorquin, 2023-10-24:
Thanks for clarifying. 

I saw the seed in the `extract_waveforms`. I just didn't know what it means over there. I am gald that is not the same case as the other test that I pointed.

PR #2121 (2023-10-23): Use default cutouts for peak-sign test
Fixes #2120 

Turns out that we recently changed the default `ms_before`/`ms_after` but in one of the tests we were forcing previous defaults.