SpikeInterface/spikeinterface #2127: Cannot load mearec dataset

florian6973, 2023-10-24:
Hi,

Here is a bug noticed when trying to use mearec. Would you have any ideas about how to solve it please?

MWE (spikeinterface 0.98.0, mearec 1.9.0):
```
import spikeinterface as si
import spikeinterface.extractors as se

local_path = si.download_dataset(remote_path='mearec/mearec_test_10s.h5')
recording, sorting_true = se.read_mearec(local_path)
```

Error:
> Traceback (most recent call last):
  File "C:\Users\fl\Documents\Git\dp\test_mearec.py", line 8, in <module>
    recording, sorting_true = se.read_mearec(local_path)
  File "C:\Users\fl\miniconda3\envs\dp\lib\site-packages\spikeinterface\extractors\neoextractors\mearec.py", line 109, in read_mearec
    recording = MEArecRecordingExtractor(file_path)
  File "C:\Users\fl\miniconda3\envs\dp\lib\site-packages\spikeinterface\extractors\neoextractors\mearec.py", line 34, in __init__
    NeoBaseRecordingExtractor.__init__(self, all_annotations=all_annotations, **neo_kwargs)
  File "C:\Users\fl\miniconda3\envs\dp\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 185, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "C:\Users\fl\miniconda3\envs\dp\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 25, in __init__
    self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
  File "C:\Users\fl\miniconda3\envs\dp\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 63, in get_neo_io_reader
    neo_reader = neoIOclass(**neo_kwargs)
TypeError: __init__() got an unexpected keyword argument 'load_spiketrains'

Temporary fix:
I add
```
del neo_kwargs["load_spiketrains"]
del neo_kwargs["load_analogsignal"]
```
before calling NeoBaseRecordingExtractor in `spikeinterface\extractors\neoextractors\mearec.py`,
but I assume this is not the correct way...

Thanks for your help,

Best,

florian6973

zm711, 2023-10-25:
@florian6973 

I'm wondering if you do 

```python
pip install -U spikeinterface
```
to install the latest version of spikeinterface (0.98.2) if that fixes the issue.

I just tried the set of commands on the dev version (0.99.0.dev) and it worked fine with no problems. Could you try updating to 0.98.2 and reporting if that fixes the problem. If it doesn't the instructions to install the dev version are [here](https://spikeinterface.readthedocs.io/en/latest/installation.html).

florian6973, 2023-10-25:
Thanks for your prompt reply! Yes updating to 0.98.2 fixes the issue. Thanks!

closed 2023-10-25 by florian6973