SpikeInterface/spikeinterface #2131: Import sorted Kilosort3 data that was run in MATLAB

depreterc, 2023-10-25:
Hi newbie here,

How do you load in Kilosort3 output that has been previously sorted outside of spikeinterface? We would like to bring this sorted data into spikeinterface for post processing using the spikeinterface tools.

Thanks!

zm711, 2023-10-25:
@depreterc 

For complete newbies (we all start somewhere :) ) There is a ton of information in the [documentation]( https://spikeinterface.readthedocs.io/en/latest/). Most raw file formats and sorters will have reading functions.

for example, 
```python
import spikeinterface as si
import spikeinterface.extractors as se

kilosort_sorting = se.read_kilosort(folder_path="kilosort-folder") #put in your folder path here!
```

You'll also want to load your raw data (the recording). There are tons of 'extractors' that SpikeInterface
[extractors](https://spikeinterface.readthedocs.io/en/latest/modules/extractors.html#read-one-recording) (check this page).

So you would do
```python
my_recording = se.read_xxx() # the read function needs to be your raw data format

# then it's decent practice to
kilosort_sorting.register_recording(my_recording)
```

After this point you can explore any of the functions in the documentation that you need, apply them, and if you run into any issues open a new issue! 

Give this a try and let us know how it goes.

depreterc, 2023-10-25:
se.read_kilosort was exactly what we needed!

Thank you! :)

closed 2023-10-26 by alejoe91