SpikeInterface/spikeinterface #2132: The `get_extremum_channels` should have an `auto` option.

alejoe91, 2023-10-26:
Currently the `get_extremum_channels` (and many other functions) require a `peak_sign`. However, since the input of all these functions is usually a `WaveformExtractor`, we could add the option to automatically estimate the peak_sign from the templates.

e.g.:
```
if peak_sign == "auto":
    mean_template_value = np.mean(all_templates)
    peak_sign = "neg" if mean_template_value < 0 else "pos"
```

What do you guys think?

zm711, 2023-11-02:
I like this! I've been meaning to comment on this for a spell.

Sometimes I get pretty nasty opto artifacts that have giant positive peaks. So I would be concerned the mean might see those and get it wrong. The skew seems like it would be more robust to some of my datasets since most of the neurons themselves have nice negative peaks (albeit often smaller than the sorted artifacts).

alejoe91, 2023-11-02:
I see, on the other hand it would look at the distribution of most of the templates, so having a few "off" templates should be ok!