SpikeInterface/spikeinterface #2140: Unable to use backend of "ipywidgets" in any plotting function

Laohusong, 2023-10-30:
When I use "ipywidget" backend, it raises errors. It happens when I am trying to analyze a 170-min-long, 384-channel, neuropixel data. But matplotlib style always works. 

For example, I am trying to run 
`w = si.plot_unit_templates(we, backend="ipywidgets")`
 it will raise error like this:
 ---------------------------------------------------------------------------
TraitError                                Traceback (most recent call last)
[d:\laohu_pipeline\laohu_pipeline.ipynb](file:///D:/laohu_pipeline/laohu_pipeline.ipynb) Cell 35 line 1
----> [1](vscode-notebook-cell:/d%3A/laohu_pipeline/laohu_pipeline.ipynb#X46sZmlsZQ%3D%3D?line=0) w = si.plot_unit_templates(we, backend="ipywidgets")

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\spikeinterface\widgets\base.py:120](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/spikeinterface/widgets/base.py:120), in define_widget_function_from_class.<locals>.widget_func(*args, **kwargs)
    117 @copy_signature(widget_class)
    118 def widget_func(*args, **kwargs):
    119     W = widget_class(*args, **kwargs)
--> 120     W.do_plot(W.backend, **W.backend_kwargs)
    121     return W.plotter

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\spikeinterface\widgets\base.py:53](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/spikeinterface/widgets/base.py:53), in BaseWidget.do_plot(self, backend, **backend_kwargs)
     51 plotter = self.possible_backends[backend]()
     52 self.check_backend_kwargs(plotter, backend, **backend_kwargs)
---> 53 plotter.do_plot(self.plot_data, **backend_kwargs)
     54 self.plotter = plotter

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\spikeinterface\widgets\ipywidgets\unit_templates.py:7](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/spikeinterface/widgets/ipywidgets/unit_templates.py:7), in UnitTemplatesPlotter.do_plot(self, data_plot, **backend_kwargs)
      6 def do_plot(self, data_plot, **backend_kwargs):
----> 7     super().do_plot(data_plot, **backend_kwargs)
      8     self.controller["plot_templates"].layout.visibility = "hidden"

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\spikeinterface\widgets\ipywidgets\unit_waveforms.py:77](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/spikeinterface/widgets/ipywidgets/unit_waveforms.py:77), in UnitWaveformPlotter.do_plot(self, data_plot, **backend_kwargs)
     74 for w in self.controller.values():
     75     w.observe(self.updater)
---> 77 self.widget = widgets.AppLayout(
     78     center=fig_wf.canvas,
     79     left_sidebar=unit_widget,
     80     right_sidebar=fig_probe.canvas,
     81     pane_widths=ratios,
     82     footer=footer,
     83 )
     85 # a first update
     86 self.updater(None)

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\ipywidgets\widgets\widget_templates.py:157](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/ipywidgets/widgets/widget_templates.py:157), in AppLayout.__init__(self, **kwargs)
    156 def __init__(self, **kwargs):
--> 157     super().__init__(**kwargs)
    158     self._update_layout()

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\ipywidgets\widgets\widget_box.py:64](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/ipywidgets/widgets/widget_box.py:64), in Box.__init__(self, children, **kwargs)
     62 def __init__(self, children=(), **kwargs):
     63     kwargs['children'] = children
---> 64     super().__init__(**kwargs)

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\ipywidgets\widgets\widget.py:503](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/ipywidgets/widgets/widget.py:503), in Widget.__init__(self, **kwargs)
    501 """Public constructor"""
    502 self._model_id = kwargs.pop('model_id', None)
--> 503 super().__init__(**kwargs)
    505 Widget._call_widget_constructed(self)
    506 self.open()

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\ipywidgets\widgets\widget_templates.py:80](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/ipywidgets/widgets/widget_templates.py:80), in LayoutProperties.__init__(self, **kwargs)
     79 def __init__(self, **kwargs):
---> 80     super().__init__(**kwargs)
     81     self._property_rewrite = defaultdict(dict)
     82     self._property_rewrite['align_items'] = {'top': 'flex-start',
     83                                              'bottom': 'flex-end'}

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:1376](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:1376), in HasTraits.__init__(self, *args, **kwargs)
   1374 for key, value in kwargs.items():
   1375     if self.has_trait(key):
-> 1376         setattr(self, key, value)
   1377         changes[key] = Bunch(
   1378             name=key,
   1379             old=None,
   (...)
   1382             type="change",
   1383         )
   1384     else:
   1385         # passthrough args that don't set traits to super

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:750](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:750), in TraitType.__set__(self, obj, value)
    748     raise TraitError('The "%s" trait is read-only.' % self.name)
    749 else:
--> 750     self.set(obj, value)

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:723](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:723), in TraitType.set(self, obj, value)
    722 def set(self, obj: HasTraits, value: S) -> None:
--> 723     new_value = self._validate(obj, value)
    724     assert self.name is not None
    725     try:

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:756](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:756), in TraitType._validate(self, obj, value)
    754     return value
    755 if hasattr(self, "validate"):
--> 756     value = self.validate(obj, value)
    757 if obj._cross_validation_lock is False:
    758     value = self._cross_validate(obj, value)

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:2322](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:2322), in Instance.validate(self, obj, value)
   2320     return value
   2321 else:
-> 2322     self.error(obj, value)

File [c:\Users\ASUS\anaconda3\envs\si_env\Lib\site-packages\traitlets\traitlets.py:860](file:///C:/Users/ASUS/anaconda3/envs/si_env/Lib/site-packages/traitlets/traitlets.py:860), in TraitType.error(self, obj, value, error, info)
    854 else:
    855     e = "The '{}' trait expected {}, not {}.".format(
    856         self.name,
    857         self.info(),
    858         describe("the", value),
    859     )
--> 860 raise TraitError(e)

TraitError: The 'center' trait of an AppLayout instance expected a Widget or None, not the FigureCanvasAgg at '0x1e5e0274990'.

Laohusong, 2023-10-30:
Thanks a lot! It works! I found I set “%matplotlib inline” at the very beginning. 😂

closed 2023-10-30 by Laohusong

zm711, 2023-10-30:
@samuelgarcia, This might be a documentation issue. I think when to use `%matplotlib inline` and when to use `%matplotlib widget` is not clear in the documentation. I'll see if I can clarify that in the docs when I have time.

zm711, 2023-10-30:
Sounds good Sam!