SpikeInterface/spikeinterface #2142: KeyError: 'probe_name' in OpenEphysBinaryRecordingExtractor

LiesDec, 2023-10-30:
When creating a `OpenEphysBinaryRecordingExtractor` object, the probe name is required (L190)
` probe_name = probe.annotations["probe_name"]`
https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/neoextractors/openephys.py#L190
This gives the error:
`KeyError: 'probe_name'`

Based on the `probinterface.read_openephys()` method,  I think the probe name should be accessed by 
` probe_name = probe.name`
see https://github.com/SpikeInterface/probeinterface/blob/main/src/probeinterface/io.py#L1609C5-L1616C6

Thanks for considering this edit :)

LiesDec, 2023-10-30:
hi @alejoe91,

Thanks a lot :)

cheers

closed 2023-10-31 by alejoe91

nmtimme, 2023-11-10:
Just a brief addition: I was trying to sort Neuropixel data using version 0.98.2 of spike interface and I got an error like this using kilosort3 in the container. It has previously worked, but some update occurred in the container version that produced this error. Updating the version of spike interface resolved the error. Thanks!