SpikeInterface/spikeinterface #2144: Can't run Kilosort2_5 from local repository

luiztauffer, 2023-10-31:
When trying to use kilosort 2_5 from a local repository, I keep getting this error:
```
Exception: The sorter kilosort2_5 is not installed.
```

The KS repository exists locally and I'm passing the correct path so SI sorter.
I suspect the problem might be at `sorters > external> kilosort2_5.py`:
```python
def check_if_installed(kilosort2_5_path: Union[str, None]):
    if kilosort2_5_path is None:
        return False
    assert isinstance(kilosort2_5_path, str)

    if kilosort2_5_path.startswith('"'):
        kilosort2_5_path = kilosort2_5_path[1:-1]
    kilosort2_5_path = str(Path(kilosort2_5_path).absolute())

    if (Path(kilosort2_5_path) / "master_kilosort.m").is_file() or (
        Path(kilosort2_5_path) / "main_kilosort.m"
    ).is_file():
        return True
    else:
        return False
```

If you check the [Kilosort repository](https://github.com/MouseLand/Kilosort), you'll see that indeed the files `master_kilosort.m` and `main_kilosort.m` do not exist.

alejoe91, 2023-10-31:
@luiztauffer you have to checkout v2.5: https://github.com/MouseLand/Kilosort/tree/v2.5

alejoe91, 2023-10-31:
The current main is KS3

luiztauffer, 2023-10-31:
thanks @alejoe91 ! We should add this to the error track of other kilosort versions

alejoe91, 2023-10-31:
Or better in the `Installing Spike Sorters` docs section!

closed 2023-10-31 by luiztauffer