SpikeInterface/spikeinterface #2151: help with understanding Spike waveforms (basic)

kshtjkumar, 2023-11-01:
I used the following code to generate waveforms:
[image]

The results i got:
[image]

I want some help with the x axis of the graph here! is the time in milliseconds ? 

We got a review on the graph saying:
the single unit spike snippets are showing tens of ms which is abnormal. It is curious if the authors actually captured unit spikes or other signals such as EMGs.
the exact image the review was on is this:
[image]

Can someone explain how to interpret these?

alejoe91, 2023-11-01:
The x axis are is sample index. To plot timestamps in ms, you can compute the values as this:
```
import numpy as np

num_samples = we.nbefore + we.nafter
timestamps_ms = np.arange(num_samples) / we.sampling_frequency * 1000
```

alejoe91, 2023-11-01:
If you plot without an `x` (like in your case, you are only providing the `y`, i.e. the waveforms), the x axis is always going to be from 0 to num_samples - 1.

kshtjkumar, 2023-11-01:
i got this output:
[image]

kshtjkumar, 2023-11-01:
i got this as output:
[image]

alejoe91, 2023-11-01:
Which output? The link is broken on my side

kshtjkumar, 2023-11-01:
[image]

kshtjkumar, 2023-11-01:
[image]

kshtjkumar, 2023-11-01:
i am not able to upload the image ! it keeps saying like above!

alejoe91, 2023-11-01:
Anyways, you should be getting from 0 to 7ms if your sampling frequency is 30kHz

kshtjkumar, 2023-11-01:
yep the x axis is from 0 to 7 !
So now,  i can use this for all waveform sorting ?

alejoe91, 2023-11-01:
Yes, that is the same for all waveforms. It is set when you extract waveforms with the `ms_before`\`ms_after`. See docs here: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.extract_waveforms

kshtjkumar, 2023-11-01:
thank you so much! As always You have been of great help!

closed 2023-11-01 by kshtjkumar