SpikeInterface/spikeinterface #2164: Windows Permission Errors in SI

zm711, 2023-11-03:
## Motivation ##
As discussed in #2156, this is just a little summary of some hiccups that Windows users have bumped into dealing with the management of tempfiles or deleting files while running SpikeInterface. The total count of user complaints 5 across 3 issues. The 6th  user/4th issue is only semi-related to SpikeInterface since the recording holds a reference to the file while its open so it shouldn't be expected to be delete-able while the recording object exists. I will say I tested `overwrite` with a waveform extractor today and ran into no issues doing a folder overwrite. So my potential concern there was actually not correct. 

## Errors caused by SpikeInterface ##

#1871--3 users ran into WinError32 Error: maybe related to SC2 deleting temp files while running?

#1396 `check_your_install.py` fails due to PermissionError: [WinError 32] The process cannot access the file because it is being used by another process. I fixed by creating a separate cleanup script for Windows and skipping clean up in the check_your_install on Windows

#1225 WinError32 annoyed a user so much they switched to Linux (maybe that's the secret agenda of SI after all :) ). Error was related to `n_jobs` and docker.

## Errors only semi-related to SpikeInterface (ie not internal to package but happening while interacting with the package) ##

#1790: trying to clean up file while SpikeInterface was still holding a reference to them in Windows (WinError32)

## Testing ##

As mentioned in #2138, I tried running the testing suite locally on my Windows machine and was met with tons of WinError32 Errors.  I also tried the suggestion of changing the read-only bit on Windows as suggested in the [python documentation](https://docs.python.org/3/library/shutil.html) and it did not work. Maybe my reading was wrong, but I thought they were getting rid of the onerror and switching to onexec instead so for older python we would need to provide the read-only func to onerror and in future python version it would need to go through onexec. But maybe I was wrong. Either way my current environment for spikeinterface is python 3.11 so my test was using the onerror parameter. And the failure was still WinError32.

## Suggestions ##

In #2156, I had suggested at least adding some explicit testing of the folder generation and deletion into the core tests to ensure that the core writing/removing functions work on Mac/Linux/Windows and I still think that is a good idea since I, for one, don't know when the WinError 32 will exactly occur (and they are super annoying when they do occur).  Or to be more precise I know they occur when things are happening in the same process (like all of pytest), but I never know how to predict when SpikeInterface will generate new processes or use the same process for an operation.

@h-mayorquin, tagging you since the issue linking was so you could re-read over them.

h-mayorquin, 2023-11-04:
Thanks a lot for organizing this.
Yes, I don't know how clear I was in the other PR but I believe this should be tested and fixed.
What I was also saying is that I think it does not make sense to make stand alone tests for this. Reason being is that I believe this is mainly a multiprocessing related issue.

Reading the issues together reinforces my view that this is mainly a multiprocessing issue. We know that in opposition to linux, where files are only scheduled for removal if they are being used, windows throws an error if you attempt a shutil On top of that (our) multiprocessing has some bugs where the file handles are not unreferenced. The latter, you can see it in the CI tests and sometimes running pytests locally which from time to time breaks my test folder -in linux- in a similar way.

So I believe that the tests need to be focused on that specific contex to be sure that we are solving the problem: removal of files and directories in a multiprocessing context. We need to look at the code base where that happens (you already located some cases) and create tests in those context or in a simplified version of it.

In a way this is to be expected. This is a combination of a three complicated techniques with known footgunts: 1) Multiprocessing, 2) IO in an OS independent way, 3) Global variables. And now we are even adding sharing memory to the mix.

Two things that I believe could be long-hanging fruit to alleviate somehow this problem and reduce the complexity of future maintenance in general:
* Do we need to handle the temporary directory through the globals? Mixing globals with temporary files sounds like an avenue to create this type of errors and depending on the context this might keep the folders marked as referenced even if they are not.
* Do we need to be removing all the files ourselves? Is this just a convenience thing? Maybe we should just delegate OS tasks to the OS instead of doing it ourselves.

This is some knowledge that I lack, no rethorical questions. Maybe there is another way of solving the original problem that the use of global variables and handling removal ourselves were intendeded to solve.

A thidd possibility is to use the ignore_errors flag in `shutil.rmtree`:
https://docs.python.org/3/library/shutil.html#shutil.rmtree

This will hide the shallow bugs and I will avoid the programs crashing in some cases.

The first thing anyway will be to run the full tests on windows just to see if we at least get some reliable case that we can test.

zm711, 2023-11-04:
Thanks @h-mayorquin! I think having an appropriate test/test framework will serve everyone in the long-term better rather than just tossing a few random ad-hoc tests into the suite. So I appreciate all the thought you are putting into this! On Monday, I'll try the ignore_errors flag myself in the test suite and see what happens, I'm curious how it will go. 

> The first thing anyway will be to run the full tests on windows just to see if we at least get some reliable case that we can test.

I agree for sure, but I hear there's preponderated democracy here :)

h-mayorquin, 2023-11-05:
I was getting the error on windows while these tests with pytest:

#2170

Linking here for tracking.

samuelgarcia, 2023-11-06:
Hi Zach,
thanks a lot for this summary.
I recently installed a window with double boot on my laptop!
I am a  bit busy for the next 2 weeks but I promise to try to spent some time on this.
I have the intuition that we can fix issue some of then with some array.copy() when underlying memap is used (for plotting for instance).
Lets how full tests on window will go.

zm711, 2023-11-06:
Thanks Sam. Appreciate you checking it out too. Try not to fall in love with Windows during this "trial" period.

samuelgarcia, 2023-11-22:
The fight is not terminated but some bug are fixed here : #2221 
For instance ethe spykingcircus2 one.

zm711, 2023-12-04:
Just wanted to update this for the current issues on main ( I just pulled main before running the test suite).

```
FAILED src/spikeinterface/sorters/external/tests/test_tridesclous.py::TridesclousCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...
FAILED src/spikeinterface/sorters/internal/tests/test_spykingcircus2.py::SpykingCircus2SorterCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...
FAILED src/spikeinterface/sorters/internal/tests/test_tridesclous2.py::Tridesclous2SorterCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...FAILED src/spikeinterface/sorters/tests/test_find_folders.py::test_find_recording_folders - AssertionError: assert 1 == 3
FAILED src/spikeinterface/sorters/tests/test_launcher.py::test_run_sorters_with_dict - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...
FAILED src/spikeinterface/sorters/tests/test_runsorter.py::test_run_sorter_docker - ModuleNotFoundError: No module named 'docker'
FAILED src/spikeinterface/sorters/tests/test_runsorter.py::test_run_sorter_singularity - ModuleNotFoundError: No module named 'spython'
FAILED src/spikeinterface/sortingcomponents/tests/test_benchmark_matching.py::test_run_matching_vary_parameter[parameters1-fraction_misclassed] - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: '35T...
FAILED src/spikeinterface/sortingcomponents/tests/test_benchmark_matching.py::test_run_matching_vary_parameter[parameters2-fraction_missing] - AssertionError: folder 35TH2K7B_sorting0 already exists, choose another name or use overwrite=True
FAILED src/spikeinterface/widgets/tests/test_widgets.py::TestWidgets::test_plot_sorting_summary - TypeError: AverageWaveformItem.__init__() got an unexpected keyword argument 'waveform_percentiles'
FAILED src/spikeinterface/widgets/tests/test_widgets.py::TestWidgets::test_plot_traces - OSError: cannot load library 'libgobject-2.0-0.dll': error 0x780.  Additionally, ctypes.util.find_library() did not...
FAILED src/spikeinterface/widgets/tests/test_widgets.py::TestWidgets::test_plot_unit_templates - TypeError: AverageWaveformItem.__init__() got an unexpected keyword argument 'waveform_percentiles'
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_spykingcircus - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_mountainsort4 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_mountainsort5 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_tridesclous - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_ironclust - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_waveclus - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_hdsort - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_kilosort1 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers.py::test_combinato - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers_gpu.py::test_kilosort2 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers_gpu.py::test_kilosort2_5 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers_gpu.py::test_kilosort3 - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers_gpu.py::test_yass - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
ERROR src/spikeinterface/sorters/external/tests/test_singularity_containers_gpu.py::test_pykilosort - AssertionError: folder C:\Users\ZacharyMcKenzie\Documents\GitHub\spikeinterface\test_folder\core\tmp_folder\toy alr...
```

Total counts were (~15 failed, 14 errors).

samuelgarcia, 2023-12-07:
Merci Zach. they are not related to the permissionn error but there are still some.
I will take a look quite soon.

zm711, 2023-12-07:
I buried the lead by including so many (and the bottom ones are not the beloved Win32). The top few for (tridesclous and spkyingcircus) are permission errors. I recopy just the Win32 error below :) 

```
FAILED src/spikeinterface/sorters/external/tests/test_tridesclous.py::TridesclousCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...
FAILED src/spikeinterface/sorters/internal/tests/test_spykingcircus2.py::SpykingCircus2SorterCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...
FAILED src/spikeinterface/sorters/internal/tests/test_tridesclous2.py::Tridesclous2SorterCommonTestSuite::test_with_run - PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\...FAILED src/spikeinterface/sorters/tests/test_find_folders.py::test_find_recording_folders - AssertionError: assert 1 == 3
```

JoeZiminski, 2024-05-16:
Permission errors also occurs when building docs, sporadically in sphinx-gallery examples, leading the examples to not be generated.

JoeZiminski, 2024-06-05:
The test `test_principal_components.test_compute_for_all_spikes` is failing on windows, only in the `sparse=True` case.

zm711, 2024-09-20:
reminder to retest @JoeZiminski and me with current state of code :)

JoeZiminski, 2024-09-26:
Nice one @zm711 I reran:

<details>

```
======================================================================================================================================================================================== short test summary info ======================================================================================================================================================================================== 
FAILED src/spikeinterface/extractors/tests/test_cellexplorerextractor.py::CellExplorerSortingTest::test_open - ImportError: To use the CellExplorerSortingExtractor install pymatreader
FAILED src/spikeinterface/extractors/tests/test_cellexplorerextractor.py::CellExplorerSortingTest::test_pickling - ImportError: To use the CellExplorerSortingExtractor install pymatreader
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MearecRecordingTest::test_neo_annotations - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MearecRecordingTest::test_open - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MearecRecordingTest::test_pickling - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MearecSortingTest::test_open - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MearecSortingTest::test_pickling - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::NeuroScopeSortingTest::test_open - ModuleNotFoundError: No module named 'lxml'
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::NeuroScopeSortingTest::test_pickling - ModuleNotFoundError: No module named 'lxml'
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MaxwellRecordingTest::test_neo_annotations - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MaxwellRecordingTest::test_open - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::MaxwellRecordingTest::test_pickling - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::BiocamRecordingTest::test_neo_annotations - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::BiocamRecordingTest::test_open - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::BiocamRecordingTest::test_pickling - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::EDFRecordingTest::test_neo_annotations - ValueError: Requires pyedflib
FAILED src/spikeinterface/extractors/tests/test_neoextractors.py::EDFRecordingTest::test_open - ValueError: Requires pyedflib
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_of_ragged_arrays[True] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_of_ragged_arrays[False] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_start_time[True] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_start_time[False] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_start_time_from_series[True] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_sorting_extraction_start_time_from_series[False] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_multiple_unit_tables[True] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors.py::test_multiple_unit_tables[False] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
FAILED src/spikeinterface/extractors/tests/test_nwbextractors_streaming.py::test_recording_s3_nwb_fsspec[True] - ImportError: DLL load failed while importing _errors: The specified module could not be found.
[... cut ...]
========================================================================================================================================================== 58 failed, 2876 passed, 113 skipped, 361 warnings, 48 errors in 1514.33s (0:25:14) ===========================================================================================================================================================
```

</details>

Not so bad 58 / 2876! I just installed in a fresh env and ran, not sure if some can be solved with simple stuff I have not closely inspected the outputs.

zm711, 2024-12-03:
It looks like most of the errors are related to missing packages etc. With Heberto having said up Windows testing in CI I think we can close this and focus on improving our test suite in general now!

closed 2024-12-03 by zm711