SpikeInterface/spikeinterface #2178: Implement GFIX preprocessing

alejoe91, 2023-11-07:
From https://billkarsh.github.io/SpikeGLX/help/catgt_tshift/catgt_tshift/:

"-gfix=0.40,0.10,0.02. This requires that a transient artifact must attain a peak amplitude of at least ||0.40 mV||, must reach its peak as least as rapidly as ||0.1mV/sample-tick||, and the transient is over when the amplitude settles back to ||0.02 mV|| or smaller"

WeissShahaf, 2024-05-21:
any news on implementing GFIX?
it's a great tool and the only thing missing for switching from catGT functionality in spikeinterface

alejoe91, 2024-05-21:
@WeissShahaf we are holgin a hackahton at th end of the month. I'll put this in as a project ;)

alejoe91, 2024-05-21:
https://github.com/SpikeInterface/SpikeInterface-Hackathon-Edinburgh-May24/issues/39

alejoe91, 2026-10-26:
This is in since 0.104.0: https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.preprocessing.detect_saturation_periods

closed 2026-10-26 by alejoe91