SpikeInterface/spikeinterface #2180: Alignement in extracting waveform

Kayv-cmb, 2023-11-07:
Hi SpikeInterface,

I am using the waveform extractor to collect the waveform of each spike after a sorting. I am looking for specific change in the waveform and I would need the spike waveform to be aligned on their peak and I have the feeling that they are not. Is i t doable? Or do I have to manually implement that ?
If I understood the get_waveforms() function will get me the waveform (spike, sample, channel) with spike going at max 500 since I specified that and sample should contain 3ms before the peak and 4ms after the peak, knowing that my sampling rate is 30kHz aka 210sample which is what I have, but the spike do not look like they are aligned.
```
we = si.WaveformExtractor.create(raw_rec, sorting, '/nemo/lab/schaefera/working/raw_data/raw_NP/sorted_batch/2023-10-18_15-18-29/recording1/experiment1/waveforms', remove_if_exists=True)
we.set_params(ms_before=3., ms_after=4., max_spikes_per_unit=500)
we.run_extract_waveforms(n_jobs=-1, chunk_size=30000,verbose=True)
print(we)
we.get_waveforms(555)[:,:,:]

i,j,k = np.unravel_index(we.get_waveforms(555).argmax(), we.get_waveforms(555).shape)
we.get_waveforms(555)[i,j,k]

wave_temp = we.get_waveforms(555)[:,:,k]
wave_switch = np.swapaxes(wave_temp,0,1)
plt.plot(wave_switch)
```

For example I am plotting the waveform of this particular cluster on the channel with the biggest waveform is detected which I would expect the closest channel to the units.
However in the plot all of the spike detected on it seems not to be aligned. How can I fix that?
[image]

Is it possible to get some help on to aligned them or what I have is supposed to be normal?

alejoe91, 2023-11-07:
@Kayv-cmb I think that the problem is that you are doing `argmax` instead of  `argmin`!!! I recommend using spikeinterface functions for what you're trying to do:

```
# easier API
we = si.extract_waveforms(raw_rec, sorting, '/nemo/lab/schaefera/working/raw_data/raw_NP/sorted_batch/2023-10-18_15-18-29/recording1/experiment1/waveforms', 
                          remove_if_exists=True, ms_before=3., ms_after=4., max_spikes_per_unit=500, sparse=False,
                          n_jobs=-1, chunk_duration="1s",verbose=True)
print(we)

# get "extremum" channel indices for each waveforms
extremum_channel_indices = si.get_template_extremum_channel(we, outputs="index")
unit_id = 555
single_waveforms = we.get_waveforms(unit_id)[:,:,extremum_channel_indices[unit_id]]
# YOU DON'T NEED THISwave_switch = np.swapaxes(wave_temp,0,1)
plt.plot(single_waveforms.T)
```

Kayv-cmb, 2023-11-08:
@alejoe91 Thanks a lot!!! I didn't know the functions that was doing what I wanted directly, It seems way better with this

closed 2023-11-08 by Kayv-cmb