SpikeInterface/spikeinterface #2181: sw.plot_unit_summary(we, unit_id=...), is plotting only dummy probe

wardfadel, 2023-11-07:
sw.plot_unit_summary(we, unit_id=...), is plotting a dummy probe, even though I created a probe and connected it to the recordings.
it seems that the new probe is not seen by the waveform

probe_x = pi.generate_multi_columns_probe(
    num_columns= 2,
    num_contact_per_column= 128,
    xpitch= 6,
    ypitch= 12,
    y_shift_per_column= np.array([6, 0]),
    contact_shapes= "square",
    contact_shape_params= {"width": 5})

probe_x.set_device_channel_indices(channel_indices)
recording.set_probe(probe_x ,in_place= True)

alejoe91, 2023-11-07:
How are you computing waveforms? Can you `print(recording.get_channel_locations())`?

wardfadel, 2023-11-07:
Kilosort2.5  was used, waveforms were loaded, we = si.load_waveforms(filepath, with_recording = True)
print(recording.get_channel_locations())  gives:
[[   0.    6.]
 [   0.   18.]
 [   0.   30.]
 [   0.   42.]
 [   0.   54.]
 [   0.   66.]
 [   0.   78.]
 [   0.   90.]
 [   0.  102.]
 [   0.  114.]
 [   0.  126.]
 [   0.  138.]
 [   0.  150.]
 [   0.  162.]
 [   0.  174.]
 [   0.  186.]
 [   0.  198.]
 [   0.  210.]
 [   0.  222.]
 [   0.  234.]
 [   0.  246.]
 [   0.  258.]
 [   0.  270.]
 [   0.  282.]
 [   0.  294.]
 [   0.  306.]
 [   0.  318.]
 [   0.  330.]
 [   0.  342.]
 [   0.  354.]
 [   0.  366.]
 [   0.  378.]
 [   0.  390.]
 [   0.  402.]
 [   0.  414.]
 [   0.  426.]
 [   0.  438.]
 [   0.  450.]
 [   0.  462.]
 [   0.  474.]
 [   0.  486.]
 [   0.  498.]
 [   0.  510.]
 [   0.  522.]
 [   0.  534.]
 [   0.  546.]
 [   0.  558.]
 [   0.  570.]
 [   0.  582.]
 [   0.  594.]
 [   0.  606.]
 [   0.  618.]
 [   0.  630.]
 [   0.  642.]
 [   0.  654.]
 [   0.  666.]
 [   0.  678.]
 [   0.  690.]
 [   0.  702.]
 [   0.  714.]
 [   0.  726.]
 [   0.  738.]
 [   0.  750.]
 [   0.  762.]
 [   0.  774.]
 [   0.  786.]
 [   0.  798.]
 [   0.  810.]
 [   0.  822.]
 [   0.  834.]
 [   0.  846.]
 [   0.  858.]
 [   0.  870.]
 [   0.  882.]
 [   0.  894.]
 [   0.  906.]
 [   0.  918.]
 [   0.  930.]
 [   0.  942.]
 [   0.  954.]
 [   0.  966.]
 [   0.  978.]
 [   0.  990.]
 [   0. 1002.]
 [   0. 1014.]
 [   0. 1026.]
 [   0. 1038.]
 [   0. 1050.]
 [   0. 1062.]
 [   0. 1074.]
 [   0. 1086.]
 [   0. 1098.]
 [   0. 1110.]
 [   0. 1122.]
 [   0. 1134.]
 [   0. 1146.]
 [   0. 1158.]
 [   0. 1170.]
 [   0. 1182.]
 [   0. 1194.]
 [   0. 1206.]
 [   0. 1218.]
 [   0. 1230.]
 [   0. 1242.]
 [   0. 1254.]
 [   0. 1266.]
 [   0. 1278.]
 [   0. 1290.]
 [   0. 1302.]
 [   0. 1314.]
 [   0. 1326.]
 [   0. 1338.]
 [   0. 1350.]
 [   0. 1362.]
 [   0. 1374.]
 [   0. 1386.]
 [   0. 1398.]
 [   0. 1410.]
 [   0. 1422.]
 [   0. 1434.]
 [   0. 1446.]
 [   0. 1458.]
 [   0. 1470.]
 [   0. 1482.]
 [   0. 1494.]
 [   0. 1506.]
 [   0. 1518.]
 [   0. 1530.]
 [   6. 1524.]
 [   6. 1512.]
 [   6. 1500.]
 [   6. 1488.]
 [   6. 1476.]
 [   6. 1464.]
 [   6. 1452.]
 [   6. 1440.]
 [   6. 1428.]
 [   6. 1416.]
 [   6. 1404.]
 [   6. 1392.]
 [   6. 1380.]
 [   6. 1368.]
 [   6. 1356.]
 [   6. 1344.]
 [   6. 1332.]
 [   6. 1320.]
 [   6. 1308.]
 [   6. 1296.]
 [   6. 1284.]
 [   6. 1272.]
 [   6. 1260.]
 [   6. 1248.]
 [   6. 1236.]
 [   6. 1224.]
 [   6. 1212.]
 [   6. 1200.]
 [   6. 1188.]
 [   6. 1176.]
 [   6. 1164.]
 [   6. 1152.]
 [   6. 1140.]
 [   6. 1128.]
 [   6. 1116.]
 [   6. 1104.]
 [   6. 1092.]
 [   6. 1080.]
 [   6. 1068.]
 [   6. 1056.]
 [   6. 1044.]
 [   6. 1032.]
 [   6. 1020.]
 [   6. 1008.]
 [   6.  996.]
 [   6.  984.]
 [   6.  972.]
 [   6.  960.]
 [   6.  948.]
 [   6.  936.]
 [   6.  924.]
 [   6.  912.]
 [   6.  900.]
 [   6.  888.]
 [   6.  876.]
 [   6.  864.]
 [   6.  852.]
 [   6.  840.]
 [   6.  828.]
 [   6.  816.]
 [   6.  804.]
 [   6.  792.]
 [   6.  780.]
 [   6.  768.]
 [   6.  756.]
 [   6.  744.]
 [   6.  732.]
 [   6.  720.]
 [   6.  708.]
 [   6.  696.]
 [   6.  684.]
 [   6.  672.]
 [   6.  660.]
 [   6.  648.]
 [   6.  636.]
 [   6.  624.]
 [   6.  612.]
 [   6.  600.]
 [   6.  588.]
 [   6.  576.]
 [   6.  564.]
 [   6.  552.]
 [   6.  540.]
 [   6.  528.]
 [   6.  516.]
 [   6.  504.]
 [   6.  492.]
 [   6.  480.]
 [   6.  468.]
 [   6.  456.]
 [   6.  444.]
 [   6.  432.]
 [   6.  420.]
 [   6.  408.]
 [   6.  396.]
 [   6.  384.]
 [   6.  372.]
 [   6.  360.]
 [   6.  348.]
 [   6.  336.]
 [   6.  324.]
 [   6.  312.]
 [   6.  300.]
 [   6.  288.]
 [   6.  276.]
 [   6.  264.]
 [   6.  252.]
 [   6.  240.]
 [   6.  228.]
 [   6.  216.]
 [   6.  204.]
 [   6.  192.]
 [   6.  180.]
 [   6.  168.]
 [   6.  156.]
 [   6.  144.]
 [   6.  132.]
 [   6.  120.]
 [   6.  108.]
 [   6.   96.]
 [   6.   84.]
 [   6.   72.]
 [   6.   60.]
 [   6.   48.]
 [   6.   36.]
 [   6.   24.]
 [   6.   12.]
 [   6.    0.]]

alejoe91, 2023-11-07:
Do the locations look correct? can you also print: `print(we.get_channel_locations())`?

Can you add a screenshot of the plot too?

alejoe91, 2023-11-07:
then it was not correctly saved when you exported waveforms. Can you try to extract the waveforms again?

wardfadel, 2023-11-07:
Yea, there was no probe attached to the waveform so a dummy one was generated by the Spikeinterface, it should be attached apriori. Is there a solution to connect the same recordings probe to the waveform too (or more generally to replace an original probe)? and thanks a lot for the help

alejoe91, 2023-11-07:
Can you share the full script?

wardfadel, 2023-11-07:
Sure, still not organized, w = sw.plot_unit_summary(we, unit_id=39) is plotting a 1d 256 dummy probe

recording_file = r"filepath.dat"

recording = si.read_binary(recording_file, num_channels=256, sampling_frequency=20000, dtype="int16", time_axis=1)

probe_x = pi.generate_multi_columns_probe(
    num_columns= 2,
    num_contact_per_column= 128,
    xpitch= 6,
    ypitch= 12,
    y_shift_per_column= np.array([6, 0]),
    contact_shapes= "square",
    contact_shape_params= {"width": 5})

channel_indices = np.arange(256)
channel_indices[128:256] = channel_indices[128:256][::-1]
probe_x.set_device_channel_indices(channel_indices)
# print(probe_x.device_channel_indices)
# plot_probe(probe_x, with_contact_id=True, with_device_index=True)

# x=plot_probe(probe_x)
# plt.show()

recording.set_probe(probe_x ,in_place= True)
# recording.set_probegroup(probe_x)

# probe_x= pi.Probe()

recording.annotate(is_filtered=False)
channel_ids = recording.get_channel_ids()
fs = recording.get_sampling_frequency()
num_chan = recording.get_num_channels()
num_segments = recording.get_num_segments()

we = si.load_waveforms(folder='H:\\Spike_Sorting\\......\\waveforms', with_recording = True, )

w = sw.plot_unit_summary(we, unit_id=39)

plt.savefig('my_plot33.png')

# Get the user's home directory
user_home = os.path.expanduser("~")

# Define the "Downloads" directory path
downloads_dir = os.path.join(user_home, "Downloads")

# Set the current working directory to the "Downloads" directory
os.chdir(downloads_dir)

# Create a full path to the PNG file
png_file_name = "my_plot33.png"  # Replace with the actual file name
png_file_pathname = os.path.join(downloads_dir, png_file_name)

# Save the plot as an SVG file
svg_file_name = "my_plot33.svg"
svg_file_pathname = os.path.join(downloads_dir, svg_file_name)
# convert SVG to EMF
save_fig(svg_file_pathname, dpi=600, verbose=True, conv_svg_to_emf=True)

alejoe91, 2023-11-07:
`we = si.load_waveforms(folder='H:\\Spike_Sorting\\......\\waveforms', with_recording = True, )` assumes you had pre-generated the waveforms, maybe with a different recording. You can "overwrite" the recording with:

```
we.set_recording(recording=recording)
```

To point to the new recording. Note that that will change the folder information, so use it with caution. I recommend re-extracting waveforms with a recording with the correct locations.

wardfadel, 2023-11-07:
Thank you Alessio, it worked.  We may want to plot the total units distribution over the probe (based on the max_channel) or something similar.

alejoe91, 2023-11-09:
@wardfadel can you open another issue related to the distribution plot? So that it gets better searchable by other users. Closing this :)

closed 2023-11-09 by alejoe91

wardfadel, 2023-11-10:
it looks like only the shape and dimensions of the probe have been attached to the waveform, but channels indices are not, it is indexed based on the 1d 256 dummy probe. Although I added the channel indices to the probe:

channel_indices = np.arange(256)

# Reversing the order of elements from index 128 to 255
channel_indices[128:256] = channel_indices[128:256][::-1]

probe_x.set_device_channel_indices(channel_indices)

recording.set_probe(probe_x ,in_place= True)

# print(probe_x.device_channel_indices)
plot_probe(probe_x,   with_channel_index = True )
plt.title('probe_x')
plt.show()

alejoe91, 2023-11-10:
@wardfadel can you use 

```
code blocks??
```

Otherwise the formatting is very hard to read...

alejoe91, 2023-11-10:
you should use `with_device_index`, not `with_channel_index` (we removed that ambiguous argument in the latest release)

wardfadel, 2023-11-10:
I already used it with_device_index and the order was correct, but what is with_channel_index then, and is it connected to the channel indices cuz it looks not. More importantly, when I am plotting the unit location 
w = sw.plot_unit_locations(we,plot_all_units = True , with_channel_ids= True,  unit_ids = [96] )
the location of the unit is not based on the new probe dimensions but on a 1d dimension, so all the units are located on the left side of the probe and on a wrong order.
I added some .svg plots:
[image]
[image]
[image]
[image]

alejoe91, 2023-11-10:
@wardfadel can you please edit your comments and use `code blocks`?

wardfadel, 2023-11-10:
```

recording_file = r"H:\\Spike_Sorting\\CTX_mouse\\256ch.....dat"

recording = si.read_binary(recording_file, num_channels=256, sampling_frequency=20000, dtype="int16", time_axis=1)

probe_x = pi.generate_multi_columns_probe(
    num_columns= 2,
    num_contact_per_column= 128,
    xpitch= 6,
    ypitch= 12,
    y_shift_per_column= np.array([6, 0]),
    contact_shapes= "square",
    contact_shape_params= {"width": 5})

channel_indices = np.arange(256)

# Reshape the 1D array into a 2D array with 2 rows
channel_indices_2d = channel_indices.reshape((2, -1), order='F')

# Reverse the order of columns
channel_indices_2d = channel_indices_2d[:, ::-1]

# Reshape the 2D array back into a 1D array
channel_indices = channel_indices_2d.flatten()

# Set the modified channel order to your probe
probe_x.set_device_channel_indices(channel_indices)
probe_x.set_contact_ids(channel_indices)

print(channel_indices)

recording.set_probe(probe_x ,in_place= True)

plot_probe(probe_x, with_device_index= True  ) # with_channel_index = True
plt.title('probe_x')
plt.show()

we = si.load_waveforms(folder='H:\\Spike_Sorting\\CTX_mouse\\256ch\\...\\waveforms', with_recording = True, )
we.set_recording(recording=recording, allow_unfiltered=True)
w = sw.plot_unit_locations(we,plot_all_units = True , with_channel_ids= True,  unit_ids = [96] , plot_legend = True)
w = sw.plot_unit_probe_map(we, channel_ids= [147], unit_ids = [96])
w = sw.plot_unit_summary(we, unit_id=96)

```

wardfadel, 2023-11-10:
```python

recording_file = r"H:\\Spike_Sorting\\CTX_mouse\\256ch.....dat"

recording = si.read_binary(recording_file, num_channels=256, sampling_frequency=20000, dtype="int16", time_axis=1)

probe_x = pi.generate_multi_columns_probe(
    num_columns= 2,
    num_contact_per_column= 128,
    xpitch= 6,
    ypitch= 12,
    y_shift_per_column= np.array([6, 0]),
    contact_shapes= "square",
    contact_shape_params= {"width": 5})

channel_indices = np.arange(256)

# Reshape the 1D array into a 2D array with 2 rows
channel_indices_2d = channel_indices.reshape((2, -1), order='F')

# Reverse the order of columns
channel_indices_2d = channel_indices_2d[:, ::-1]

# Reshape the 2D array back into a 1D array
channel_indices = channel_indices_2d.flatten()

# Set the modified channel order to your probe
probe_x.set_device_channel_indices(channel_indices)
probe_x.set_contact_ids(channel_indices)

print(channel_indices)

recording.set_probe(probe_x ,in_place= True)

plot_probe(probe_x, with_device_index= True  ) # with_channel_index = True
plt.title('probe_x')
plt.show()

we = si.load_waveforms(folder='H:\\Spike_Sorting\\CTX_mouse\\256ch\\...\\waveforms', with_recording = True, )
we.set_recording(recording=recording, allow_unfiltered=True)
w = sw.plot_unit_locations(we,plot_all_units = True , with_channel_ids= True,  unit_ids = [96] , plot_legend = True)
w = sw.plot_unit_probe_map(we, channel_ids= [147], unit_ids = [96])
w = sw.plot_unit_summary(we, unit_id=96)

```

wardfadel, 2023-11-13:
Hi Alessio, should I expect a reply? and have a nice day

alejoe91, 2023-11-13:
Hi, your waveform extractor had wrong locations when it was computed, so all unit locations are also wrong. I recommend re-computing waveforms and othe postprocessing functions using the new "corrected" recording

wardfadel, 2023-11-13:
Thanks a lot, I will check it out

wardfadel, 2023-11-13:
Thanks a lot, I will check it out, but this means that the waveform object couldn't update the attached probe, although when plotting the probe, the shape of the new probe with the correct channel indices where plotted.
Is there a possibility that I can only update the .JSON file in the recording info of the wavefrom folder, cuz I noticed now that the dimension of the probe in the probegroup.json is still 1d 256 channels even after the code update.

alejoe91, 2023-11-13:
The problem is that some postprocessing function use the channel locations and get saved to the folder. If you computed that information with the wrong probe information, the only option is to recompute. Since many functions use channel locations, for example through sparsity, I recommend recomputing everything from scratch using the correct channel locations

wardfadel, 2023-11-16:
Now, all is good, thanks for the help! I re-extracted the waveforms and attached the probe to the recording and then we.set_recording(recording=recording, allow_unfiltered=False) which was still necessary!