SpikeInterface/spikeinterface #2184: SpikeGLX readin problem & question about public recording recommendation

ZoeChen96, 2023-11-09:
Hi,
I asked the question [](https://github.com/SpikeInterface/spikeinterface/issues/2177#issue-1981140312) yesterday about spikeGLX. I have 4 spikeGLX datasets, and I found only 1 of the 4 can't work after I rename the file. However, I tried with two other resources and they both work:
[https://github.com/jenniferColonell/SpikeGLX_Datafile_Tools/blob/main/Python/DemoReadSGLXData/readSGLX.py](url)
[https://github.com/jcouto/pnc_spks](url)
The error is below. Can you have a look?
```
{
	"name": "KeyError",
	"message": "'snsMnMaXaDw'",
	"stack": "---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb Cell 20 line 4
      <a href='vscode-notebook-cell://ssh-remote%2Bchen2/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb#X25sdnNjb2RlLXJlbW90ZQ%3D%3D?line=0'>1</a> import spikeinterface.extractors as se
      <a href='vscode-notebook-cell://ssh-remote%2Bchen2/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb#X25sdnNjb2RlLXJlbW90ZQ%3D%3D?line=2'>3</a> folder = \"/imec/other/macaw/projectdata/neuropixels_datasets/set2\"
----> <a href='vscode-notebook-cell://ssh-remote%2Bchen2/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb#X25sdnNjb2RlLXJlbW90ZQ%3D%3D?line=3'>4</a> rec = se.read_spikeglx(folder_path=folder)
      <a href='vscode-notebook-cell://ssh-remote%2Bchen2/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb#X25sdnNjb2RlLXJlbW90ZQ%3D%3D?line=4'>5</a> a = rec.get_traces()
      <a href='vscode-notebook-cell://ssh-remote%2Bchen2/imec/other/macaw/chen14/pysimulator/maindevelop.ipynb#X25sdnNjb2RlLXJlbW90ZQ%3D%3D?line=5'>6</a> print(a)

File /imec/other/macaw/chen14/anaconda43/envs/tfqu1016/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/spikeglx.py:51, in SpikeGLXRecordingExtractor.__init__(self, folder_path, load_sync_channel, stream_id, stream_name, all_annotations)
     49 def __init__(self, folder_path, load_sync_channel=False, stream_id=None, stream_name=None, all_annotations=False):
     50     neo_kwargs = self.map_to_neo_kwargs(folder_path, load_sync_channel=load_sync_channel)
---> 51     NeoBaseRecordingExtractor.__init__(
     52         self, stream_id=stream_id, stream_name=stream_name, all_annotations=all_annotations, **neo_kwargs
     53     )
     55     # open the corresponding stream probe for LF and AP
     56     # if load_sync_channel=False
     57     if \"nidq\" not in self.stream_id and not load_sync_channel:

File /imec/other/macaw/chen14/anaconda43/envs/tfqu1016/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:185, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    156 def __init__(
    157     self,
    158     stream_id: Optional[str] = None,
   (...)
    163     **neo_kwargs: Dict[str, Any],
    164 ) -> None:
    165     \"\"\"
    166     Initialize a NeoBaseRecordingExtractor instance.
    167 
   (...)
    182 
    183     \"\"\"
--> 185     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    187     kwargs = dict(all_annotations=all_annotations)
    188     if block_index is not None:

File /imec/other/macaw/chen14/anaconda43/envs/tfqu1016/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:25, in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     23 def __init__(self, block_index, **neo_kwargs):
     24     if not hasattr(self, \"neo_reader\"):  # Avoid double initialization
---> 25         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     27     if self.neo_reader.block_count() > 1 and block_index is None:
     28         raise Exception(
     29             \"This dataset is multi-block. Spikeinterface can load one block at a time. \"
     30             \"Use 'block_index' to select the block to be loaded.\"
     31         )

File /imec/other/macaw/chen14/anaconda43/envs/tfqu1016/lib/python3.8/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:64, in _NeoBaseExtractor.get_neo_io_reader(cls, raw_class, **neo_kwargs)
     62 neoIOclass = getattr(rawio_module, raw_class)
     63 neo_reader = neoIOclass(**neo_kwargs)
---> 64 neo_reader.parse_header()
     66 return neo_reader

File ~/.local/lib/python3.8/site-packages/neo/rawio/baserawio.py:179, in BaseRawIO.parse_header(self)
    166 def parse_header(self):
    167     \"\"\"
    168     This must parse the file header to get all stuff for fast use later on.
    169 
   (...)
    177 
    178     \"\"\"
--> 179     self._parse_header()
    180     self._check_stream_signal_channel_characteristics()

File ~/.local/lib/python3.8/site-packages/neo/rawio/spikeglxrawio.py:81, in SpikeGLXRawIO._parse_header(self)
     80 def _parse_header(self):
---> 81     self.signals_info_list = scan_files(self.dirname)
     83     # sort stream_name by higher sampling rate first
     84     srates = {info['stream_name']: info['sampling_rate'] for info in self.signals_info_list}

File ~/.local/lib/python3.8/site-packages/neo/rawio/spikeglxrawio.py:254, in scan_files(dirname)
    252 if meta_filename.exists() and bin_filename.exists():
    253     meta = read_meta_file(meta_filename)
--> 254     info = extract_stream_info(meta_filename, meta)
    256     info['meta_file'] = str(meta_filename)
    257     info['bin_file'] = str(bin_filename)

File ~/.local/lib/python3.8/site-packages/neo/rawio/spikeglxrawio.py:415, in extract_stream_info(meta_file, meta)
    412 channel_gains = np.ones(num_chan)
    414 # there are differents kinds of channels with different gain values
--> 415 mn, ma, xa, dw = [int(e) for e in meta['snsMnMaXaDw'].split(sep=',')]
    416 per_channel_gain = np.ones(num_chan, dtype='float64')
    417 per_channel_gain[0:mn] = 1. / float(meta['niMNGain'])

KeyError: 'snsMnMaXaDw'"
}
```

alejoe91, 2023-11-15:
Hi, can you share the `.meta` files?

Regarding NP2 data with ground-truth, I'm not sure there are any. Are you looking for simulated data? In that case, you can easily generate some with [MEArec](https://github.com/SpikeInterface/MEArec)

ZoeChen96, 2023-11-15:
Hi Alessio, yes sure, here it is: 

> acqApLfSy=384,0,1
> appVersion=20230323
> fileCreateTime=2020-01-07T11:17:23
> fileName=G:/bernardo_tacca_sample/NP2.0/SC035_010720_g0_t0.exported.imec1.ap.bin
> fileSHA1=FF17CB550A95347677C7C414A4140D5D995F96B2
> fileSizeBytes=13859737430
> fileTimeSecs=599.9999833330181
> firstSample=84968814
> gateMode=Immediate
> imAiRangeMax=0.5
> imAiRangeMin=-0.5
> imCalibrated=true
> imDatApi=2.11
> imDatBs_fw=1.1.128
> imDatBsc_fw=2.0.19
> imDatBsc_hw=2.1
> imDatBsc_pn=NP2_QBSC_00
> imDatBsc_sn=259
> imDatFx_hw=1.7
> imDatFx_pn=NPM_FLEX_0
> imDatHs_fw=0.0
> imDatHs_pn=000000000
> imDatHs_sn=1440
> imDatPrb_dock=1
> imDatPrb_pn=PRB2_1_2_0640_0
> imDatPrb_port=2
> imDatPrb_slot=3
> imDatPrb_sn=19011111614
> imDatPrb_type=21
> imLEDEnable=false
> imMaxInt=8192
> imRoFile=C:/SpikeGLX/Release_v20190919-phase20/IMRO_Tables/NPtype21_bank0_ref1.imro
> imSampRate=29999.4325
> imStdby=
> imTrgRising=false
> imTrgSource=0
> nSavedChans=385
> snsApLfSy=384,0,1
> snsSaveChanSubset=0:384
> syncImInputSlot=3
> syncSourceIdx=2
> syncSourcePeriod=1
> trigMode=Immediate
> typeImEnabled=3
> typeNiEnabled=1
> typeThis=imec
> userNotes=Imec0: left ALM; Imec1: left Mdibrain; Imec2: right Medulla\nSingle shank for Midbrain, 4shank probes for ALM and Medulla
> ~imroTbl=(21,384)(0 1 1 0)(1 1 1 1)(2 1 1 2)(3 1 1 3)(4 1 1 4)(5 1 1 5)(6 1 1 6)(7 1 1 7)(8 1 1 8)(9 1 1 9)(10 1 1 10)(11 1 1 11)(12 1 1 12)(13 1 1 13)(14 1 1 14)(15 1 1 15)(16 1 1 16)(17 1 1 17)(18 1 1 18)(19 1 1 19)(20 1 1 20)(21 1 1 21)(22 1 1 22)(23 1 1 23)(24 1 1 24)(25 1 1 25)(26 1 1 26)(27 1 1 27)(28 1 1 28)(29 1 1 29)(30 1 1 30)(31 1 1 31)(32 1 1 32)(33 1 1 33)(34 1 1 34)(35 1 1 35)(36 1 1 36)(37 1 1 37)(38 1 1 38)(39 1 1 39)(40 1 1 40)(41 1 1 41)(42 1 1 42)(43 1 1 43)(44 1 1 44)(45 1 1 45)(46 1 1 46)(47 1 1 47)(48 1 1 48)(49 1 1 49)(50 1 1 50)(51 1 1 51)(52 1 1 52)(53 1 1 53)(54 1 1 54)(55 1 1 55)(56 1 1 56)(57 1 1 57)(58 1 1 58)(59 1 1 59)(60 1 1 60)(61 1 1 61)(62 1 1 62)(63 1 1 63)(64 1 1 64)(65 1 1 65)(66 1 1 66)(67 1 1 67)(68 1 1 68)(69 1 1 69)(70 1 1 70)(71 1 1 71)(72 1 1 72)(73 1 1 73)(74 1 1 74)(75 1 1 75)(76 1 1 76)(77 1 1 77)(78 1 1 78)(79 1 1 79)(80 1 1 80)(81 1 1 81)(82 1 1 82)(83 1 1 83)(84 1 1 84)(85 1 1 85)(86 1 1 86)(87 1 1 87)(88 1 1 88)(89 1 1 89)(90 1 1 90)(91 1 1 91)(92 1 1 92)(93 1 1 93)(94 1 1 94)(95 1 1 95)(96 1 1 96)(97 1 1 97)(98 1 1 98)(99 1 1 99)(100 1 1 100)(101 1 1 101)(102 1 1 102)(103 1 1 103)(104 1 1 104)(105 1 1 105)(106 1 1 106)(107 1 1 107)(108 1 1 108)(109 1 1 109)(110 1 1 110)(111 1 1 111)(112 1 1 112)(113 1 1 113)(114 1 1 114)(115 1 1 115)(116 1 1 116)(117 1 1 117)(118 1 1 118)(119 1 1 119)(120 1 1 120)(121 1 1 121)(122 1 1 122)(123 1 1 123)(124 1 1 124)(125 1 1 125)(126 1 1 126)(127 1 1 127)(128 1 1 128)(129 1 1 129)(130 1 1 130)(131 1 1 131)(132 1 1 132)(133 1 1 133)(134 1 1 134)(135 1 1 135)(136 1 1 136)(137 1 1 137)(138 1 1 138)(139 1 1 139)(140 1 1 140)(141 1 1 141)(142 1 1 142)(143 1 1 143)(144 1 1 144)(145 1 1 145)(146 1 1 146)(147 1 1 147)(148 1 1 148)(149 1 1 149)(150 1 1 150)(151 1 1 151)(152 1 1 152)(153 1 1 153)(154 1 1 154)(155 1 1 155)(156 1 1 156)(157 1 1 157)(158 1 1 158)(159 1 1 159)(160 1 1 160)(161 1 1 161)(162 1 1 162)(163 1 1 163)(164 1 1 164)(165 1 1 165)(166 1 1 166)(167 1 1 167)(168 1 1 168)(169 1 1 169)(170 1 1 170)(171 1 1 171)(172 1 1 172)(173 1 1 173)(174 1 1 174)(175 1 1 175)(176 1 1 176)(177 1 1 177)(178 1 1 178)(179 1 1 179)(180 1 1 180)(181 1 1 181)(182 1 1 182)(183 1 1 183)(184 1 1 184)(185 1 1 185)(186 1 1 186)(187 1 1 187)(188 1 1 188)(189 1 1 189)(190 1 1 190)(191 1 1 191)(192 1 1 192)(193 1 1 193)(194 1 1 194)(195 1 1 195)(196 1 1 196)(197 1 1 197)(198 1 1 198)(199 1 1 199)(200 1 1 200)(201 1 1 201)(202 1 1 202)(203 1 1 203)(204 1 1 204)(205 1 1 205)(206 1 1 206)(207 1 1 207)(208 1 1 208)(209 1 1 209)(210 1 1 210)(211 1 1 211)(212 1 1 212)(213 1 1 213)(214 1 1 214)(215 1 1 215)(216 1 1 216)(217 1 1 217)(218 1 1 218)(219 1 1 219)(220 1 1 220)(221 1 1 221)(222 1 1 222)(223 1 1 223)(224 1 1 224)(225 1 1 225)(226 1 1 226)(227 1 1 227)(228 1 1 228)(229 1 1 229)(230 1 1 230)(231 1 1 231)(232 1 1 232)(233 1 1 233)(234 1 1 234)(235 1 1 235)(236 1 1 236)(237 1 1 237)(238 1 1 238)(239 1 1 239)(240 1 1 240)(241 1 1 241)(242 1 1 242)(243 1 1 243)(244 1 1 244)(245 1 1 245)(246 1 1 246)(247 1 1 247)(248 1 1 248)(249 1 1 249)(250 1 1 250)(251 1 1 251)(252 1 1 252)(253 1 1 253)(254 1 1 254)(255 1 1 255)(256 1 1 256)(257 1 1 257)(258 1 1 258)(259 1 1 259)(260 1 1 260)(261 1 1 261)(262 1 1 262)(263 1 1 263)(264 1 1 264)(265 1 1 265)(266 1 1 266)(267 1 1 267)(268 1 1 268)(269 1 1 269)(270 1 1 270)(271 1 1 271)(272 1 1 272)(273 1 1 273)(274 1 1 274)(275 1 1 275)(276 1 1 276)(277 1 1 277)(278 1 1 278)(279 1 1 279)(280 1 1 280)(281 1 1 281)(282 1 1 282)(283 1 1 283)(284 1 1 284)(285 1 1 285)(286 1 1 286)(287 1 1 287)(288 1 1 288)(289 1 1 289)(290 1 1 290)(291 1 1 291)(292 1 1 292)(293 1 1 293)(294 1 1 294)(295 1 1 295)(296 1 1 296)(297 1 1 297)(298 1 1 298)(299 1 1 299)(300 1 1 300)(301 1 1 301)(302 1 1 302)(303 1 1 303)(304 1 1 304)(305 1 1 305)(306 1 1 306)(307 1 1 307)(308 1 1 308)(309 1 1 309)(310 1 1 310)(311 1 1 311)(312 1 1 312)(313 1 1 313)(314 1 1 314)(315 1 1 315)(316 1 1 316)(317 1 1 317)(318 1 1 318)(319 1 1 319)(320 1 1 320)(321 1 1 321)(322 1 1 322)(323 1 1 323)(324 1 1 324)(325 1 1 325)(326 1 1 326)(327 1 1 327)(328 1 1 328)(329 1 1 329)(330 1 1 330)(331 1 1 331)(332 1 1 332)(333 1 1 333)(334 1 1 334)(335 1 1 335)(336 1 1 336)(337 1 1 337)(338 1 1 338)(339 1 1  [... cut ...] 69:1)(0:1:169:1)(0:0:170:1)(0:1:170:1)(0:0:171:1)(0:1:171:1)(0:0:172:1)(0:1:172:1)(0:0:173:1)(0:1:173:1)(0:0:174:1)(0:1:174:1)(0:0:175:1)(0:1:175:1)(0:0:176:1)(0:1:176:1)(0:0:177:1)(0:1:177:1)(0:0:178:1)(0:1:178:1)(0:0:179:1)(0:1:179:1)(0:0:180:1)(0:1:180:1)(0:0:181:1)(0:1:181:1)(0:0:182:1)(0:1:182:1)(0:0:183:1)(0:1:183:1)(0:0:184:1)(0:1:184:1)(0:0:185:1)(0:1:185:1)(0:0:186:1)(0:1:186:1)(0:0:187:1)(0:1:187:1)(0:0:188:1)(0:1:188:1)(0:0:189:1)(0:1:189:1)(0:0:190:1)(0:1:190:1)(0:0:191:1)(0:1:191:1)

ZoeChen96, 2023-11-15:
> Hi, can you share the `.meta` files?
> 
> Regarding NP2 data with ground-truth, I'm not sure there are any. Are you looking for simulated data? In that case, you can easily generate some with [MEArec](https://github.com/SpikeInterface/MEArec)

For the datasets, I don't want simulated data, I already know and generated some simulated datasets using MEArec, thanks for mentioning that!
Instead, I want some real recordings with curated results (which can be used as ground truth)

alejoe91, 2023-11-15:
What is the name of this `.meta` file? do you have more than one in the folder?

alejoe91, 2023-11-15:
Note that for how the reader is structured, the `meta` file must contain the `imec` substring in its name. Can you try to add it if it's not there?

ZoeChen96, 2023-11-15:
The name is `SC035_010720_g0_t0.exported.imec1.ap.meta`, I think it is with imec substring. There is only this meta here

alejoe91, 2023-11-15:
right, you need to remove `.exported`

ZoeChen96, 2023-11-15:
Ok, I will try! I don't have permission to change the name, it will take some time

ZoeChen96, 2023-11-15:
It works, thank you!

closed 2023-11-15 by ZoeChen96

alejoe91, 2023-11-15:
Actually, it should work with your file name..not sure why it's not working! @samuelgarcia any idea?

samuelgarcia, 2023-11-15:
I think we have to change rules using regexp instead of spliting on dot for getting fields like (ap vs lf) in spikeglx.