SpikeInterface/spikeinterface #2185: changing color of the channels in plot_timeseries (basic)

kshtjkumar, 2023-11-09:
hi, I am trying to change the colours of the time series plot, is it possible to give a particular colour  to a particular channel id ?

recording_rhs.set_channel_groups(channel_ids=recording_rhs.get_channel_ids(), groups=['red', 'green', 'blue', 'black'])
w2 = sw.plot_timeseries({"Raw Data": recording_rhs},color_groups=True, time_range=(120, 420),clim = (-50, 50),  order_channel_by_depth= False, show_channel_ids=True)#,backend='ipywidgets' )

something like this, but the channel colors are being very different !

kshtjkumar, 2023-11-09:
fig,ax=plt.subplots(4,1,sharex=True)
clr =  ['C0', 'C1', 'C2', 'C3']
plt.subplots_adjust(hspace=0.001)
for i in [0,1,2,3] :
    sw.plot_timeseries({"":recording_rhs}, color = clr[i],time_range=(120, 420),channel_ids=recording_rhs.channel_ids[i:i+1],ax = ax[i],show_channel_ids=True)
    if i==0:
        ax[i].spines[['bottom']].set_visible(False)
        ax[i].set_xticks([])
        ax[i].get_legend().remove()
#         ax[i].set_xlabel('')
#         ax[i].set_xticklabels([])
    elif i in [1,2]:
        ax[i].spines[['top','bottom']].set_visible(False)
        ax[i].set_xticks([])
        ax[i].get_legend().remove()
    elif i==3:
        ax[i].spines[['top']].set_visible(False)
        ax[i].get_legend().remove()

kshtjkumar, 2023-11-09:
[image]

alejoe91, 2023-11-10:
Yeah we should add an option to change the colors. For the values, you can use `return_scaled=True` to plot in uV. Currently plotting in mV is not supported.