SpikeInterface/spikeinterface #2187: Problem with Sorting on one Drive (c:) and outputting results to Another Drive (d:) on Windows

zm711, 2023-11-09:
## Issue

As discussed in #1891, when using `run_sorter`, it errors if two different drives are specified on Windows. (I'm stealing Heberto's example from the discussion): So saving from one drive to another works fine (as tested).

```python
from spikeinterface.core.datasets import download_dataset
from spikeinterface.extractors.neoextractors.mearec import (
    MEArecRecordingExtractor
)
from spikeinterface.core import load_extractor

local_file_path = download_dataset(remote_path="mearec/mearec_test_10s.h5")
recording = MEArecRecordingExtractor(local_file_path)

file_path = "C:\\Users\\heberto\\spikeinterface_datasets\\ephy_testing_data\\mearec\\mearec_test_10s.h5"
recording = MEArecRecordingExtractor(file_path=file_path)

MEArecRecordingExtractor: 32 channels - 32.0kHz - 1 segments - 320,000 samples - 10.00s 
                          float32 dtype - 39.06 MiB
  file_path: [C:\Users\heberto\spikeinterface_datasets\ephy_testing_data\mearec\mearec_test_10s.h5](file:///C:/Users/heberto/spikeinterface_datasets/ephy_testing_data/mearec/mearec_test_10s.h5)

folder_path = Path("D:\\folder_to_write_stuff")
folder_path.is_dir(), folder_path.anchor, folder_path.root

recording.save(folder=folder_path, overwrite=True)

write_binary_recording with n_jobs = 1 and chunk_size = 32000
write_binary_recording: 100%|##########| 10/10 [00:02<00:00,  3.67it/s]
BinaryFolderRecording: 32 channels - 32.0kHz - 1 segments - 320,000 samples - 10.00s 
                       float32 dtype - 39.06 MiB

import os

folder_path = "D:\\folder_to_write_stuff"
print(os.listdir(folder_path))

['provenance.json', 'probe.json', 'properties', 'traces_cached_seg0.raw', 'binary.json', 'si_folder.json']
```

But if instead we try to run sorter with the same dataset we get:

```python

  Cell In[6], line 1
    ss.run_sorter('tridesclous', recording, output_folder = folder_path)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\runsorter.py:147 in run_sorter
    return run_sorter_local(**common_kwargs)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\runsorter.py:170 in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\sorters\basesorter.py:141 in initialize_folder
    recording.dump(rec_file, relative_to=output_folder)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\base.py:569 in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\base.py:602 in dump_to_json
    dump_dict = self.to_dict(

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\base.py:406 in to_dict
    dump_dict = _make_paths_relative(dump_dict, relative_to)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\base.py:983 in _make_paths_relative
    return recursive_path_modifier(d, func, target="path", copy=True)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\core_tools.py:824 in recursive_path_modifier
    recursive_path_modifier(kwargs, func, copy=False)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\core_tools.py:847 in recursive_path_modifier
    dc[k] = func(v)

  File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\base.py:982 in <lambda>
    func = lambda p: os.path.relpath(str(p), start=relative)

  File ~\anaconda3\envs\spykes\Lib\ntpath.py:754 in relpath
    raise ValueError("path is on mount %r, start on mount %r" % (

ValueError: path is on mount 'C:', start on mount 'D:'
```
I will note that the `folder_to_write_stuff` and the `sorter_output` folders are created on the `D:` drive so error is after the folder generation.

## Solution

Not sure, but can move our new thread to discuss here. Maybe this is directly linked #2186 and both can be fixed by a similar global solution. As @h-mayorquin said, the `rec.save` works fine, so the resolving machinery between local drives isn't the main problem (as it is in #2186, which fails save and run_sorter), but maybe the order of the operations is contributing. 

## Notes

I would say this is a bit higher priority since many people would want to sort on C: where they have a fast drive and then move to either a storage drive or a portable drive. 

I marked as bug since from my (windows-centric) perspective this seems like expected behavior to just work (I know it's not trivial to implement), but if you think it falls more in the enhancement, I could also see it labeled that way instead if the guarantee is to work on the (`C:`) for Windows only and everything else is bonus.

h-mayorquin, 2023-11-10:
Thanks for writing such a detailed description, Zach. I have a question here, is this a vital feature or a convenience one?

Like, can't the same results be achieved by just having the data and the output in the same drive? I can see how it is convenient but I would like to understand if there is a situation where is unavoidable.

zm711, 2023-11-10:
It's mostly just that the drive fills up and there aren't guard rails around many SI functions (see #1063), which recently happened to me because I couldn't save my files from c -> d (I was doing a test of supporting with multiple sorters to do comparison, which is a major benefit of SI, so just ran them one after another). Same issue with wavefrom extraction which #1063 directly mentions.

So yes this is convenience in some sense. From a library perspective I could see calling this an enhancement rather than a bug, but as a user it feels pretty terrible to constantly be searching for space on my C drive to make sure I can save a file (can I test one sorter, two sorters, five sorters, hard to know how many temp files etc they will generate). Then doing something and having it crash anyway because I mis-estimated space needs. (I'm actively working on approval for Linux for myself so I can avoid these issues, but until that point...) 

So if the library wants to fully support Windows I would say vital, Windows OS uses the drive system rather than the root folder to tree system, so getting this error feels like Windows just isn't as supported. At minimum if this is low priority I would like a better windows error message so that knew users know that SI currently requires Windows users to sort and save on the same drive.

Does that make sense? I'm torn from a user vs developer perspective...

(I didn't open the 3rd docker issue because I'm not sure how to word it or give the correct details since I do the sorting locally).

zm711, 2023-11-10:
One other tiny point that is a bit of a sticking point related to drive filling up is that Kilosort requires a binary file written so you end up with two giant files in the same folder --> the original data file and the nearly equally sized giant binary file. So you eat up a lot of space that way. (This isn't SI's fault, but it makes the space issue more pressing if you want to use any of the KS's as a comparison sorter.

zm711, 2023-12-05:
I'm going to close this for now and update with a newer issue so that people don't stumble upon this old problem if they run into the pickling error seen in #2279.