SpikeInterface/spikeinterface #2188: Issue Loading Waveforms with Recording Attached

litzj01, 2023-11-09:
I'm having issues loading my waveforms with my recording attached.  A lot of the widgets require this and I want to view the data in the SpikeInterface gui. If I try to set the load waveforms 'with_recording=True', I get a long error message that the recording cannot be loaded.  Any suggestions?

I'm using the following code to load my data:

recording = se.read_biocam('/Users/litzj/092223_Raw/92223_GreenChirpND1_01.brw') 

sorting = ss.read_sorter_folder(output_folder='/Users/litzj/spikeinterface/herdingspikes_output', register_recording=True, sorting_info=True) 

we = si.load_waveforms(sorting=sorting, folder="/Users/litzj/092223_Raw/preprocessed/82_wt_100Hz_filter/waveforms_folder", with_recording=False) 

we.set_recording(recording, allow_unfiltered=True) 

(As a note, the path to the folder that my waveforms are stored in suggests that the data was filtered, but it actually wasn't. The dataset is too large for my computer to store the filtered object.)

samuelgarcia, 2023-11-10:
Hi.
The solution you are doing is correct to load waveforms when you do not have anymore the recording.
I do not not understand the error you have.

Note:
  * many diplsy using widgets module do not need the recording for plot
  * spikeinterface-gui need the recording at the moement but I can make a mode without recording if this is needed.

alejoe91, 2023-11-10:
A note on your note: you can also do:
```
recording_f = spre.bandpass_filter(recording)

...

we.set_recording(recording_f) 
```

This will NOT save the filtered data, but filter on the fly!

litzj01, 2023-11-10:
Thanks for your reply! I know I can still utilize a lot of the widgets, but I wanted to be able to analyze the waveforms more in-depth. My issue was that I was getting an error trying to use the waveform post-processing functions that said the recording wasn't attached even after I had run that code. This had been an issue for days, but of course today it suddenly seemed to work in my notebook. 
 
It took a pretty long time to open the GUI (about 30 min) and I got the following note, if that matters: "C:\Users\litzj\AppData\Local\Temp\1\ipykernel_25024\2730146880.py:6: DeprecationWarning: 'exec_' will be removed in the future. Use 'exec' instead. app.exec_()" 

I saw someone else was having a similar issue with large datasets loading, but I didn't see that it was resolved. I tried to run the command "sigui /Users/litzj/092223_Raw/preprocessed/82_wt_100Hz_filter/waveforms_folder" from the terminal, but then got the same error as before "Exception: The recording could not be loaded. You can use the `with_recording=False` argument".

My original data file is about 9GB and the waveforms folder is about 4.5GB and there are 122 units detected.  This is one of my smallest datasets though (most are ~90GB and have up to 8,000+ units).  Do you think the GUI would be able to handle a larger dataset?

samuelgarcia, 2023-11-13:
Opening the Gui should not be so long!!!!
The only computation which is done is the noise level estimation at the opening when this is not already done in the waveform extractor. the ssecond openning should be really faster.

Thanks for the feeback I will remove the DeprecationWarning.

4.5GB is pretty big for waveforms did you made a sparse extraction ?

Normally the GUI is doing lazy loading so this is done for big dataset. I must have a look do you have the latest version ?

litzj01, 2023-11-13:
I didn't specify the sparsity when extracting the waveforms, so the default sparsity should have been used. I am using a biocam chip with 4096 channels which might contribute to the large size. I should have the latest version of SpikeInterface as I installed it in mid-October from the source install. The GUI also was quite laggy.

zm711, 2024-05-10:
@litzj01, we have gone through some bigger refactoring and updates. If you want to install from main and try the gui from main let us know if the issues still persist.

zm711, 2024-06-19:
I'll close this now, but reopen or open a new issue if something new has come up.

closed 2024-06-19 by zm711