SpikeInterface/spikeinterface #2200: Issue exporting to phy

litzj01, 2023-11-13:
I'm having issues with exporting my data to phy. Every time I run the line `phy template-gui params.py `, I get the error `ValueError: attempt to get argmax of an empty sequence`. I raised the issue with Phy (see the thread [here](https://github.com/cortex-lab/phy/issues/1233)) and they think it might be a problem with Spike Interface.  It seems to be similar to issue #1945, but I don't have any mono-channel templates (my smallest template is (58,8)). 
Any help would be appreciated!

Here is the full error:
`16:40:13.138 [D] init:68 Start capturing exceptions.
16:40:13.155 [D] model:619 Loading spike clusters.
16:40:13.230 [D] model:569 No channel shank file found.
16:40:13.230 [D] model:692 Loading templates.
16:40:13.239 [D] model:720 Templates are sparse.
16:40:13.240 [W] model:667 Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
16:40:13.240 [D] model:730 Loading the whitening matrix.
16:40:13.241 [D] model:434 Whitening matrix file not found.
16:40:13.246 [D] model:737 Loading the inverse of the whitening matrix.
16:40:13.331 [D] model:766 Loading features.
16:40:13.339 [D] model:781 Features are sparse.
16:40:13.340 [D] model:803 Loading template features.
16:40:13.342 [D] model:504 Load cluster_channel_group.tsv.
16:40:13.345 [D] model:504 Load cluster_group.tsv.
16:40:13.347 [D] model:504 Load cluster_si_unit_ids.tsv.
16:40:13.866 [D] context:100 Initialize joblib cache dir at C:\Users\litzj\092223_Raw\preprocessed\82_wt_100Hz_filter\phy_folder_for_hdspikes\.phy.
16:40:13.867 [D] context:101 Reducing the size of the cache if needed.
16:40:13.874 [D] base:102 Add filter high_pass.
16:40:13.875 [D] config:31 Load config file C:\Users\litzj\.phy\phy_config.py.
16:40:13.876 [D] plugin:145 Loading 0 plugins.
16:40:13.879 [D] context:209 The file C:\Users\litzj\092223_Raw\preprocessed\82_wt_100Hz_filter\phy_folder_for_hdspikes\.phy\new_cluster_id.pkl doesn't exist.
16:40:13.891 [D] context:185 Save data to C:\Users\litzj\092223_Raw\preprocessed\82_wt_100Hz_filter\phy_folder_for_hdspikes\.phy\spikes_per_cluster.pkl.
16:40:13.915 [D] gui:463 Creating GUI.
16:40:13.942 [D] state:46 Load C:\Users\litzj.phy\TemplateGUI\state.json for GUIState.
16:40:13.988 [E] init:62 An error has occurred (ValueError): attempt to get argmax of an empty sequence
Traceback (most recent call last):
File "", line 198, in _run_module_as_main
File "", line 88, in run_code
File "C:\Users\litzj\miniconda\envs\phy2\Scripts\phy.exe_main.py", line 7, in
sys.exit(phycli())
^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\core.py", line 1157, in call
return self.main(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\core.py", line 1078, in main
rv = self.invoke(ctx)
^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\core.py", line 1688, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\core.py", line 1434, in invoke
return ctx.invoke(self.callback, **ctx.params)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\core.py", line 783, in invoke
return _callback(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\click\decorators.py", line 33, in new_func
return f(get_current_context(), *args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps_init.py", line 159, in cli_template_gui
template_gui(params_path, **kwargs)
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 218, in template_gui
gui = controller.create_gui()
^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps\base.py", line 1700, in create_gui
self.supervisor.attach(gui)
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\cluster\supervisor.py", line 950, in attach
self._create_views(
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\cluster\supervisor.py", line 765, in _create_views
gui, data=self.cluster_info, columns=self.columns, sort=sort)
^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\cluster\supervisor.py", line 925, in cluster_info
return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\cluster\supervisor.py", line 925, in
return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\cluster\supervisor.py", line 750, in get_cluster_info
out[key] = func(cluster_id)
^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps\base.py", line 1194, in get_best_channel_label
return self._get_channel_labels([self.get_best_channel(cluster_id)])[0]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\utils\context.py", line 154, in memcached
out = f(*args, **kwargs)
^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps\base.py", line 1188, in get_best_channel
channel_ids = self.get_best_channels(cluster_id)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\utils\context.py", line 154, in memcached
out = f(*args, **kwargs)
^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 151, in get_best_channels
template = self.model.get_template(template_id)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phylib\io\model.py", line 957, in get_template
return self._get_template_sparse(template_id, unwhiten=unwhiten)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\phylib\io\model.py", line 924, in _get_template_sparse
best_channel = channel_ids[np.argmax(amplitude)]
^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\numpy\core\fromnumeric.py", line 1229, in argmax
return _wrapfunc(a, 'argmax', axis=axis, out=out, **kwds)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\litzj\miniconda\envs\phy2\Lib\site-packages\numpy\core\fromnumeric.py", line 59, in _wrapfunc
return bound(*args, **kwds)
^^^^^^^^^^^^^^^^^^^^
ValueError: attempt to get argmax of an empty sequence`

Here is the `plot_probe`:
[image]

@zm711

litzj01, 2023-11-17:
I've have since tried a couple of new things since my previous post, but am still not having success. 
I realized the default sparsity might not fit my probe very well, so I re-extracted the waveforms with the following code:
```
we_sparse = si.extract_waveforms(recording, sorting, "/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/" + 'waveforms_sparse', overwrite=True, allow_unfiltered=True, sparse=True, method="radius", radius_um=70, **global_job_kwargs) 

sparsity = si.compute_sparsity(we_sparse, method="radius", radius_um=70)

we_sparse2 = we_sparse.save(we_sparse, sparsity=sparsity, folder="waveforms_sparse2")
```

Then I run the following codes to export to phy:
```
amplitudes = spost.compute_spike_amplitudes(we_sparse) 
unit_locations = spost.compute_unit_locations(we_sparse) 
spike_locations = spost.compute_spike_locations(we_sparse) 
correlograms, bins = spost.compute_correlograms(we_sparse) 
similarity = spost.compute_template_similarity(we_sparse)

sexp.export_to_phy(we_sparse, "/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/" + 'phy_folder_for_hdspikes', verbose=True, sparsity=we_sparse2) 
```

The export seems to be successful, but phy is still unable to launch. Here is the new error I get after running `phy template-gui params.py --debug`:

`12:05:14.520 [D] __init__:68          Start capturing exceptions.
12:05:14.749 [D] model:619            Loading spike clusters.
12:05:14.959 [D] model:569            No channel shank file found.
12:05:14.960 [D] model:692            Loading templates.
12:05:14.996 [D] model:720            Templates are sparse.
12:05:14.997 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
12:05:14.997 [D] model:730            Loading the whitening matrix.
12:05:14.998 [D] model:434            Whitening matrix file not found.
12:05:15.007 [D] model:737            Loading the inverse of the whitening matrix.
12:05:15.390 [D] model:766            Loading features.
12:05:15.433 [D] model:781            Features are sparse.
12:05:15.433 [D] model:803            Loading template features.
12:05:15.436 [D] model:504            Load `cluster_group.tsv`.
12:05:15.444 [D] model:504            Load `cluster_channel_group.tsv`.
12:05:15.448 [D] model:504            Load `cluster_si_unit_ids.tsv`.
qt.qpa.xcb: X server does not support XInput 2
failed to get the current screen resources
qt.qpa.gl: QXcbConnection: Failed to initialize GLX
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib64/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib64/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib64/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib64/dri)
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
12:05:29.275 [D] context:100          Initialize joblib cache dir at `/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/phy_folder_for_hdspikes/.phy`.
12:05:29.284 [D] context:101          Reducing the size of the cache if needed.
12:05:29.340 [D] base:102             Add filter `high_pass`.
12:05:29.342 [D] config:31            Load config file `/home/users/litzj/.phy/phy_config.py`.
12:05:29.344 [D] plugin:145           Loading 0 plugins.
12:05:29.350 [D] context:209          The file `/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/phy_folder_for_hdspikes/.phy/new_cluster_id.pkl` doesn't exist.
12:05:29.489 [D] context:185          Save data to `/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/phy_folder_for_hdspikes/.phy/spikes_per_cluster.pkl`.
12:05:29.587 [D] gui:463              Creating GUI.
12:05:30.386 [D] state:46             Load /home/users/litzj/.phy/TemplateGUI/state.json for GUIState.
12:05:35.111 [D] gui:718              Add view ClusterView to GUI.
12:05:36.093 [D] gui:718              Add view SimilarityView to GUI.
12:05:36.204 [D] gui:718              Add view WaveformView to GUI.
12:05:36.208 [D] base:337             Set state for WaveformView.
12:05:36.229 [D] gui:718              Add view CorrelogramView to GUI.
12:05:36.231 [D] base:337             Set state for CorrelogramView.
12:05:36.267 [D] gui:718              Add view ISIView to GUI.
12:05:36.270 [D] base:337             Set state for ISIView.
12:05:36.309 [D] gui:718              Add view FeatureView to GUI.
12:05:36.314 [D] base:337             Set state for FeatureView.
12:05:36.369 [D] gui:718              Add view AmplitudeView to GUI.
12:05:36.373 [D] base:337             Set state for AmplitudeView.
12:05:36.424 [D] gui:718              Add view FiringRateView to GUI.
12:05:36.426 [D] base:337             Set state for FiringRateView.
12:05:56.911 [D] gui:718              Add view TraceView to GUI.
12:05:56.989 [D] base:337             Set state for TraceView.
12:05:56.990 [D] base:494             Set color scheme to depth.
12:06:06.609 [D] base:1626            Adding default color schemes to TraceView.
12:06:21.921 [D] gui:718              Add view ProbeView to GUI.
12:06:21.925 [D] base:337             Set state for ProbeView.
12:06:21.927 [W] gui:671              Could not create view TemplateFeatureView.
12:06:21.961 [D] gui:718              Add view ClusterScatterView to GUI.
12:06:21.964 [D] base:337             Set state for ClusterScatterView.
[... cut ...]
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 165, resource id: 90, major code: 130 (Unknown), minor code: 47
qt.qpa.xcb: QXcbConnection: XCB error: 170 (Unknown), sequence: 180, resource id: 90, major code: 146 (Unknown), minor code: 20
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
QOpenGLWindow::beginPaint: Failed to create context
QOpenGLWindow::beginPaint: Failed to make context current
QOpenGLFunctions created with non-current context
Segmentation fault (core dumped)
`

zm711, 2023-11-17:
@litzj01,

I think there are a couple issues:
1) spikeinterface issues
2) phy problems

For the phy problems could you copy that to your old thread (in the phy repo) and put it in ` instead so we can read it better. The seg fault is likely due to a faulty phy install that we can work on in the phy issue tracker instead.

What is your spikeinterface version? As of `0.99.0` the default is `sparse=True` so sparsity is automatically calculated. For your shape of probe you may need to play with your sparsity_kwargs, but the code you're displaying doesn't quite make sense.

```python
we_sparse = si.extract_waveforms(recording, sorting, "/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/" + 'waveforms_sparse', overwrite=True, allow_unfiltered=True, sparse=True, method="radius", radius_um=70, **global_job_kwargs) 
```
This is fine if that is how you want to do your sparsity, but then your next two lines don't make sense. You shouldn't calculate your sparsity on top of your sparse waveforms. You already have the sparsity inside of `we_sparse`. Then you should export to phy with `we_sparse`.

```python
amplitudes = spost.compute_spike_amplitudes(we_sparse) 
unit_locations = spost.compute_unit_locations(we_sparse) 
spike_locations = spost.compute_spike_locations(we_sparse) 
correlograms, bins = spost.compute_correlograms(we_sparse) 
similarity = spost.compute_template_similarity(we_sparse)

sexp.export_to_phy(we_sparse, "/home/exacloud/gscratch/Sivyerlab/092123_Primate/sorted/no_filter/" + 'phy_folder_for_hdspikes', verbose=True, ) #no sparsity since we_sparse is already sparse
```

You may need to play with your sparsity values (like `radius_um` to work with your probe), but once you have appropriate sparsity you just export that `waveform_extractor` to phy without layering sparsity on top. The other option would be to get dense waveforms and then do a sparsity mask instead.

litzj01, 2023-11-17:
Thanks for your reply! I got pretty confused reading the SI documentation on sparsity which is how I ended up adding the two lines about calculating sparsity post waveform extraction. I can re-extract the waveforms and export to phy without those lines. I am still trying to figure out the right settings for the sparsity. A radius 70um should define a 3 x 3 grid of electrodes surrounding each unit. 

My recordings are pretty large, so extracting the dense waveforms isn't a good option.

I should have the latest version of spikeinterface. I installed it from source a weeks ago.

zm711, 2023-11-17:
That documentation was based on the older defaults where the function would do dense waveforms and then you would calculate a sparsity and then apply the sparsity onto the dense waveforms to generate sparse waveforms. Now the default is to just calculate sparsity. I'm working on updating those docs currently do try to reflect that, so it is helpful to know they are confusing thanks! 

I think you would just want to play around with the radius to have sparse, but not too sparse for Phy. So you could try playing with the `radius_um` to see if you can find the ideal distance for your probe.

I was just asking about version to see what your `extract_waveform` default behavior was. The most recent will have `sparse=True`.

litzj01, 2023-11-17:
I believe the default is `sparse=True`, but the default radius was 40um which would be too small for my probe.

zm711, 2023-11-17:
Yep. Exactly. `sparse=True` and `radius_um=40`, but if that is too sparse for your probe you can increase that value like you're trying.

zm711, 2023-11-21:
@litzj01 
Since you got the sparsity figured out based on your response in the [Phy Issue](https://github.com/cortex-lab/phy/issues/1233), I'll close this issue, but if something else comes up with the Spikeinterface side feel free to open a new issue! Also I'll attach the `sortingview` portion of the docs [here](https://spikeinterface.readthedocs.io/en/latest/modules/curation.html) in case you want to see if `sortingview` is something you would want to test along with Phy.

closed 2023-11-21 by zm711