SpikeInterface/spikeinterface #2201: Concatenating files to sort together

vatsunlab, 2023-11-14:
Hello! I have multiple ~3 minute long recordings that I want to sort together. But when I try this:
[image]

I get this runtime warning, and the size of the concatenated file is less than the size of any of the individual files:
[image]

Could you please let me know what I'm doing wrong? Greatly appreciate any help, thanks!

alejoe91, 2023-11-14:
Hi! Everything looks good and I think that the problem is in the overflow error (so it only affects the print statement). The number of samples and duration looks correct.

vatsunlab, 2023-11-14:
THanks. Continuing on, if I do:
[image]

I get this error:
[image]

Could you identify the problem? Thanks.

alejoe91, 2023-11-14:
@samuelgarcia @h-mayorquin could you take a look at this? Seems related to the pre-allocation of the memmap objects

vatsunlab, 2023-11-14:
Sure, the files are here: https://pitt-my.sharepoint.com/:f:/g/personal/vatsun_pitt_edu/ElNgdDIZN_1DvWj8FNpoEWQBdchbvEiQlPZLTaTm23q31w?e=9LgUyU
I will delete them after you grab it. Thank you!

h-mayorquin, 2023-11-14:
OK, I started downloading them. I will let you know when I am done. 

Can you share the complete script that you ran for the error? That is, I will need a `script.py` file that gets you the error above.

vatsunlab, 2023-11-14:
Hi, the script is in that folder too.

alejoe91, 2023-11-14:
It should!

h-mayorquin, 2023-11-17:
@vatsunlab the branch in https://github.com/SpikeInterface/spikeinterface/pull/2223 should solve your error. Could you test it?

lakritzthemarmoset, 2023-11-21:
> @vatsunlab the branch in #2223 should solve your error. Could you test it?

Hello! I am working in vatsunlab and attempted to implement the fix. It seemed to run the concatenation and writing of the binary recording but then this error happened: 

[image]

Not sure how to handle this. 

Thank you very much for the help!

h-mayorquin, 2023-11-22:
Hi, @lakritzthemarmoset .

Glad that the other problem was fixed.
Can you open a new issue and share with us the full script that produces the error?

You seem to be using docker in windows and trying to save your results in a mounted path (I see C: and D:) which is not supported yet. But share your script in a new issue and we will be able to help. Also, this is the same data you shared with use right?

I am closing this as the original issue was addressed.

closed 2023-11-22 by h-mayorquin

lakritzthemarmoset, 2023-11-22:
> Hi, @lakritzthemarmoset .
>
> Glad that the other problem was fixed. Can you open a new issue and share with us the full script that produces the error?
>
> You seem to be using docker in windows and [...]

Hi @h-mayorquin, 

Thank you again for the help with the concatenation, I really appreciate it!

The docker problem was a simple fix of needing to open the Docker Desktop, which I had seen others online had the same problem in different scenarios (not with SpikeInterface). After that, everything ran smoothly. Do you still think a new issue needs to be opened? If so, I can get the full script after the holidays.

h-mayorquin, 2023-11-23:
Not at all. Happy to hear it was something simple to solve! 

Happy holidays and happy analysis!

commit 3abc55eb (2023-11-22): Merge pull request #2223 from h-mayorquin/fix_concatenate_segments

Fix #2201