SpikeInterface/spikeinterface #2206: SpikingCircus2 throwing error with multi-segment recordings

FrancescoNegri, 2023-11-15:
Hi!
I was trying to run SpykingCircus2 with a multi-segment recording (3 segments for a total of ~18 mins) since, according to my understanding of the [documentation](https://spikeinterface.readthedocs.io/en/latest/modules/sorters.html#handling-multi-segment-recordings), only SpykingCircus2 and Tridesclous support multi-segment data.

However, the `run_sorter(...)` function throws this error:
```
{
	"name": "ValueError",
	"message": "This sorter spykingcircus2 does not handle multi-segment recordings, use si.concatenate_recordings(...)",
	"stack":  ...
}
```

Is such a behaviour expected?
Thank you in advance :)

yger, 2023-11-15:
SC2 should work with multi segment, I'll investigate since I don't know where and why this message is raised...

samuelgarcia, 2023-11-15:
Oups. This is a bug. I guess a bad assert somewhere. Pierre you fix this ?

alejoe91, 2023-11-15:
Problem is here: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/basesorter.py#L50

You need to set it to True in the SC2 class

alejoe91, 2023-11-15:
Note that also TDC2 is missing it!

yger, 2023-11-16:
This has been fixed in #2208

closed 2023-11-16 by yger