SpikeInterface/spikeinterface #2214: Issue with reading .pl2 files

Bosswang12138, 2023-11-15:
Hello everyone,

I'm trying to use the spikeinterface.extractors.read_plexon2() function to read a .pl2 file using the following code:

`import spikeinterface.extractors as se` 
`from pathlib import Path`

`folderPath = Path(r"D:\PROGRAM")`
`filename = Path("16ch1.pl2")`
`fullfilename = folderPath.joinpath(filename)`

`recording = se.read_plexon2(file_path=fullfilename, stream_id='6')`

However, I'm facing the following error:

---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)

File [d:\Anaconda\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\plexon2.py:30](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:30), in Plexon2RecordingExtractor.__init__(self, file_path, stream_id, stream_name, all_annotations)
     [28](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:28) def __init__(self, file_path, stream_id=None, stream_name=None, all_annotations=False):
     [29](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:29)     neo_kwargs = self.map_to_neo_kwargs(file_path)
---> [30](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:30)     NeoBaseRecordingExtractor.__init__(
     [31](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:31)         self, stream_id=stream_id, stream_name=stream_name, all_annotations=all_annotations, **neo_kwargs
     [32](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:32)     )
     [33](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/plexon2.py:33)     self._kwargs.update({"file_path": str(file_path)})

File [d:\Anaconda\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:185](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:185), in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    [156](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:156) def __init__(
    [157](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:157)     self,
    [158](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:158)     stream_id: Optional[str] = None,
   (...)
    [163](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:163)     **neo_kwargs: Dict[str, Any],
    [164](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:164) ) -> None:
    [165](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:165)     """
    [166](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:166)     Initialize a NeoBaseRecordingExtractor instance.
    [167](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:167) 
   (...)
    [182](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:182) 
    [183](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:183)     """
--> [185](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:185)     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    [187](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:187)     kwargs = dict(all_annotations=all_annotations)
    [188](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:188)     if block_index is not None:

File [d:\Anaconda\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:25](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:25), in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     [23](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:23) def __init__(self, block_index, **neo_kwargs):
     [24](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:24)     if not hasattr(self, "neo_reader"):  # Avoid double initialization
---> [25](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:25)         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     [27](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:27)     if self.neo_reader.block_count() > 1 and block_index is None:
     [28](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:28)         raise Exception(
     [29](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:29)             "This dataset is multi-block. Spikeinterface can load one block at a time. "
     [30](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:30)             "Use 'block_index' to select the block to be loaded."
     [31](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:31)         )

File [d:\Anaconda\envs\si_env\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:64](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:64), in _NeoBaseExtractor.get_neo_io_reader(cls, raw_class, **neo_kwargs)
     [62](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:62) neoIOclass = getattr(rawio_module, raw_class)
     [63](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:63) neo_reader = neoIOclass(**neo_kwargs)
---> [64](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:64) neo_reader.parse_header()
     [66](file:///D:/Anaconda/envs/si_env/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py:66) return neo_reader

File [d:\Anaconda\envs\si_env\lib\site-packages\neo\rawio\baserawio.py:178](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:178), in BaseRawIO.parse_header(self)
    [165](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:165) def parse_header(self):
    [166](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:166)     """
    [167](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:167)     This must parse the file header to get all stuff for fast use later on.
    [168](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:168) 
   (...)
    [176](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:176) 
    [177](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:177)     """
--> [178](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:178)     self._parse_header()
    [179](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/baserawio.py:179)     self._check_stream_signal_channel_characteristics()

[... 1 lines cut ...]
    [237](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/plexon2rawio/plexon2rawio.py:237) for spike_channel_idx, spike_header in enumerate(self.header['spike_channels']):
    [238](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/plexon2rawio/plexon2rawio.py:238)     schannel_name = spike_header['name'].split('.')[0]
[... cut ...]
    [655](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/plexon2rawio/pypl2/pypl2lib.py:655) if not result:
    [656](file:///D:/Anaconda/envs/si_env/lib/site-packages/neo/rawio/plexon2rawio/pypl2/pypl2lib.py:656)     self._print_error()

ArgumentError: argument 2: TypeError: expected c_char_Array_5 instance instead of bytes

Has anyone encountered a similar issue with the read_plexon2() function or has any suggestions on how to resolve this error?

Any help or guidance would be greatly appreciated. Thank you!

alejoe91, 2023-11-16:
Hi, 

SpikeItnerface uses NEO for IO. Can you open an issue there?
https://github.com/NeuralEnsemble/python-neo/issues

Bosswang12138, 2023-11-16:
> Hi,
> 
> SpikeInterface uses NEO for IO. Can you open an issue there? https://github.com/NeuralEnsemble/python-neo/issues

"Hi,

Thank you for your suggestion! I'll open an issue in the NEO repository to address this matter.

Best regards,"

zm711, 2025-02-05:
We've worked on pl2 a bit. Since this is over 1 year old I'll close as stale, but if our current work with Neo hasn't been enough just reopen!

closed 2025-02-05 by zm711