SpikeInterface/spikeinterface #2222: `plot_unit_waveforms` and `plot_unit_templates` display incorrectly for probegroup

zm711, 2023-11-17:
Here is one dataset:

[image]

Here's another dataset:

[image]

The difference between the datasets are that the second has a probegroup attached rather than a probe. But other than that they should be equivalent. 

In fact the x distance for the probe is 25 um and for the probegroup, the distance on that shank is only 20 um so this isn't a distance thing. I've tried setting the figsize myself and that didn't help.

This is currently in the `matplotlib` backend.

alejoe91, 2023-11-20:
Interesting and weird! @zm711 can you show us how you are attaching the probe/probegroup?

zm711, 2023-11-20:
This the probegroup we made on the probe interface issue together. and it is just a 

```python
recording.set_probegroup(multishank)
```

for the weird looking one.

alejoe91, 2023-11-20:
and the other one is `recording.set_probe(multishank)`?

Can you share the probeinterface JSON?
```
pi.write_probeinterface("multishank.json", multishank)
```

zm711, 2023-11-20:
> and the other one is recording.set_probe(multishank)

No. So it is a different probe. It's just one of the commercial probes.

so,
```python
recording.set_probe(commercial_probe)
```

Yep. It's on my work computer, so I'll send it later today (both the probe and the probe group).

zm711, 2023-11-20:
[multishank.json](https://github.com/SpikeInterface/spikeinterface/files/13415379/multishank.json)
[singleprobe.json](https://github.com/SpikeInterface/spikeinterface/files/13415380/singleprobe.json)
@alejoe91, here are two json files. The multishank was a probegroup and the singleprobe was set with a `set_probe`.

zm711, 2023-12-01:
I figured it out:

So the probe I was using is not aligned in the x (ie it makes a v), so the `get_waveforms_scales` function fails to get the correct x offset. I'm not sure of the exact correct solution but I'll link a PR that can at least start a discussion.
To get this corrected image I just found that if the `delta_x` is too small (ie in my case of the slowly v electrodes the delta x was .5 which is basically 0 and was making too small an offset.
[image]

PR #2287 (2024-01-19): Add a more robust `delta_x` to `unit_waveforms`
Fixes #2222 

## The Problem
Basically I go through this in the issue, but what I figured out is that when using a Cambridge neurotech H7 probe which has a slanting and slowly changing x position for the contacts the `delta_x` was being calculated as 0.5 which was leading to the waveforms being super squished.

[image]

I think the basic logic of this function before (but correct me if I'm wrong), was that in the case of `x_chans.size==1` then the x axis is all stacked so let's force an `delta_x` of 40 to prevent this type of skinny waveforms. If [... cut]