SpikeInterface/spikeinterface #2224: Issues with sorting through mountainsort using mda file

claudio1212, 2023-11-17:
Hi, as I commented on a separate thread, I'm having some issues using Open Ephys legacy files to do sorting in spikeinterface. In particular, In some session these files can be correctly loaded and sorted while in others I'm faced with an error related to the presence of possible gaps in the file (as I mentioned in the other tread).

 To try an alternative to loading continuous files, I used previously created mda files containing the data I wanted I wanted to analyze. Using these files I could always correctly load and sort the data, but I had as a results very few units (5 found in 32 recording channels, for example). These results are in contrast with those that I obtained  both using mountainsort3 and by using the same sorter that I am considering in this thread (mountainsort5) in spikeinterface, but after loading the continuous files of that recording and not the mda file. In these latter cases I found at least one cluster per recording channel with various channels having two or three of them.

I was then wondering If I can actually use mda files as an alternative, given though I saw they can be correctly loaded also in those sessions in which trying to load continuous OE files returns me the above mentioned error, or If these files for some reasons are not actually and always correctly loaded leading to differences in the number of units found.

alejoe91, 2023-11-20:
@magland

magland, 2023-11-20:
Hi @claudio1212 I think this issue belongs in either mountainsort4 or mountainsort5 repo

https://github.com/magland/mountainsort4

https://github.com/flatironinstitute/mountainsort5

Although I'm not sure how much I can help since it has been a very long time since I have looked at or thought about mountainsort3. :)

claudio1212, 2023-11-26:
Thanks for the reply, so essentially I figure there is no actual way to know if the results I'm obtaining via mountainsort 5 are correct or not by comparing them with mountainsort 3, am I right? This was my main issue at the moment

claudio1212, 2023-11-26:
Thanks for the reply, I'm sorry I haven't replied yet. I'll try to ask this in the repo you linked.

alejoe91, 2023-11-27:
Moved discussion to sorter repos