SpikeInterface/spikeinterface #2233: tridesclous2 parameters change

rat-h, 2023-11-20:
It seems the parameters for tridesclous2 were renamed/changed. I need a map of old parameters to new parameters. All my optimization for our data doesn't work any more (-.-)

old version:
```json
{
        "apply_preprocessing": true,
        "detection": {
            "detect_threshold": 5,
            "exclude_sweep_ms": 0.4,
            "peak_sign": "neg"
        },
        "filtering": {
            "freq_max": 8000.0,
            "freq_min": 300
        },
        "general": {
            "local_radius_um": 100,
            "ms_after": 3.5,
            "ms_before": 2.5
        },
        "hdbscan_kwargs": {
            "allow_single_cluster": true,
            "cluster_selection_method": "leaf",
            "core_dist_n_jobs": -1,
            "min_cluster_size": 25
        },
        "job_kwargs": {
            "chunk_duration": "1s",
            "n_jobs": -1,
            "progress_bar": true
        },
        "localization": {
            "max_distance_um": 1000,
            "optimizer": "minimize_with_log_penality"
        },
        "matching": {
            "peak_shift_ms": 0.2
        },
        "selection": {
            "min_n_peaks": 20000,
            "n_peaks_per_channel": 5000
        },
        "waveforms": {
            "max_spikes_per_unit": 300
        }
}
```
the new one
```json
{
    "apply_preprocessing": true,
    "waveforms": {
        "ms_before": 0.5,
        "ms_after": 1.5,
        "radius_um": 120.0
    },
    "filtering": {
        "freq_min": 300.0,
        "freq_max": 12000.0
    },
    "detection": {
        "peak_sign": "neg",
        "detect_threshold": 5,
        "exclude_sweep_ms": 1.5,
        "radius_um": 150.0
    },
    "selection": {
        "n_peaks_per_channel": 5000,
        "min_n_peaks": 20000
    },
    "features": {},
    "svd": {
        "n_components": 6
    },
    "clustering": {
        "split_radius_um": 40.0,
        "merge_radius_um": 40.0
    },
    "templates": {
        "ms_before": 1.5,
        "ms_after": 2.5
    },
    "matching": {
        "peak_shift_ms": 0.2,
        "radius_um": 100.0
    },
    "job_kwargs": {
        "n_jobs": -1
    },
    "save_array": true
}
```

rat-h, 2023-11-20:
BTW, `get_sorter_params_description('spykingcircus2')` and `get_sorter_params_description('tridesclous2')` returns an empty dictionary.

zm711, 2023-11-20:
Yep that is on the to-do docket from #1802. And I would say even more important now that people want to use the sorter :)

rat-h, 2023-11-20:
`tridesclous2` was (and hopefully is) the best sorter for our data so far.

samuelgarcia, 2024-02-06:
Hi I made some update in tridesclous2.
get_sorter_params_description has change a bit.

zm711, 2024-03-25:
@rat-h, please see the `get_sorter_params_description('tridesclous2')` for the current params. I'll close this but if anything is unclear just open a new issue :)

closed 2024-03-25 by zm711