SpikeInterface/spikeinterface #2234: Support for syncing data from multiple Neuropixels as well as an NIDQ device when using SpikeGLX for acquisition

manimoh, 2023-11-21:
Hi!
I was wondering if there was a functionality on syncing all the data-streams from 1 or more NPX probes as well as the analog digital channels on the NIDQ device, when using SpikeGLX for acquisition. While SpikeGLX provides command-line functionality to sync individual timestamps (spike timing, event onset etc) across various devices using their 'TPrime' command, there is no way to synchronize the the continuous data-streams acquired on different devices (for example LFP channels from 2 different Neurpixels probes, and an analog channel on the NIDQ device which actually recordings the running speed of the animals).

I see that there is some work in progress [here](https://github.com/SpikeInterface/spikeinterface/blob/38d82d4f8ce1430f5e3375a907e5c3c557a78711/src/spikeinterface/extractors/neuropixels_utils.py#L107), but was wondering if that function will satisfy the requirements I am asking about, or are there alternate ways to achieve it!

samuelgarcia, 2023-11-22:
Hi Manish.
This is on my todo list since almost 1.5 years!
I swear I will do it for christmas but I prefer to not tell which year :)

I have private code somewhere that is doing this I need to propagate to that place this should be quite easy.
Leave this issue open and feel free to write "ping" in the discussion every 2 weeks.

manimoh, 2023-11-22:
I actually would appreciate this functionality for my own use -- happy to help debug/test it :)

manimoh, 2023-12-04:
"ping" @samuelgarcia :)

manimoh, 2024-01-24:
Sorry for being annoying, but "ping"

samuelgarcia, 2024-01-29:
Hi @manimoh.
Thank you the recurent ping.
I am quite busy witha big refactporing the month.
Do not stop the ping! After this big work I will try to implement your need in spikeinterface.
Sorry for the very long delay.
Have a nice day.

manimoh, 2024-04-24:
"Ping"

TheChymera, 2024-06-25:
@samuelgarcia still interested in this? we need it for conversion of our neuropixels data.

guptadivyansh, 2025-03-13:
Hi @samuelgarcia , how can we can help with implementing this?

alejoe91, 2025-03-13:
@guptadivyansh if you have some solutions it would be great to draft a PR so we can discuss about it! This will probably be in this file: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/neoextractors/spikeglx.py

Do you have time to give it a try?