SpikeInterface/spikeinterface #2237: Phy output is sparse

DradeAW, 2023-11-22:
Hi,

Since a few weeks ago, all of my `export_to_phy` have sparse waveforms.
I use `wvf_extractor = si.extract_waveforms(..., sparse=False)` followed by `export_to_phy(wvf_extractor, ...)`.

So basically same as before, but adding that the WaveformExtractor object is not sparse.
But the output in phy is still sparse (8 channels shown instead of 64).

I'm not sure what broke it or exactly when this started happening.

zm711, 2023-11-22:
Could you do one quick check and see if the problem is occurring at the level of the wvf_extractor so type

```python
wvf_extractor.is_sparse()
```

I think Sam, you, and me made changes to those parts of the code most recently so I think we need to see at which level this is happening (because it shouldn't be).

DradeAW, 2023-11-22:
```
>>> wvf_extractor.is_sparse()
False
>>> 
```

zm711, 2023-11-22:
So in this case phy code should run

```python
used_sparsity = ChannelSparsity.create_dense(wvf_extractor)
```

Do you want to try that and just see what is returned in `used_sparsity`? 

The thing I changed is that `template_inds` is used as a sparsity mask by phy, so if the extractor is dense then you don't need to save that file so that Phy sees that your files are dense. So the other thing that would be useful is when you run

```
phy template-gui params.py --debug
```
Does phy see your templates as sparse or dense? (The message should occur in the log of the console you're using).

DradeAW, 2023-11-22:
Using the debug mode, phy says `Templates are dense.`
Also, `templates.npy` has the correct shape.

This is weird!

zm711, 2023-11-22:
Which view is actually showing up as sparse? the `Waveform View` or the `Template View`. Maybe with our edits we missed something else and that is causing this issue. 

> EDIT: for exports done about a month ago, it says Templates are sparse. But I'm getting all my channels ^^
Maybe in the dense format, phy only shows a few channels?

Basically before SpikeInterface was giving a dummy mask so Phy was applying a sparsity mask that did nothing and labeling the data as sparse when it shouldn't have. That's why I made the change to not apply the sparsity mask. But if this is the problem you could fix this by just deleting this if statement on a branch and trying to export:

https://github.com/SpikeInterface/spikeinterface/blob/3abc55ebd29fe652989eeb311337645432c9c183/src/spikeinterface/exporters/to_phy.py#L206

Then it should say your waveforms are sparse, but it's a dummy mask. This would likely tell us that something else needs to be fixed to change this behavior. Does that make sense to try?

DradeAW, 2023-11-22:
The `WaveformView` is giving a sparse output.

I tried by setting my `WaveformExtractor` a dense sparsity (all ones), and the output is what I expect (all my channels showing up).

DradeAW, 2023-11-22:
I am computing with `compute_pc_features=False`, to it has nothing to do with features :)

I agree it would be nice to fix `export_to_phy`, but in the mean time I'm using this hack ^^