SpikeInterface/spikeinterface #2241: Event times precede start of recording time Open ephys recording

LiesDec, 2023-11-22:
When I compare 

- the timestamps of TTL events 
- and the start timestamp of my recording 

the TTL events **precede** the start time, which seems impossible.

When I use a different package to read the data ( [open-ephys-python ](https://github.com/open-ephys/open-ephys-python-tools/tree/main))  I get **the same event timestamps** but a **different recording start timestamp**.
So it seems that the recording timestamps are read wrongly. 

It is important that the timestamps are correct 'cause otherwise you cannot synchronise the recordings with other acquisition devices. 

### Code with spike interface:

```
import spikeinterface.full as si

#read events 
events=si.read_openephys_event(datapath / animal / session )
events_times = events.get_event_times(channel_id="Neuropixels PXI Sync")

#read recording from probe A (stream 0)
recording = si.read_openephys(datapath / animal / session , stream_id='0')
start_time = recording.get_time_info()['t_start']

print('TTL timestamps: {}'.format(events_times))
print('start time recording: {}'.format(start_time))

```
Output:
```
TTL timestamps: array([ 136.60946667,  137.60946667,  138.60843333, ..., 3150.2826    ,
       3151.28180516, 3152.2816    ])
start time recording: 158.99466666666666
```
### Code with open-ephys-python

```
from open_ephys.analysis import Session

# Read experiment 1, recording 1 first record node and first recording (experiment1,recording1)
recording = Session(directory).recordnodes[0].recordings[0] 

#Read event ON state (1)  from probe A (stream 0) 
events = recording.events[(recording.events['stream_index']==0) & (recording.events['state']==1)]

#Read timestamps from the recording from probeA (stream 0)
time = recording.continuous[0].timestamps

print('TTL timestamps: {}'.format( np.array(events['timestamp'])))
print('start time recording: {}'.format(time[0]))
```
Output:

```
TTL timestamps: [ 136.60946667  137.60946667  138.60843333 ... 3150.2826     3151.28177183
 3152.28156667]
start time recording: 136.39443333333332
```

zm711, 2023-11-22:
@LiesDec,

SpikeInterface relies on [Neo](https://github.com/NeuralEnsemble/python-neo) for reading open-ephys. It would be helpful if you could also open the issue over there since any issues with the file reading would need to be fixed at the Neo level and then propagated to SpikeInterface. There have been some patches to Neo, but since Neo hasn't had a release for a while it might be beneficial to try to install neo from source ([instructions here](https://spikeinterface.readthedocs.io/en/latest/installation.html)) to see if that has been patched already. If not then posting your issue to Neo is the way to go.

alejoe91, 2023-11-22:
Thanks @zm711 

@LiesDec can you try to read the recording with the `load_sync_timestamps=True` argument? 

Then you can get the start time with:
```
recording.get_times()[0]
```

LiesDec, 2023-11-22:
@alejoe91 that worked! 
I didn't know that option was the one I should have used. 
Thank you for the quick reply! :)

alejoe91, 2023-11-22:
Yeah we've been adding several options for OpenEphys + NP. Lagging a bit behind on documentation, but we'll get there!

Essentially, this option loads the synchronized timestamps directly from Open Ephys, see [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/neoextractors/openephys.py#L220-L237)

Ok to close this?

LiesDec, 2023-11-23:
@alejoe91 Absolutely! ☺️
Great that you have been adding several options for openEphys + NP. I'll make sure to check out the source code more often! ☺️

closed 2023-12-04 by alejoe91