SpikeInterface/spikeinterface #2242: No way to access spikeGLX Digital Channels on the 'nidq' stream

manimoh, 2023-11-22:
Hi,

While `read_spikeglx(spikeglx_folder, 'nidq')` allows you to access the analog I/O channels, the signal on the digital i/O channels (TTL pulses) are missing -  these often contain events related to experimental control and would be important to capture from a downstream analysis point of view.

alejoe91, 2023-11-22:
Hi @manimoh 

Is there any code to read them in Python? We should need to implement `read_spikeglx_events`! (We have one for Open Ephys)

manimoh, 2023-11-22:
Unfortunately all I could find is that the 'XD' (digital) not really parsed in neo's [spikeglxrawio](https://github.com/NeuralEnsemble/python-neo/blob/b8989a15ff057db062fa5aefecfb7c0d12adde2f/neo/rawio/spikeglxrawio.py#L63), so I don't know where else to look at :/

samuelgarcia, 2023-11-23:
There is no clear API on neo side to read digtal stream.
This is on my todo list since very long time.

manimoh, 2023-11-23:
Okay I did some digging and here's what I found:
The digital channels (if present and saved) are encoded as an 8 bit word, where the digital channel 0 refers to the least significant bit. And each sample essentially stores the state of the the 8 digital channels in this word. 

I essentially confirmed this by inserting the following code after this [line](https://github.com/NeuralEnsemble/python-neo/blob/b8989a15ff057db062fa5aefecfb7c0d12adde2f/neo/rawio/spikeglxrawio.py#L105)  in the neo's spikeglxrawio.py 
```
extracted_word = data[:,8] # Extract the 9th value 8 analog channels, and 1 'digital' word`
all_samples = ['{0:08b}'.format(x) for x in extracted_word]
```
And then seeing if the transitions in the bits based on `all_samples` made sense.

I am attaching 2 test cases that can help you - 1 of them has only sync pulses on digital channel 0, and the other one has sync pulses on digital channels 2 and 3. The sync pulses are at 1 Hz, and the TTL pulse widths are ~1 sec each
[Sync0.zip](https://github.com/SpikeInterface/spikeinterface/files/13452648/Sync0.zip)
[Sync0TTL2and3.zip](https://github.com/SpikeInterface/spikeinterface/files/13452684/Sync0TTL2and3.zip)

P.S: The second zip file contains a .7z compression inside because i was not able to decrease the filesize to less than 25 MB, I apologize in advance for the inconvenience!

yarikoptic, 2023-12-16:
@samuelgarcia how could we may be help to get a fix for this going?  may be point to few ideas on where to introduce what and then @TheChymera with @manimoh could take a stab at implementing it?

alejoe91, 2023-12-18:
Hi @yarikoptic, @manimoh and @TheChymera

Yes this will need to be implemented in NEO, specifically in the [`SpikeGLXRawIO`](https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/spikeglxrawio.py)
Can you open an issue there? 

Once that is implemented, on the SpikeItnerface side we just need a very thin layer on top of NEO events, similar to this [OpenEphys](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/neoextractors/openephys.py#L258-L286) implementation.

manimoh, 2024-01-29:
Added a [PR](https://github.com/SpikeInterface/spikeinterface/pull/2442) to spikeinterface, pending merging of [PR](https://github.com/NeuralEnsemble/python-neo/pull/1383) in NEO

zm711, 2024-09-20:
PRs merged so I'm closing this :)

closed 2024-09-20 by zm711