SpikeInterface/spikeinterface #2249: channel_id confusion

aghatpande, 2023-11-23:
Thank you for this excellent package. I am new and have a question about channel IDs assigned by default by the 'read_neuralynx' (or perhaps any other RecordingExtractor) method. I used it to read in my 64 channel data where each channel is a separate file e.g. '['CSC1.ncs' 'CSC10.ncs' 'CSC11.ncs' 'CSC12.ncs]' When I query the recording object for channel ID's using 'recording.get_channel_ids' I get a list like so: ['62' '66' '60' '63']. How do these channel id's relate to my list of files? Are the IDs randomly generated integers with the files being imported as listed in the neuralynx data folder? I did try to search the documentation extensively but didn't come across a unambiguous answer. Any insight into will be much appreciated.

h-mayorquin, 2023-11-28:
Hi, @aghatpande 
I don't remember how the channel_ids are set but looking at the neo code it seems that the channel name properties should be there. 

Can you load your recording again adding the property 

`recording = read_neuralynx(your_arguments, all_annotations=True)`
And then printing  `recording.get_property_keys()`

Also, are you files large? Maybe you can share them with us so I can figure out the mapping.

h-mayorquin, 2023-11-28:
I suspect that @samuelgarcia was talking from the point of view of developers. But if this works for you that's great. You can always rename your channels with the rename channel function.

Good luck.

closed 2023-11-28 by h-mayorquin