SpikeInterface/spikeinterface #2257: Cluster quality and waveform (basic)

kshtjkumar, 2023-11-23:
Hi, I have a question regarding the quality of my sorting, while i run this code (i use tetrode, and mountainsort5 for sorting):
f```
ig, ax = plt.subplots()
num_samples = we.nbefore + we.nafter
timestamps_ms = np.arange(num_samples) / we.sampling_frequency * 1000
for i, unit_id in enumerate(sorting_KS.unit_ids[:1]):
    template = we.get_template(unit_id)
    color = colors[i]
    
    ax.plot(timestamps_ms,wf[:, :, 0].T, color=color, lw=0.3, alpha = 0.2)
    ax.plot(timestamps_ms,template[:, 0].T, color='red', lw=3) 
plt.xlabel("Time (mS)")
plt.ylabel("Amplitude (micro Volt)")
plt.title("Avg_Waveform")
```

is my understanding  correct here:
The below line give me all the waveforms of the spikes in the 1st cluster coming from channel number 0:
ax.plot(timestamps_ms,wf[:, :, 0].T, color=color, lw=0.3, alpha = 0.2)
and this line gives the average of the clusters of the channel 0:
ax.plot(timestamps_ms,template[:, 0].T, color='red', lw=3) 

even tho while recording we see good spiking activity after sorting the wave forms look deformed and skewed.

Upon evaluating the quality matrix the ISI violation of the cluster comes to be 0 or less than 0.5. How to interpret this? If waveform is good then the isi ration is bad and vice versa, in most of my cases. How to be sure that the quality of my recording is good? 

<img width="479" alt="Screenshot 2023-11-24 at 3 57 43 AM" src="https://github.com/SpikeInterface/spikeinterface/assets/68465910/ed64ed92-22ec-4db3-86a4-fa6d261b0575">

<img width="648" alt="Screenshot 2023-11-24 at 3 58 23 AM" src="https://github.com/SpikeInterface/spikeinterface/assets/68465910/8885a120-5ebc-46ca-a282-c46c444cc822">

zm711, 2023-11-27:
@kshtjkumar,

instead of trying to pull things out yourself I would suggest starting with some of the internal widgets, [for example](https://spikeinterface.readthedocs.io/en/latest/modules_gallery/widgets/plot_3_waveforms_gallery.html) shows how to pull out waveforms 
For waveforms:

```python
unit_ids = sorting.unit_ids[:4]

sw.plot_unit_waveforms(we, unit_ids=unit_ids)
```

for templates
```python
unit_ids = sorting.unit_ids

sw.plot_unit_templates(we, unit_ids=unit_ids, ncols=5)
```

I'm not sure where you are getting `wf` from in your example above? Are you reading the raw file yourself or using an SI based function. Is your `wf` filtered or not? I'm wondering if you could try the same plotting using the internal SI tools I've listed and then post those so we can see what is happening there. Otherwise, it would be beneficial to have your whole script so we know where each variable is coming from. 

To summarize:

1) Could you try with the `widgets` module to redo your plotting
2) Could you post your whole script so we can see where things are coming from

kshtjkumar, 2023-11-27:
hi, tried the above widgets code!
[image]

kshtjkumar, 2023-11-27:
@zm711 

here is my full code:

```
import spikeinterface.full as si  # import core only
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.widgets as sw
import pyintan
import spikeinterface as si
import spikeinterface.extractors as se
import spikemetrics as st
import matplotlib.pyplot as plt
import numpy as np
import warnings
warnings.simplefilter('ignore') 
%matplotlib widget

file = "17_feb_lko_230217_155836_full_merged.rhs"
#file ='21_11_23_merged_3.rhs'
#file = "15th_nov_merged_7.rhs"
reader = se.read_intan(file,stream_name ='RHD2000 amplifier channel',use_names_as_ids=True)
recording_rhs = reader
print(reader.channel_ids)
recording_rhs.annotate(is_filtered = False)
channel_ids = recording_rhs.get_channel_ids()
fs = recording_rhs.get_sampling_frequency()
num_chan = recording_rhs.get_num_channels()
num_segments = recording_rhs.get_num_segments()
recording_rhs_1 = recording_rhs

print("Channel_ids = ", channel_ids)
print("Sampling_frequency = ", fs)
print("Number of Channels = ", num_chan)
print("Number of segments = ", num_segments)
print('Total_rec_duration = ', recording_rhs.get_total_duration())

#ecog = ['D-000','D-002', 'D-004', 'D-006']
#ecog = ['B-004', 'B-005', 'B-006', 'B-007']
ecog = ['B-006', 'B-007', 'B-008' ,'B-009']
#ecog = ['B-004']
recording_rhs = recording_rhs.channel_slice(ecog)

import probeinterface as pi
number_of_channel = 4
posn = np.zeros((number_of_channel,2))
radius = 50
for i in range(number_of_channel):
    theta = i/number_of_channel*2*np.pi
    x = np.cos(theta)* radius
    y = np.sin(theta)* radius
    posn[i] = [x,y]
    print(i,theta,x,y)
probe = pi.Probe(ndim = 2)
probe.set_contacts(posn, shapes = 'square', shape_params={'width' : 20 , 'height' :20 })
probe.set_contact_ids([f"Ch{i}" for i in range (number_of_channel)])
probe.create_auto_shape()
pi.plotting.plot_probe(probe, with_contact_id=True)
probe.to_dataframe(complete=True)
probe.set_device_channel_indices(np.random.permutation(number_of_channel))
probe.to_dataframe(complete=True)
recording_rhs.set_probe(probe, in_place = True)
probe_loaded = recording_rhs.get_probe()
probe_loaded.to_dataframe(complete=True)
recording_rhs.get_channel_locations()
#sw.plot_probe_map(recording_rhs, with_channel_ids = True)
recording_rhs.get_property('contact vector')

recording_ecog = spre.bandpass_filter(recording_rhs, freq_min=300, freq_max = 6000 ) #bandpass filter
recording_notch_ecog = spre.notch_filter(recording_ecog,q = 50)  #notch_filter 
#rec_ecog11 = recording_notch_ecog.channel_slice(['B-011', 'B-029', 'B-030' ,'B-031'])
rec_ecog = spre.common_reference(recording_notch_ecog, operator="median", reference="global")  #rereferencing the data 
#rec_ecog = recording_rhs
sorting_KS = ss.run_mountainsort5(rec_ecog)
#sorting_KS = ss.run_pykilosort(rec_ecog)

print("KS found", len(sorting_KS.get_unit_ids()), "units")
sorting_KS = sorting_KS.remove_empty_units()
print("KS found", len(sorting_KS.get_unit_ids()), "non empty units")

rec_ecog.annotate(is_filtered=True)
we = si.extract_waveforms(rec_ecog, sorting_KS, folder="waveforms_kshtj094416596", load_if_exists=False, overwrite=True )
print(we)
spost.compute_spike_amplitudes(we)

for unit in sorting_KS.get_unit_ids():
    waveforms = we.get_waveforms(unit_id= unit)
    spiketrain = sorting_KS.get_unit_spike_train(unit)
    print('Unit', unit, "num_waveforms:" ,(waveforms.shape[2]), "num spikes:", (len(spiketrain)))

num_samples = we.nbefore + we.nafter
timestamps_ms = np.arange(num_samples) / we.sampling_frequency * 1000
    
unit_ids = sorting_KS.unit_ids
for unit_id in unit_ids[:]:
    fig, ax = plt.subplots()
    template = we.get_template(unit_id=unit_id, mode='average')
    print(template.shape)
    ax.plot(timestamps_ms,template)
    ax.set_title(f'{unit_id}')

plt.show()
plt.xlabel("Time (ms)")
plt.ylabel("Amplitude (micro volt)")

colors = ['green', 'red', 'blue','yellow','Olive', 'Teal', 'Fuchsia','black','orange', 'purple',"brown"]
#['Olive', 'Teal', 'Fuchsia','green']
fig, ax = plt.subplots()
for i, unit_id in enumerate(sorting_KS.unit_ids[:]):
    wf = we.get_waveforms(unit_id)
    color = colors[i]
    ax.plot(timestamps_ms,wf[:, :, 0].T, color=color, lw=0.4)
    #plt.legend(["1","2","3"])
    
plt.xlabel("Time (mS)")
plt.ylabel("Amplitude (micro Volt)")
plt.title("Waveform")

qm = sqm.compute_quality_metrics(we)#, sparsity=sparsity_radius[0], verbose=True)
display(qm)

pc= spost.compute_principal_components(we, n_components= 3)
colors = ['Olive', 'Teal', 'Fuchsia','black', "blue",'red','green','yellow','orange','magenta']
fig, ax = plt.subplots()
for i, unit_id in enumerate(sorting_KS.unit_ids[:]):
    comp = pc.get_projections(unit_id)
    print(comp.shape)
    color = colors[i]
   ax.scatter(comp[:, 0, 3], comp[:, 1, 3], color=color)
    plt.legend(['1', '2','3','4'])
    plt.title("PCA Clusters")

```

kshtjkumar, 2023-11-27:
`wf`
is taken from this page:
https://spikeinterface.readthedocs.io/en/0.92.0/modules/toolkit/plot_2_postprocessing.html

zm711, 2023-11-27:
That looks like a nice tetrode, but all we need to confirm is that those are the actual mappings between the device and the data matrix.  So you know for that the left contact (`dev0`) is actually on your tetrode in that location? For example your data should be a matrix `(num_samples, num_channels)`. So to get the 1st channel I would do `matrix[:, 0]` (because python is 0 indexed). And based on your plot above the 0's channel is the left most point, medium height point on the tetrode. If that is true for your recording set-up then your tetrode is fine, but if that left contact is actually the 4th channel then your set-up would be wrong. 

Often the mapping for omnetics means that the numbers aren't a simple `np.arange` but often it is something more like `[1, 3,2, 0]`. So really we just need to confirm that you know which contact on your probe goes to which place via the omnetics connector. 

For some example mappings for `Cambridge Neurotech` see [here](https://github.com/SpikeInterface/probeinterface/blob/70d77be9197b21e263df51f2604beb2eef2a29fe/src/probeinterface/wiring.py#L9C1-L50). But what you'll likely notice right away is that these mapping appear to be a random order of numbers but they are not. They are the exact order of electrode contacts and data matrix. Since you're not using a commercial setup you have to make the wiring between contact to data matrix and this is really the most crucial step. (And this is why I really want to make sure you do this right, so that you analyze your data correctly). 

As another note,  is there a reason you want to start at -50? You could just have your deepest contact be 0 and have your top contact be 100 instead? It's all rather arbitrary, but when you use plotting functions if you're trying to think about the depth of your probes it might be nicer to not have to remember an offset of 50 and just have the offset be 0.

kshtjkumar, 2023-11-29:
@zm711  could you please help with the rest of the code if the probe code looks fine?

zm711, 2023-12-04:
So for generating your probe, there's one small thing you can change
for `square` you only need to specify `width`, because the square as the same values. The `height` keyword is for `rect` shape. 

For the notch filter that makes sense to prenotch the data, especially if you have bad line noise. The bandpass is already done by ms5, but if your recording is super noisy than I can understand trying to clean it up as much as possible. So yeah up to point it looks like you're in okay shape (again, I think opening an issue in probeinterface to really carefully go through the wiring could be beneficial, but it seems okay--the issue being if you link up the wrong channels then a spike detected on one of your channels, might not actually be close to any of your other channels and so the waveforms would look bad even though other metrics would look okay--this happened in your first image which was my original concern). 

The other thing that might be helpful for you is to got the ms5 repo and read about how the sorting works. ms5 currently gives you 3 scheme options (spikeinterface defaults to scheme2), but for your data it may be better to use a different scheme to get better sorting results. If you read the ms5 docs or open an issue there they could potentially help you figure out the best way to use their sorter [ms5](https://github.com/flatironinstitute/mountainsort5). 

Moving on during your `extract_waveforms` I would need to know your version of SI, because the default was switched to automatically do sparse, but for your tetrode you likely don't want to be sparse so we would want to make sure your extractor is dense. Could you do:

```python
we.is_sparse()
```
and see what that returns.

Otherwise if all those other steps worked then you just have to pick what your cutoffs are (ie you should read the qc docs and see which work well with tetrodes and then create cutoffs based on your preferred metrics). 

tl;dr

The code looks okay for sorting as long as the correct tetrode channels are being fed in. Not all qc will work great for tetrodes to read up on the qc metrics you actually want to use. Your waveforms look a little strange, but your amplitude distribution looks fine (albeit a little small).

kshtjkumar, 2023-12-04:
```
we.is_sparse()
False
```
By qc docs do you mean the quality metrics doc? if not Can you please point me to it. 

" you just have to pick what your cutoffs are" by this do you imply which clusters to reject based on isi_violations and other things ? 

My si version is `0.98.0.dev0`

kshtjkumar, 2023-12-07:
hi @zm711 , so i cudnt install mountainsort4, the issues is still open in the relevant repository. I updated spikeinterface to `'0.99.1'`
Doing the same steps give me two issues, 1st while running; 
`sorting_rec = ss.run_sorter("mountainsort5", rec_ecog_ref, output_folder='mountainsort5_output1' )`
I have to mention each time the output folder, earlier i didnt do this , is there a way to over write it? I t wud be nice not to make so many folders and rename each in every run.
second , now once its all sorted, i run : `we.is_sparse()` it returns `True`, and i dont think we want that for tetrode right? Can I make it dense ? 
Other than that when i run : 
```
unit_ids = sorting.unit_ids
sw.plot_unit_templates(we, unit_ids=unit_ids, ncols=5)
```

the output is weird: 
[image]

I am not getting waveforms on all channels as i got before! 

Can you please guide where am I going wrong here ?

zm711, 2023-12-07:
>I have to mention each time the output folder, earlier i didnt do this , is there a way to over write it? 

So unfortunately yes for the moment. The code base is working on adding overwrite options to most of the functions, but not everything has an overwrite yet. Sorry! 

> Can I make it dense ?

You may or may not want want sparse waveforms for your tetrode. For instance if you had a tetrode where each contact was 100 um away then it might be more similar to 4 monotrodes rather than a true tetrode. If your tetrode contacts are all within 10 um then even with the default parameters the sparsity should change too much in that case.
But you can run your `si.extract_waveforms(...., sparse=False)` and that will return dense waveforms. It used to be that sparse had a default of False, but this default was switched to True for more recent versions since it is more common for people to use high-density MEAs these days vs tetrodes.

Because you have sparse waveforms your figures might look different. I would do redo with `sparse=False` and see what that figure looks like when you go back to dense waveforms with the more recent version of spikeinterface (0.99.1).

closed 2024-06-19 by zm711