SpikeInterface/spikeinterface #2261: AttributeError when setting docker=True in spikeinterface.sorters.run_sorter

pantelisantonoudiou, 2023-11-26:
Hello, I get a consistent error when I run the `run_sorter` function with `docker=True`. I have tried this with a number of sorters such as `'ironclust'`, `'kilosort3'`, `'mountainsort5'`, `'herdingspikes'` and I always get the same error. The same code works with local sorters such as `'spykingcircus2'` and `'tridesclous2'`. This is my first time using docker and docker desktop so this may not be a bug; the images seem to have pulled just fine though.

**1)** OS, IDE: windows 11, VS code.

**2)** spikeinterface version:  I have installed spikeinterface using the [windows conda yml file](https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/full_spikeinterface_environment_windows.yml). Here is the full list of my environment packages [env.txt](https://github.com/SpikeInterface/spikeinterface/files/13465583/env.txt).

**3)** Python code:
```python
import spikeinterface.extractors as se
import spikeinterface.sorters as ss

test_recording, _ = se.toy_example(
    duration=30,
    seed=0,
    num_channels=32,
    num_segments=1,
    firing_rate=10,
    num_units=10,
)

sorter = 'kilosort3'
sorting = ss.run_sorter(sorter_name=sorter,
    recording=test_recording,
    output_folder=sorter+"_output",
    verbose=True,
    docker_image=True)
```

**4)** Error:

```python
File ...\site-packages\spikeinterface\sorters\runsorter.py:141, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    139         else:
    140             container_image = singularity_image
--> 141     return run_sorter_container(
    142         container_image=container_image,
    143         mode=mode,
...
    extractor = extractor_class(**new_kwargs)
File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/numpyextractors.py", line 137, in __init__
    if spikes.size == 0:
AttributeError: 'list' object has no attribute 'size'
```

alejoe91, 2023-11-27:
Closing then :)

samuelgarcia, 2023-11-27:
@alejoe91 We should have a way to push the recording generator inside a docker with pickle no ?
I re-open this.

PR #2271 (2023-11-30): run_sorter in container check json or pickle
run_sorter in container check json or pickle.

This close #2261.
closed 2023-11-30 by samuelgarcia