SpikeInterface/spikeinterface #2262: How to use "wobble" method in spyking-circus2?

zjm199502, 2023-11-26:
Hello sir/madam,

I want to use `wobble` matching methods in SI-spykingcircus2. but how to decide the params? 
`required_kwargs_keys = ["nbefore", "nafter", "templates"]`

How should  `nbefore` , `nafter`  and  `templates ` be defined in spykingcircus2?

Does anyone have idea on this issue?

Thank you!

Jinming

zm711, 2023-11-27:
@yger-a spykingcircus2 question

yger, 2023-11-27:
This is not straightforward to change the matching engine currently in SC2. I can make a PR to patch that, but meanwhile, if you are interested, you'll have to hack the src/sortingcomponents/sorters/internals/spyking_circus2.py file to manually pass templates/nbefore/nafter and change the matching engine to wobble. Let me know if you are interested

zjm199502, 2023-11-27:
> This is not straightforward to change the matching engine currently in SC2. I can make a PR to patch that, but meanwhile, if you are interested, you'll have to hack the [...]

Hello dear yger, I do interest in using wobble in SC2.  Actually I moved from SC to SC2, cause SC2 seems much more accurate in my ephys data. But currently the circus-omp methods is too slow for me although I have tested this in several computers (include Windows and Mac, use all 8 jobs). I saw in the SI document that wobble could be much faster than circus-omp and i tried to manually define templates/nbefore/nafter in spyking_circus2.py, but i didn't get the right way to define them. So it's very kind if you could make a PR to patch that, or just provide a example will also help a lot!

Thank you very much!

yger, 2023-11-27:
Ok, I'll make a quick PR. But did you tried to update from main? The OMP peeler has been improved and should be faster now in the main branch (not released yet)

zjm199502, 2023-11-27:
@yger  I'd say that the speed of the newer SC2 in 0.100.0 is much faster than I thought it shall be !!  what a great updating!  In 0.98.2 it takes more than 30 mins to matching, but now it is 5 mins only (I import a 30000Hz sampling, 10 mins continuous)! 
[image]
If the PR we mentioned before takes long of your time, it's not hurry.

Thank you very much again!

yger, 2023-11-30:
This should be possible via #2276, that might be merged soon

zjm199502, 2023-12-01:
> This should be possible via #2276, that might be merged soon

Thank you very much. I'll ty it later!